mavlink/mavros #1216: setpoint_raw/attitude control latency

virtualclone, 2019-04-10:
----

### Issue details

I'm using `setpoint_raw/attitude` to control body_rate and thrust in gazebo, however, I found that there's a latency between desired command and actual body_rate as in figure (Blue line is desired command and red line is actual body_rate)
[image]
Is this normal?
If yes, is there any method to solve this kind of problem?
Thanks

### MAVROS version and platform

Mavros: 0.29.0
ROS: Kinetic
Ubuntu: 16.04

vooon, 2019-04-10:
I don't think that ~1/4 sec delay is normal. Check that you don't overload your computer. Try lower rates.

virtualclone, 2019-04-11:
I reduce the rates from 50 to 30 in my control node. There's a little improvement in latency (it's around 0.15s now.)
[image]
And also I check both CPU and memory usage are around 60%.
So how long is the reasonable delay between desired command and actual body_rate?
Thanks!

Jaeyoung-Lim, 2019-04-11:
@virtualclone have you tried increasing the gains of your attitude rate controller? It is hard to say this is simply introduced from a communication delay from what you have shown

virtualclone, 2019-04-12:
@Jaeyoung-Lim , yes, I tried increasing the gains of attitude rate controller, but the problem still occurs.

sergioma295, 2019-06-07:
I am having a big problem. I have created a new issue. Can somebody help me please? #1248