mavlink/mavros #1217: Landcur request failed

iscoyizj, 2019-04-10:
----

### Issue details

Hi,

I met a few problems when using rosrun mavros landcur. The request failed as the log said.

Thank you very much for your help.

### MAVROS version and platform

Mavros: 0.27.0-20-g5ac2b2b
ROS: Kinetic
Ubuntu: 16.04

### Autopilot type and version

PX4

[v1.8.0-1489-g55b9f76] 

### Node logs

```
:~$ rosrun mavros mavcmd landcur 0 0
Landing on current coord: Lat: 47.3977513 Long: 8.5456073
Landing on current coord: Lat: 47.3977513 Long: 8.5456073
Request failed. Check mavros logs. ACK: 1
Landing on current coord: Lat: 47.3977513 Long: 8.5456073
service [/mavros/cmd/land] returned no response

```

### Diagnostics

```
~$ rostopic echo -n1 /diagnostics | tee /tmp/diag.yml
header: 
  seq: 155
  stamp: 
    secs: 1554899398
    nsecs: 294880322
  frame_id: ''
status: 
  - 
    level: 0
    name: "mavros: FCU connection"
    message: "connected"
    hardware_id: "udp://:14540@127.0.0.1:14557"
    values: 
      - 
        key: "Received packets:"
        value: "15108"
      - 
        key: "Dropped packets:"
        value: "0"
      - 
        key: "Buffer overruns:"
        value: "0"
      - 
        key: "Parse errors:"
        value: "0"
      - 
        key: "Rx sequence number:"
        value: "137"
      - 
        key: "Tx sequence number:"
        value: "142"
      - 
        key: "Rx total bytes:"
        value: "3291383"
      - 
        key: "Tx total bytes:"
        value: "110021"
      - 
        key: "Rx speed:"
        value: "23863.000000"
      - 
        key: "Tx speed:"
        value: "835.000000"
  - 
    level: 0
    name: "mavros: GPS"
    message: "3D fix"
    hardware_id: "udp://:14540@127.0.0.1:14557"
    values: 
      - 
        key: "Satellites visible"
        value: "10"
      - 
        key: "Fix type"
        value: "3"
      - 
        key: "EPH (m)"
        value: "0.00"
      - 
        key: "EPV (m)"
        value: "0.00"
  - 
    level: 0
    name: "mavros: Heartbeat"
    message: "Normal"
    hardware_id: "udp://:14540@127.0.0.1:14557"
    values: 
      - 
        key: "Heartbeats since startup"
        value: "189"
      - 
        key: "Frequency (Hz)"
        value: "0.923067"
      - 
        key: "Vehicle type"
        value: "Quadrotor"
      - 
        key: "Autopilot type"
        value: "PX4 Autopilot"
      - 
        key: "Mode"
        value: "AUTO.LAND"
      - 
        key: "System status"
        value: "Standby"
  - 
    level: 0
    name: "mavros: System"
    message: "Normal"
    hardware_id: "udp://:14540@127.0.0.1:14557"
    values: 
      - 
        key: "Sensor present"
        value: "0x00000000"
      - 
        key: "Sensor enabled"
        value: "0x00000000"
      - 
        key: "Sensor helth"
        value: "0x00000000"
      - 
        key: "CPU Load (%)"
        value: "0.0"
      - 
        key: "Drop rate (%)"
        value: "0.0"
      - 
        key: "Errors comm"
        value: "0"
      - 
        key: "Errors count #1"
        value: "0"
      - 
        key: "Errors count #2"
        value: "0"
      - 
        key: "Errors count #3"
        value: "0"
      - 
        key: "Errors count #4"
        value: "0"
  - 
    level: 0
    name: "mavros: Battery"
    message: "Normal"
    hardware_id: "udp://:14540@127.0.0.1:14557"
    values: 
      - 
        key: "Voltage"
        value: "12.15"
      - 
        key: "Current"
        value: "-1.0"
      - 
        key: "Remaining"
        value: "100.0"
  - 
    level: 0
    name: "mavros: Time Sync"
    message: "Normal"
    hardware_id: "udp://:14540@127.0.0.1:14557"
    values: 
      - 
        key: "Timesyncs since startup"
        value: "661"
      - 
        key: "Frequency (Hz)"
        value: "9.999900"
      - 
        key: "Last RTT (ms)"
        value: "0.421504"
      - 
        key: "Mean RTT (ms)"
        value: "0.490559"
      - 
        key: "Last remote time (s)"
        value: "255.222383000"
      - 
        key: "Estimated time offset (s)"
        value: "1554899143.113989353"
---

```

### Check ID

```
$ rosrun mavros checkid
OK. I got messages from 1:1.

---
Received 5969 messages, from 1 addresses
sys:comp   list of messages
  1:1     32, 33, 2, 4, 230, 0, 1, 105, 74, 140, 141, 111, 76, 241, 242, 147, 245, 24, 26, 30, 31

```

iscoyizj, 2019-04-14:
Hi？

iscoyizj, 2019-04-21:
hello？

TSC21, 2019-04-24:
@iscoyizj without any further details, it's kinda impossible to help you. What are you trying to do exactly? What's the output of the flight controller? (not the one that comes from MAVROS, but rather what you inspect on QGC).

iscoyizj, 2019-04-25:
First I use 
``` 
mavros mavcmd takeoffcur 0 0 3 
```
to make the drone takeoff,then I tried  
```
mavros mavcmd landcur 
```
which then have the problems. But I now replace the mavcmd landcur with set flight mode to auto.land and avoid the problem. I will run the node another day and check the flight controller. Thank you for your help.

sergioma295, 2019-06-07:
I am having a big problem. I have created a new issue. Can somebody help me please? #1248