mavlink/mavros #1221: CON: Got HEARTBEAT, connected. FCU: No valid autopilot

huhu1030, 2019-04-18:
### Issue details

Hi guys,

I'm trying to launch Mavros on my drone (Intel Aero) with the command 
_roslaunch mavros apm.launch fcu_url:="udp://:14555@127.0.0.1:14555"_
But I have this weird issue **"Got HEARTBEAT, connected. FCU: No valid autopilot"** and also **"ERROR: I got 1 addresses, but not your target1:1"** (when I launch rosrun mavros checkid).
I search through the web the whole day, and apparently, the issue is coming from the system_id and component_id not being set correctly.
The problem is that I checked these values on QGroundControl and they are 1/1. These values are by default 1/1 in the apm.launch file.

If someone can point me in the right direction I'll be happy.
Thank you very much.

### MAVROS version and platform

Mavros: bot sure
ROS: Kinetic
Ubuntu: 16.04

### Autopilot type and version

[ X] ArduPilot
[  ] PX4

Version: 3.5.3

### Node logs
roslaunch mavros apm.launch fcu_url:="udp://:14555@127.0.0.1:14555"
... logging to /home/drone02/.ros/log/7960effa-3c78-11e9-9a16-4a65e9c04346/roslaunch-drone02-2604.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://drone02:36828/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_mav_type: ONBOARD_CONTROLLER
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/rangefinder_pub/field_of_view: 0.0
 * /mavros/distance_sensor/rangefinder_pub/frame_id: lidar
 * /mavros/distance_sensor/rangefinder_pub/id: 0
 * /mavros/distance_sensor/rangefinder_pub/send_tf: False
 * /mavros/distance_sensor/rangefinder_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/rangefinder_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/rangefinder_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/rangefinder_sub/id: 1
 * /mavros/distance_sensor/rangefinder_sub/orientation: PITCH_270
 * /mavros/distance_sensor/rangefinder_sub/subscriber: True
 * /mavros/fake_gps/eph: 2.0
 * /mavros/fake_gps/epv: 2.0
 * /mavros/fake_gps/fix_type: 3
 * /mavros/fake_gps/geo_origin/alt: 408.0
 * /mavros/fake_gps/geo_origin/lat: 47.3667
 * /mavros/fake_gps/geo_origin/lon: 8.55
 * /mavros/fake_gps/gps_rate: 5.0
 * /mavros/fake_gps/mocap_transform: True
 * /mavros/fake_gps/satellites_visible: 5
 * /mavros/fake_gps/tf/child_frame_id: fix
 * /mavros/fake_gps/tf/frame_id: map
 * /mavros/fake_gps/tf/listen: False
[... 335 lines cut ...]
        value: "665"
      - 
        key: "Frequency (Hz)"
        value: "10.001075"
      - 
        key: "Last RTT (ms)"
        value: "0.503029"
      - 
        key: "Mean RTT (ms)"
        value: "0.661527"
      - 
        key: "Last remote time (s)"
        value: "1551482443.961940765"
      - 
        key: "Estimated time offset (s)"
        value: "9223372036.854604721"
---

### Check ID

ERROR. I got 1 addresses, but not your target 1:1
---
Received 332 messages, from 1 addresses
sys:comp   list of messages
  1:240   0, 2, 76, 111

huhu1030, 2019-04-18:
I updated Ardupilot from 3.5.3 to 3.7.0. Still the same issue..

vooon, 2019-04-19:
You use same port for bind and remote, mavros simply got it's own heartbeats. See "Vehicle type".
You should use different ports, like `udp://:14550@localhost:14555` (flight firmware should listen to purt 14555).

huhu1030, 2019-04-23:
Okay, it was indeed a port problem. Thank you very much.
Now I'm trying to launch a basic node that arm and takeoff. But as soon as it arms the drone, the remote address changes to another IP and thus Mavros is losing connection. Do you know why?

vooon, 2019-04-23:
Some other computer sending the data to port that mavros binds to. Because UDP is connectless, libmavconn uses address of the last packet to talk back. Try to use other port, e.g `udp://:14540@`.

huhu1030, 2019-04-23:
if I use 14540 it's not connecting to fcu.

vooon, 2019-04-25:
Because you should first configure your AP to use another port.

huhu1030, 2019-04-26:
I've fixed this by changing the IP to localhost ip: 127.0.0.1.
But can you tell me where I can change the port ? Thanks.

vooon, 2019-04-28:
That is really depends on flight firmware you are using. E.g. as i remember, APM may be tuned by command line args.

huhu1030, 2019-05-07:
I don't know about APM but you can change the port inside the Mavlink-router conf file.
I'm facing another issue. It appears that I can connect to the fcu only 1 when system_id target_id.
I'm changing the SYSMAV_ID to 25, same with the target_id but It doesn't seems to work.

`[ INFO] [1557197554.695418149]: MAVROS started. MY ID 25.1, TARGET ID 25.1`

CheckId returns : 
`ERROR. I got 1 addresses, but not your target 25:1

---
Received 1708 messages, from 1 addresses
sys:comp   list of messages
  1:1     0, 1, 2, 22, 129, 136, 152, 147, 150, 24, 27, 29, 30, 32, 33, 163, 36, 165, 42, 178, 182, 62, 65, 74, 193, 164, 111, 241, 116, 125
`

huhu1030, 2019-05-07:
I've fixed it.
Once you modify the SYSMAV_ID on GCS, you need to reboot.
Now the target_id == SYSMAV_ID and FCU is connected.
Thanks Vooon for your time and help.

closed 2019-05-08 by vooon