mavlink/mavros #1223: [Bug] Eigen Assertions with PX4 master branch - Possibly ARM-related.

ChuplesKai, 2019-04-22:
This is only bug and feature tracker, please use it
to report bugs or request features.

----

### Issue details

_When attempting to start a MAVROS node on an NVIDIA Jetson TX2, an Eigen assertion fails during initialization.  Seems specific to the TX2 (or possibly ARM-Architecture), and stems from mavros::extra_plugins::OdometryPlugin::handle_odom()_

### MAVROS version and platform

Mavros: latest - ?0.28.0?
ROS: Kinetic
Ubuntu: 16.04

### Autopilot type and version

[  ] ArduPilot
[ X ] PX4

Version: master development branch

### Node logs

```
Check Attached
```

### Diagnostics

```
Check Attached
```

### Check ID

```
Check Attached
```
[checkid.txt](https://github.com/mavlink/mavros/files/3104372/checkid.txt)
[diagnostics.txt](https://github.com/mavlink/mavros/files/3104373/diagnostics.txt)
[terminal_out.txt](https://github.com/mavlink/mavros/files/3104374/terminal_out.txt)

ChuplesKai, 2019-04-22:
As additional information, I've been working with @TSC21 to try resolving this issue.  I performed a GDB trace of the call, which I'm attaching here.

Interestingly, it seems to be the same issue reported here: http://discuss.px4.io/t/mavros-error-in-master-firmware/9312 .  I confirmed that switching to the PX4 stable release (1.8.2) does prevent the assertion; however, our application requires the vision functionality of the latest PX4.  

Specifically, the issue happens during initialization of MAVROS, when at some point during construction, the Mavlink Odom plugin attempts to handle a message.  For some reason, an Eigen assertion is being thrown, seemingly related to a Block Matrix operation.  My assumption is that this is somehow related to a covariance matrix of some sort, but I am unsure.

@TSC21 reports never running into this issue on an x86 platform, and I will be testing this for my own set-up as well.  I will update when I have completed that test.

[gdb_bt.txt](https://github.com/mavlink/mavros/files/3104393/gdb_bt.txt)

vooon, 2019-04-23:
I've added some more eigen aligned allocators, perhaps that may help.
Can you please check on the master?

ChuplesKai, 2019-04-23:
Pulled and re-built.  Double-checked to confirm that I have commit 81ca775; however, the assertion is still raised.  I'm attaching the GDB trace, which seems to have slightly changed, but not really in a significant fashion (i.e. still looks like the same problem from the same plugin).

I don't know if it's helpful for debugging purposes, but I am running Eigen 3.3.9.
[gdb_bt.txt](https://github.com/mavlink/mavros/files/3108406/gdb_bt.txt)

vooon, 2019-04-25:
Well, that needs to go deeper into method code and i do not have enough time now.
For now you can blacklist odometry plugin.

ChuplesKai, 2019-04-25:
@vooon Okay, I have blacklisted the odom plugin for now, and it seems to be the only plugin which was raising such an issue.  Thank you for looking into this.

sergioma295, 2019-06-07:
I am having a big problem. I have created a new issue. Can somebody help me please? #1248

hyonlim, 2019-07-22:
Hi,
I've reproduced same error for current master (on Jetson TX2, 16.04 LTS, ROS Kinetic).
How would we solve this problem? Any idea?

RBinsonB, 2020-05-07:
I was having the same error with Eigen as well on Ubuntu 16.04 LTS and ROS Kinetic and it was crashing my mavros node when I was forwarding odometry:

 ```
mavros_node: /usr/include/eigen3/Eigen/src/Core/util/XprHelper.h:70: Eigen::internal::variable_if_dynamic<T, Value>::variable_if_dynamic(T) [with T = long int; int Value = 4]: Assertion 'v == T(Value)' failed.`
 ```

Tracking down the error I realized that it was coming from Eigen and the impossibilty to inverse a Affine3d transform without crashing. Checking the library version (with `sudo dpkg -l | grep eigen`) showed that I had a beta version installed:

 ```
ii  libeigen3-dev                                    3.3~beta1-2                                           all          lightweight C++ template library for linear algebra
 ```

Default distro version of Eigen is 3.3 Beta for some reason so I removed it ( `sudo apt-get remove libeigen-dev` ) and downloaded 3.3.4 here: [https://packages.ubuntu.com/bionic/all/libeigen3-dev/download](https://packages.ubuntu.com/bionic/all/libeigen3-dev/download) and installed running `sudo dpkg -i libeigen3-dev_3.3.4-4_all.deb`

It solved the issue for me, not sure it will for everyone.