mavlink/mavros #1224: mavros/imu/data angular_velocity should be in rad/sec instead of deg/sec

geoffreyvd, 2019-04-24:
### Issue details
According to the ROS docs, the rotational velocities should be given in rad/sec, but when I access /mavros/imu/data.angular_velocity.z it is given in deg/sec

### MAVROS version and platform
Mavros: 0.18.4
ROS: Kinetic
Ubuntu: 16.04

### Autopilot type and version
[  ] ArduPilot
[ x ] PX4

Version: 1.8.2

vooon, 2019-04-25:
https://mavlink.io/en/messages/common.html#ATTITUDE it is already rad/s.
https://mavlink.io/en/messages/common.html#HIGHRES_IMU

geoffreyvd, 2019-04-25:
Mmm, well it should be. But I use /mavros/imu/data.angular_velocity.z in my application, and I am 100% sure that it's in degree/sec

geoffreyvd, 2019-04-26:
My bad, it indeed is in rad/sec. I forgot to multiply the rotation per second by 0.02 (the amount of ms associated with an imu message)

closed 2019-04-26 by geoffreyvd