mavlink/mavros #1225: Problem with /mavros/setpoint_raw/attitude

yongjianwen, 2019-04-25:
Hi, I am facing a problem with sending attitude command to the UAV via the `/mavros/setpoint_raw/attitude` topic. Previously, I have tried and succeeded in sending velocity command (in OFFBOARD mode) via the `/mavros/setpoint_velocity/cmd_vel` topic. In that case, I only sent the values in `Twist.Linear.X`, `Twist.Linear.Y`, `Twist.Linear.Z` and `Twist.Angular.Z`, and it worked perfectly with no errors. However, now that when I wanted to send attitude command in OFFBOARD mode, the QGroundControl would display an error message `Failsafe enabled: no offboard`.

[image]

I kept everything the same and the only difference is the topic used (`/mavros/setpoint_velocity/cmd_vel` versus `/mavros/setpoint_raw/attitude`). Also `rostopic echo /mavros/setpoint_raw/attitude` showed that the command has been received.

```
header:
  seq: 0
  stamp:
    secs: 0
    nsecs:        0
  frame_id: ''
type_mask: 0
orientation:
  x: 0.00569751797229
  y: -0.009772340308063
  z: 5.56814351204e-05
  w: 0.999936016594
body_rate:
  x: 0.0
  y: 0.0
  z: 0.0
thrust: 0.5789999996185
```

Could anyone help us figure out the problem causing the OFFBOARD mode to fail? Thanks in advance for helping!

Jaeyoung-Lim, 2019-04-26:
@yongjianwen Looks like you are not sending any timestamps with the messages. You also need to maintain the rate of the setpoint stream within the timeout(default 0.5s)

vooon, 2019-04-26:
In addition to @Jaeyoung-Lim, to use raw message you also should fill `type_mask`.

yongjianwen, 2019-04-26:
@Jaeyoung-Lim @vooon Thanks for your prompt replies!

Regarding the recommendation no. 1 by @Jaeyoung-Lim , I previously left the timestamp to be blank while using `/mavros/setpoint_velocity/cmd_vel` and it worked just fine (I have just double checked and it was really blank). However, since the topics used are different, I will definitely give it a try this time.

For the recommendation no. 2, I checked the frequency of `/mavros/setpoint_raw/attitude` at which Simulink is publishing, and there is no problem (over 30 Hz). Also I have made sure that Simulink is publishing the data before switching to OFFBOARD mode. So this should not be a problem.

Additionally, in order to enter OFFBOARD mode, is it necessary to fuse the pose data from Vicon?

@vooon Yeah regarding the `type_mask` value, do you happen to have found a link to its documentation? I have searched around in Google but the amount of resource seems to be scarce. The following documentation from ROS is also not very clear. Does it mean that, for my case, `type_mask = 1 + 2 + 4 = 7` since I need to use only thrust and attitude?

[image]

I will update here again regarding the outcome of the recommendations. Thanks in advance!

Jaeyoung-Lim, 2019-04-27:
@yongjianwen You can look at this project as an example: https://github.com/Jaeyoung-Lim/mavros_controllers

vooon, 2019-04-28:
Message doc: https://mavlink.io/en/messages/common.html#SET_ATTITUDE_TARGET

You should `bit or` (or sum), use constants from ros message for better understanding.
Note that in mavlink in description bits are counted starting from 1 (not usual 0).

JIMMMY0, 2019-05-01:
We finally found out that the problem is that the transmitting rate is too low for our wifi module. 
The problem is solved once we changed the wifi module or connect the companion computer using an ethernet cable.  @Jaeyoung-Lim @vooon Appreciate for all of your suggestions!

closed 2019-05-02 by vooon

pedrowunsch27, 2020-03-21:
Hello @yongjianwen ,
I have come with the same problem and I didn't find a solution with the setpoint_attitude_raw. Could you please show me the code in which you use it? It would be of great help.
Thank you!

Jaeyoung-Lim, 2020-03-25:
@pedrowunsch27 I have already linked the code in the answers above