mavlink/mavros #1226: PX4 SITL CMD: Unexpected command 22, result 0

mahdinobar, 2019-04-26:
This is only bug and feature tracker, please use it
to report bugs or request features.

----

### Issue details

I use MAVROS node to translate MAVLink messages into ROS messages so I  can communicate between ROS and Gazebo through PX4. The problem is when the commander offboard control phase finishes, I receive the warning **CMD: Unexpected command 22, result 0** then the drone lands, whereas I should continue the flight based on the velocity command from my ROS node. So my question is what does this warning message mean and how it can be corrected?

### MAVROS version and platform

Mavros: ?0.18.4?
ROS: ?Kinetic?
Ubuntu: ?16.04?

### Autopilot type and version

PX4

Version: ?v1.8.2?

JIMMMY0, 2019-04-27:
I have the same warning but it doesn't effect controlling the drone in Gazebo. It appears once after mavros finishes loading all the plugins.

vooon, 2019-04-28:
For some reason FCU sends COMMAND_ACK for command 22, while mavros do not have transaction for that command.
It might be addressed to GCS.

closed 2019-04-28 by vooon