mavlink/mavros #1227: [Bug] Arm state unstable

nicovanduijn, 2019-04-30:
### Issue details

In our system we sometimes encounter a situation where the drone is armed, but the mavros on our companion computer is publishing alternating 'True' and 'False' on the `/mavros/state/armed` topic.

I'm not sure how easy this is to reproduce. It comes and goes quite randomly on our system.

### MAVROS version and platform

Mavros: 0.29.0
ROS: Kinetic
Ubuntu: 16.04

### Autopilot type and version

[  ] ArduPilot
[ X] PX4

Version: Upstream

### Node logs
[image]

[Rosbag](https://drive.google.com/file/d/1Op0GOlDSmBwJJSSiT0m_suO52EQdfQJQ/view?usp=sharing)

vooon, 2019-04-30:
Probably you have several sources. Please catch some more full `/mavros/state`.
Also need to have diagnostics and node logs.

nicovanduijn, 2019-05-02:
It's difficult to reproduce, but I'll make sure to log `/diagnostics` and all mavros topics next time I see it.

What else do you need? I did log `mavros/state`, right? (In the bag linked above)

nicovanduijn, 2019-05-02:
Here's a bag with all mavros topics and the `/diagnostics`
[rosbag](https://drive.google.com/file/d/1OJiujZpt9mhwJPHtChD1UHHRwUrDucmD/view?usp=sharing)

nicovanduijn, 2019-05-03:
Turns out we ran into this: #1107

TSC21, 2019-05-05:
@nicovanduijn this gets fixed with #1228 right?

nicovanduijn, 2019-05-06:
Yes!

closed 2019-05-06 by nicovanduijn