mavlink/mavros #1231: NED -> ENU transformation

mrivi, 2019-05-09:
Vectors aren't correctly converted from NED to ENU frame and vice versa if a component is NAN.

----

### Issue details

Example: If I have vector (NAN, NAN, 0.7) in ned frame, I would expect to have (NAN, NAN, -0.7) in enu. However the transform gives (NAN, NAN, NAN).

### MAVROS version and platform

Mavros: 0.29.2
ROS: Melodic
Ubuntu: 18.04

### Autopilot type and version

[  ] ArduPilot
[ X ] PX4

Version: master

mrivi, 2019-05-16:
@TSC21 @vooon - could you please review this PR. Thank you

TSC21, 2019-05-16:
@mrivi netiher of the Eigen approaches using a rotation matrix worked? I mean is there a way we could replace NaNs with zero in a intermediate step or have Eigen interpret NaNs as zeros, without having to hardcode the rotation?

vooon, 2019-05-16:
Nuno, Eigen dosen't do any assumptions, calculations with NaNs definitely will result in NaNs on output.

TSC21, 2019-05-16:
> Nuno, Eigen dosen't do any assumptions, calculations with NaNs definitely will result in NaNs on output.

Ok it may be possible to consider NaNs in transforms somehow. I guess we could add an intermediate step that checks for NaNs, but probably too much work.

vooon, 2019-05-16:
In first place we should find places where vectors may become NaN (by specs!).
And only then do something, instead of changing behaviour of lib functions.

jkflying, 2019-05-17:
The approach in https://github.com/mavlink/mavros/pull/1233 uses axes reassignments (or a double reflection around x=y and z=0, if you prefer to think of it that way) instead of rotations, which is much simpler, lossless and in the case of something simple like NED <--> ENU arguable more suitable.

Rotations are generalized, true, but also introduce floating point issues (like what we're seeing here) and higher computational expense. Note that NaN isn't the only problem, we would have similar issues with +-0 and +-Inf contaminating across axes even though by definition the axes are aligned.

`StaticTF::ENU_TO_NED` is already an enum, not a rotation matrix, so I don't see this as a hack, but rather as a transform-specific optimization, since using a whole affine transform is overkill (and is lossy) for something we know by definition is static, unscaled and axes-aligned.  It could also be adapted easily to the covariances, which right now suffer from the same issue (eg. infinite variance on yaw due to unknown heading --> infinite covariance on all axes, since cos(pi/2) != 0.0 exactly ).

mrivi, 2019-05-23:
PR merged, closing

closed 2019-05-23 by mrivi