mavlink/mavros #1232: MAVROS setpoint_raw/global

peasant98, 2019-05-12:
This is only bug and feature tracker, please use it
to report bugs or request features.

----

### Issue details
Hello,

I am currently working with ArduPilot SITL with MAVROS. I am testing out the rostopic which is /mavros/setpoint_raw/global. I was curious as to how I can integrate position (gps coordinates) and velocity correctly. Do I need to use MAV_CMD_DO_CHANGE_SPEED? Or do I need to know the velocity vector of the trajectory so I can adjust the velocity components accordingly? The reason I am asking is because I would like to have the copter go from location to location much faster than the speed it goes normally. I am unsure if there is a bug, or some feature I am unaware of. Thanks!

### MAVROS version and platform

Mavros: 0.18.4
ROS: Kinetic
Ubuntu: 16.04

### Autopilot type and version

[X] ArduPilot
[  ] PX4

Version: ?3.7.1?

### Node logs

```
copy output of mavros_node. Usually console where you run roslaunch

... logging to /home/matt/.ros/log/8b221cae-74dc-11e9-b66b-f82819c0bd23/roslaunch-Lenovo-29975.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://Lenovo:46401/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_mav_type: ONBOARD_CONTROLLER
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/rangefinder_pub/field_of_view: 0.0
 * /mavros/distance_sensor/rangefinder_pub/frame_id: lidar
 * /mavros/distance_sensor/rangefinder_pub/id: 0
 * /mavros/distance_sensor/rangefinder_pub/send_tf: False
 * /mavros/distance_sensor/rangefinder_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/rangefinder_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/rangefinder_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/rangefinder_sub/id: 1
 * /mavros/distance_sensor/rangefinder_sub/orientation: PITCH_270
 * /mavros/distance_sensor/rangefinder_sub/subscriber: True
 * /mavros/fake_gps/eph: 2.0
 * /mavros/fake_gps/epv: 2.0
 * /mavros/fake_gps/fix_type: 3
 * /mavros/fake_gps/geo_origin/alt: 408.0
 * /mavros/fake_gps/geo_origin/lat: 47.3667
 * /mavros/fake_gps/geo_origin/lon: 8.55
 * /mavros/fake_gps/gps_rate: 5.0
 * /mavros/fake_gps/mocap_transform: True
 * /mavros/fake_gps/satellites_visible: 5
 * /mavros/fake_gps/tf/child_frame_id: fix
 * /mavros/fake_gps/tf/frame_id: map
 * /mavros/fake_gps/tf/listen: False
[... 425 lines cut ...]
      - 
        key: "Mean RTT (ms)"
        value: "5.889335"
      - 
        key: "Last remote time (s)"
        value: "630.292094000"
      - 
        key: "Estimated time offset (s)"
        value: "1557682647.361073732"
---

```

### Check ID

```
rosrun mavros checkid

OK. I got messages from 1:1.

---
Received 1683 messages, from 1 addresses
sys:comp   list of messages
  1:1     0, 1, 2, 132, 22, 129, 136, 152, 147, 150, 24, 27, 29, 30, 32, 33, 163, 36, 165, 42, 173, 178, 182, 62, 65, 74, 193, 164, 111, 241, 116, 125
```

TSC21, 2019-05-12:
From my understanding, that's something to with the position controller on APM rather than something you need to worry about on the MAVROS/ROS side. If you want to control do the velocity, use a velocity controller by closing the loop and send velocity setpoints instead. If you want to use the controller on the APM side, bring this question to the APM forums.

closed 2019-05-12 by TSC21

peasant98, 2019-05-13:
Alright, thanks! I'll definitely look into that because most of that work (that I need to know) is implemented in ArduCopter.