mavlink/mavros #1235: Source installation failed

BreederBai, 2019-05-16:
This is only bug and feature tracker, please use it
to report bugs or request features.

----

### Issue details

https://github.com/mavlink/mavros/blob/master/mavros/README.md#installation
Follow the tutorial in the link above, Source installation，error。

Solve the dependencies I use the following commands：
    rosdep install -y --from-paths src --ignore-src --rosdistro kinetic -r --os=debian:stretch

ROS I installed according to the following website：
  http://wiki.ros.org/ROSberryPi/Installing%20ROS%20Kinetic%20on%20the%20Raspberry%20Pi
### MAVROS version and platform

Mavros: 0.28.0
ROS: Kinetic
Raspbian:stretch

### Node logs

```
pi@raspberrypi:~/src/catkin_ws $ catkin build
-----------------------------------------------------------
Profile:                     default
Extending:          [cached] /opt/ros/kinetic
Workspace:                   /home/pi/src/catkin_ws
-----------------------------------------------------------
Build Space:        [exists] /home/pi/src/catkin_ws/build
Devel Space:        [exists] /home/pi/src/catkin_ws/devel
Install Space:      [unused] /home/pi/src/catkin_ws/install
Log Space:          [exists] /home/pi/src/catkin_ws/logs
Source Space:       [exists] /home/pi/src/catkin_ws/src
DESTDIR:            [unused] None
-----------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
-----------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
-----------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
-----------------------------------------------------------
Workspace configuration appears valid.
-----------------------------------------------------------
[build] Found '6' packages in 0.0 seconds.                                     
[build] Package table is up to date.                                           
Starting  >>> mavlink                                                          
Starting  >>> mavros_msgs                                                      
_______________________________________________________________________________
Errors     << mavros_msgs:cmake /home/pi/src/catkin_ws/logs/mavros_msgs/build.cmake.006.log
CMake Warning at /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:76 (find_package):
  Could not find a package configuration file provided by "geographic_msgs"
  with any of the following names:

    geographic_msgsConfig.cmake
    geographic_msgs-config.cmake

  Add the installation prefix of "geographic_msgs" to CMAKE_PREFIX_PATH or
  set "geographic_msgs_DIR" to a directory containing one of the above files.
  If "geographic_msgs" provides a separate development package or SDK, be
  sure it has been installed.
Call Stack (most recent call first):
  CMakeLists.txt:4 (find_package)

CMake Error at /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by "geographic_msgs"
  with any of the following names:

    geographic_msgsConfig.cmake
    geographic_msgs-config.cmake

  Add the installation prefix of "geographic_msgs" to CMAKE_PREFIX_PATH or
  set "geographic_msgs_DIR" to a directory containing one of the above files.
  If "geographic_msgs" provides a separate development package or SDK, be
  sure it has been installed.
Call Stack (most recent call first):
  CMakeLists.txt:4 (find_package)

cd /home/pi/src/catkin_ws/build/mavros_msgs; catkin build --get-env mavros_msgs | catkin env -si  /usr/bin/cmake /home/pi/src/catkin_ws/src/mavros/mavros_msgs --no-warn-unused-cli -DCATKIN_DEVEL_PREFIX=/home/pi/src/catkin_ws/devel/.private/mavros_msgs -DCMAKE_INSTALL_PREFIX=/home/pi/src/catkin_ws/install; cd -
...............................................................................
Failed     << mavros_msgs:cmake            [ Exited with code 1 ]              
Failed    <<< mavros_msgs                  [ 3.3 seconds ]                     
Abandoned <<< libmavconn                   [ Unrelated job failed ]            
Abandoned <<< mavros                       [ Unrelated job failed ]            
Abandoned <<< mavros_extras                [ Unrelated job failed ]            
Abandoned <<< test_mavros                  [ Unrelated job failed ]            
Finished  <<< mavlink                      [ 4.5 seconds ]                     
[build] Summary: 1 of 6 packages succeeded.                                    
[build]   Ignored:   None.                                                     
[build]   Warnings:  None.                                                     
[build]   Abandoned: 4 packages were abandoned.                                
[build]   Failed:    1 packages failed.                                        
[build] Runtime: 5.4 seconds total.          
```

vooon, 2019-05-16:
Check that you have `geographic_msgs` package.

BreederBai, 2019-05-16:
@vooon thank you for your help！I found that the command to resolve dependencies will give an error.
```
rosdep install --from-paths src --ignore-src -y

ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
```
I changed the command to try again,prompt me to rely on all solved
```
pi@raspberrypi:~/src/catkin_ws $ rosdep install -y --from-paths src --ignore-src --rosdistro kinetic -r --os=debian:stretch
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
mavros: No definition of [geographic_msgs] for OS [debian]
mavros_extras: No definition of [tf2_eigen] for OS [debian]
test_mavros: No definition of [control_toolbox] for OS [debian]
mavros_msgs: No definition of [geographic_msgs] for OS [debian]
Continuing to install resolvable dependencies...
#All required rosdeps installed successfully
```
But in fact, at the time of compilation, I still reported an error.
```
pi@raspberrypi:~/src/catkin_ws $ catkin build 
--------------------------------------------------------------------------------
Profile:                     default
Extending:             [env] /home/pi/src/uuid_msgs_ws/devel:/home/pi/src/geogra_ws/devel:/home/pi/src/catkin_ws/devel:/opt/ros/kinetic
Workspace:                   /home/pi/src/catkin_ws
--------------------------------------------------------------------------------
Build Space:        [exists] /home/pi/src/catkin_ws/build
Devel Space:        [exists] /home/pi/src/catkin_ws/devel
Install Space:      [unused] /home/pi/src/catkin_ws/install
Log Space:         [missing] /home/pi/src/catkin_ws/logs
Source Space:       [exists] /home/pi/src/catkin_ws/src
DESTDIR:            [unused] None
--------------------------------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
--------------------------------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
--------------------------------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
--------------------------------------------------------------------------------
Workspace configuration appears valid.
--------------------------------------------------------------------------------
[build] Found '6' packages in 0.0 seconds.                                     
[build] Updating package table.                                                
Warning: generated devel space setup files have been deleted.
Starting  >>> catkin_tools_prebuild                                            
Finished  <<< catkin_tools_prebuild                [ 15.2 seconds ]            
Starting  >>> mavlink                                                          
Starting  >>> mavros_msgs                                                      
_______________________________________________________________________________
Errors     << mavros_msgs:cmake /home/pi/src/catkin_ws/logs/mavros_msgs/build.cmake.000.log
CMake Warning at /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:76 (find_package):
  Could not find a package configuration file provided by "geographic_msgs"
  with any of the following names:

    geographic_msgsConfig.cmake
    geographic_msgs-config.cmake

  Add the installation prefix of "geographic_msgs" to CMAKE_PREFIX_PATH or
  set "geographic_msgs_DIR" to a directory containing one of the above files.
  If "geographic_msgs" provides a separate development package or SDK, be
  sure it has been installed.
Call Stack (most recent call first):
  CMakeLists.txt:4 (find_package)

CMake Error at /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by "geographic_msgs"
  with any of the following names:

    geographic_msgsConfig.cmake
    geographic_msgs-config.cmake

  Add the installation prefix of "geographic_msgs" to CMAKE_PREFIX_PATH or
  set "geographic_msgs_DIR" to a directory containing one of the above files.
  If "geographic_msgs" provides a separate development package or SDK, be
  sure it has been installed.
Call Stack (most recent call first):
  CMakeLists.txt:4 (find_package)

cd /home/pi/src/catkin_ws/build/mavros_msgs; catkin build --get-env mavros_msgs | catkin env -si  /usr/bin/cmake /home/pi/src/catkin_ws/src/mavros/mavros_msgs --no-warn-unused-cli -DCATKIN_DEVEL_PREFIX=/home/pi/src/catkin_ws/devel/.private/mavros_msgs -DCMAKE_INSTALL_PREFIX=/home/pi/src/catkin_ws/install; cd -
...............................................................................
Failed     << mavros_msgs:cmake                    [ Exited with code 1 ]      
Failed    <<< mavros_msgs                          [ 40.8 seconds ]            
Abandoned <<< libmavconn                           [ Unrelated job failed ]    
Abandoned <<< mavros                               [ Unrelated job failed ]    
Abandoned <<< mavros_extras                        [ Unrelated job failed ]    
Abandoned <<< test_mavros                          [ Unrelated job failed ]    
Finished  <<< mavlink                              [ 3 minutes and 3.0 seconds ]
[build] Summary: 2 of 7 packages succeeded.                                    
[build]   Ignored:   None.                                                     
[build]   Warnings:  None.                                                     
[build]   Abandoned: 4 packages were abandoned.                                
[build]   Failed:    1 packages failed.                                        
[build] Runtime: 3 minutes and 18.8 seconds total.        
```

vooon, 2019-05-16:
Rosdep tells you, that it can't find dep packages, that is why you have error.
Use `rosinstall_generator` to add that three packages to your ws/src (it generates yaml description that then should be passed to `wstool`).

BreederBai, 2019-05-16:
@vooon Thank you again for your patience.The path below is the path I use to install ROS from the source.
```
/home/pi/src/ros_catkin_ws/src
```
[image]

[image]

I don't know much about ROS. How can I use rosinstall_generator to add the packages I need? I would be grateful if you could give example commands.

vooon, 2019-05-16:
@BreederBai look at [source installation](https://github.com/mavlink/mavros/blob/master/mavros/README.md#source-installation), here you can get some examples, just replace mavros with other packages.

BreederBai, 2019-05-16:
@vooon There are dependencies in these dependencies, do I need to create them one by one? Is there a simple way, such as putting them into a ws/src and compiling them together?My English is not very good, in fact I am using Google Translate. I hope you can understand what I said, thank you again for your help.

vooon, 2019-05-20:
Rosinstall-generator can pull all deps, but i don't think that you really need to pull them all (and build whole ros from source).
Yes, you can put git clones (or any valid dirs with package.xml) into ws/src/. Generator just knows where to get sources (trough rosdistro).

BreederBai, 2019-05-21:
@vooon Thanks for your guidance, I finally chose to compile ROS and mavros together through source code.
```
rosinstall_generator ros_comm ros_control joystick_drivers mavlink mavros --rosdistro kinetic --deps --wet-only --tar > kinetic-custom_ros.rosinstall
```
Because of the lack of understanding of ros, I can only choose a feasible but perhaps not the best way. In fact, this method is also tried by me according to the tutorial of ros source installation. I don't know the meaning of each command and parameter installed in the source code, I can only try it. Fortunately, I succeeded, but it took a long time to install the source. It is estimated that it takes more than a dozen hours, because I am not next to the Raspberry Pi on Saturday and Sunday, so I can't know the exact time. Because I used raspbian, only the source code installed ros, so my method is not so bad. If I install ros, it is better to install mavros together, eliminating the need to repeatedly compile and install ros.

vooon, 2019-05-21:
Yes, source installation takes a lot of time because it have to compile dozen C++ files.
Especially on weak machines like raspi. But you have some ways to speedup, and the first one - to use external USB drive (or SSD) because SD cards much slower. Second one - distcc.

BreederBai, 2019-05-22:
@vooon This is a very good idea. I still have a raspberry pie in my hand, and your answer can save me a lot of time. Thank you for sharing your experience with me.

closed 2019-05-31 by TSC21