mavlink/mavros #1236: error   roslaunch mavros apm2.launch

zpzp307, 2019-05-19:
This is only bug and feature tracker, please use it
to report bugs or request features.

----

### Issue details

_Please describe the problem, or desired feature_

### MAVROS version and platform

Mavros: ?0.18.4?
ROS: ?Kinetic?
Ubuntu: ?16.04?

### Autopilot type and version

[  ] ArduPilot
[  ] PX4

Version: ?3.7.1?

### Node logs

```
copy output of mavros_node. Usually console where you run roslaunch
```

### Diagnostics

```
place here result of:
rostopic echo -n1 /diagnostics
```

### Check ID

```
rosrun mavros checkid
```

zpzp307, 2019-05-19:
error loading <rosparam> tag:  	file /opt/ros/kinetic/share/mavros/launch/apm_config.yaml contains invalid YAML: while parsing a block mapping   in "<string>", line 4, column 1:     startup_px4_usb_quirk: false     ^ expected <block end>, but found '<block mapping start>'   in "<string>", line 103, column 2:      setpoint_raw:      ^ XML is <rosparam command="load" file="$(arg config_yaml)"/> The traceback for the exception was written to the log file

TSC21, 2019-05-19:
duplicate of: https://github.com/mavlink/mavros/issues/1234

closed 2019-05-19 by TSC21