mavlink/mavros #1240: Sending STATUSTEXT Message to Ardupilot

TiffanyKalin, 2019-05-26:
This is only bug and feature tracker, please use it
to report bugs or request features.

----

### Issue details

Hello, this is more of a question. I am trying to send a custom statustext message from my ros script to Mission Planner. I am publishing my StatusText message on /mavros/statustext/send, and it is going to /rosout. However, I am not seeing it show on Ardupilot SITL (APMRover vehicle type) or on Mission Planner. I'm not sure if there's something special I have to do in order for Ardupilot to process a statustext from mavros. 

I am able to send commands just fine and can publish to velocity topics and have them received by Ardupilot. 

### MAVROS version and platform

Mavros: 0.29.2
ROS: kinetic
Ubuntu: 16.04

### Autopilot type and version

[ X] ArduPilot
[  ] PX4

Version: Rover 3.5.0

TSC21, 2019-05-26:
Did you make sure this is actually something Ardupilot os capable of handling and parse?

TiffanyKalin, 2019-05-27:
I thought that the mavros StatusText message translated to the mavlink STATUSTEXT message, which Ardupilot does seem to be able to handle and parse. I'm not sure how to check if the mavros StatusText messages in /mavros/statustext/send is getting rejected by Ardupilot or if the messages in that topic are even getting sent to Ardupilot.

TiffanyKalin, 2019-05-27:
Thank you for any help you can give.

TSC21, 2019-05-27:
Maybe an APM user/developer can help you with that. Sorry

TiffanyKalin, 2019-05-27:
No worries. I do have a mavros-specific question. When I publish something to /mavros/statustext/send, the /mavros node is subscribing to it. What does /mavros do with the information on that topic?

TiffanyKalin, 2019-05-27:
Able to get it forwarding to ardupilot. Any advice on how to forward messages to a GCS?

TSC21, 2019-05-27:
You will have to bring that to the Ardupilot forums. Closing this issue as it's not MAVROS related.

closed 2019-05-27 by TSC21

zkern5254, 2020-01-19:
This is not a Ardupilot issue. When a statustext message is published in MAVRos and a UDP broadcaster like mavproxy is used, the messages make it to the GCS fine.  When MAVRos is used as the UDP server, the messages don’t make it past the FCU. Does anyone have any thoughts on this?

Hallouda, 2020-10-16:
> Able to get it forwarding to ardupilot. Any advice on how to forward messages to a GCS?
I am facing the same issue. Could you please help me solve the problem ?

julled, 2022-02-17:
I am also facing the issue of beeing unable to send a message via MAVROS -> FCU -> GCS. 
Anyone got some hints on how to beeing able to receive them at GCS side ? Is it related to proper SYSTEM_ID & COMPONENT settings in MAVROS?

venabled, 2024-03-07:
Any recent thoughts on this?