mavlink/mavros #1241: catkin_make_isolated failed on Raspi 3 Running xubuntu

NikolaeVarius, 2019-05-26:
This is only bug and feature tracker, please use it
to report bugs or request features.

----

### Issue details

### MAVROS version and platform

Mavros: ?0.18.4?
ROS: ?Kinetic?
Ubuntu: ?16.04?

### Autopilot type and version

[  ] ArduPilot
[  ] PX4

Version: ?3.7.1?

### Node logs

```

`sudo ./src/catkin/bin/catkin_make_isolated --install -DCMAKE_BUILD_TYPE=Release --install-space /opt/ros/kinetic -j2`

returns

`Scanning dependencies of target tf2_msgs_generate_messages_lisp                                                                                                                                     
Scanning dependencies of target tf2_msgs_generate_messages_nodejs                                                                                                                                   
[  0%] Built target tf2_msgs_generate_messages_nodejs                                                                                                                                               
[  0%] Built target tf2_msgs_generate_messages_lisp                                                                                                                                                 
Scanning dependencies of target mavros                                                                                                                                                              
[  2%] Building CXX object CMakeFiles/mavros.dir/src/lib/enum_sensor_orientation.cpp.o                                                                                                              
[  4%] Building CXX object CMakeFiles/mavros.dir/src/lib/enum_to_string.cpp.o                                                                                                                       
[  7%] Building CXX object CMakeFiles/mavros.dir/src/lib/ftf_frame_conversions.cpp.o                                                                                                                
[  9%] Building CXX object CMakeFiles/mavros.dir/src/lib/ftf_quaternion_utils.cpp.o                                                                                                                 
[ 12%] Building CXX object CMakeFiles/mavros.dir/src/lib/mavlink_diag.cpp.o
[ 14%] Building CXX object CMakeFiles/mavros.dir/src/lib/mavros.cpp.o
c++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-5/README.Bugs> for instructions.
CMakeFiles/mavros.dir/build.make:158: recipe for target 'CMakeFiles/mavros.dir/src/lib/mavlink_diag.cpp.o' failed
make[2]: *** [CMakeFiles/mavros.dir/src/lib/mavlink_diag.cpp.o] Error 4
make[2]: *** Waiting for unfinished jobs....
CMakeFiles/Makefile2:2907: recipe for target 'CMakeFiles/mavros.dir/all' failed
make[1]: *** [CMakeFiles/mavros.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
<== Failed to process package 'mavros': 
  Command '['/opt/ros/kinetic/env.sh', 'make', '-j2']' returned non-zero exit status 2

Reproduce this error by running:
==> cd /home/jkim/ros_catkin_ws/build_isolated/mavros && /opt/ros/kinetic/env.sh make -j2
`
```

### Diagnostics
N/A
```
place here result of:
rostopic echo -n1 /diagnostics
```

### Check ID
N/A
```
rosrun mavros checkid
```

Jaeyoung-Lim, 2019-05-30:
Why not use catkin build?

tonybaltovski, 2019-05-30:
Add swap or increase it on the Pi.

vooon, 2019-05-31:
Perhaps i should add tips for rpi to readme, because that is quite regular.

To be able to build ROS on rpi, following things will help you:
1. use external USB drive or SSD, it much faster that SD card
2. add swap file on that drive (1 GiB is too low to build in four parallel jobs), give at least 4 GiB
3. in any problems it might be necessary to limit jobs to 2 or 1 (`-j 2`)
4. `distcc` may drastically speedup compilation

sergioma295, 2019-06-07:
I am having a big problem. I have created a new issue. Can somebody help me please? #1248