mavlink/mavros #1245: What is the initial GPS coordinate (/mavros/global_position/global) on px4 in sitl

elgarbe, 2019-06-01:
Hi, I need to config starting GPS coordinate of a second vehicle on gazebo (taht use gazebo_gps plugin). So I need to know what is the initial GPS position when using px4 in sitl with mavros and gazebo.

Thank!

TSC21, 2019-06-02:
https://dev.px4.io/en/simulation/gazebo.html#set-custom-takeoff-location

closed 2019-06-02 by TSC21

elgarbe, 2019-06-02:
is not what I ask, I would like to know what is the default value, not how to set  new one. I want to leave px4 firmware build from default values.

TSC21, 2019-06-02:
@elgarbe this is not a MAVROS related question, so this why I closed it.

Also, I shared with you the above link since from there you can actually understand where this is set, with a quick search ;) - https://github.com/PX4/sitl_gazebo/blob/master/include/gazebo_gps_plugin.h#L94-L96.

Lastly, please do not open new issues for the same exact question. If this was closed, it was for a reason, and if you were not clarified, just ask the questions here. But again, this is not a MAVROS related question/issue/bug, so this is not the right place to continue this conversation. Thank you.