mavlink/mavros #1247: Using mavros to connect with FCU over switch via TCP

mohand150, 2019-06-06:
### Issue details

I try to connect to a DropixV2 unit with mavros, I use a jetson card in which I installed Ros kinetic, the jetson card and dropix are both connect via a switch
Now, I manage to get the FCU data on my apmplanner or qgroundcontrol that I install on the jetson by specifying the **IP address (192.168.1.130) and the port (4003) with TCP Link**. (I think that the DropixV2 / Jetson TX2 works and is correctly configured) but I have trouble connecting via mavros to fcu.

The udp address I use is:
I launch the Mavros node with the apm.launch file where
	< arg  name = " fcu_url "  default ="udp: //: 14550 @ 192.168.1.130@5760" />
	< arg  name = " gcs_url "  default = " " />
where 192.168.1.9 is the IP address of the pixracer -> it does not work, i try with "udp: //: 14550@192.168 .1.130 @ 4003 ", it does not work as well

### MAVROS version and platform

Mavros: 0,30,0 
ROS: Kinetic
Ubuntu: 16.04

### Autopilot type and version

[ x ] ArduPilot
[  ] PX4

Version: 3.4.2

### Node logs

```

nvidia@tegra-ubuntu:~/catkin_ws$ roslaunch mavros apm.launch
... logging to /home/nvidia/.ros/log/8dcf7ea2-87aa-11e9-9b7a-00044bc4a5de/roslaunch-tegra-ubuntu-5446.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://tegra-ubuntu:37902/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/hrlv_ez4_pub/field_of_view: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/frame_id: hrlv_ez4_sonar
 * /mavros/distance_sensor/hrlv_ez4_pub/id: 0
 * /mavros/distance_sensor/hrlv_ez4_pub/orientation: PITCH_270
 * /mavros/distance_sensor/hrlv_ez4_pub/send_tf: True
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/laser_1_sub/id: 3
 * /mavros/distance_sensor/laser_1_sub/orientation: PITCH_270
 * /mavros/distance_sensor/laser_1_sub/subscriber: True
 * /mavros/distance_sensor/lidarlite_pub/field_of_view: 0.0
 * /mavros/distance_sensor/lidarlite_pub/frame_id: lidarlite_laser
 * /mavros/distance_sensor/lidarlite_pub/id: 1
 * /mavros/distance_sensor/lidarlite_pub/orientation: PITCH_270
 * /mavros/distance_sensor/lidarlite_pub/send_tf: True
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/sonar_1_sub/id: 2
 * /mavros/distance_sensor/sonar_1_sub/orientation: PITCH_270
 * /mavros/distance_sensor/sonar_1_sub/subscriber: True
[... 505 lines cut ...]
auto-starting new master
process[master]: started with pid [23400]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to e9c8318c-8835-11e9-9bef-00044bc4a5de
process[rosout-1]: started with pid [23415]
started core service [/rosout]
process[mavros-2]: started with pid [23435]
[ INFO] [1559810077.748363390]: FCU URL: tcp://5760@192.168.1.130@5760
[ WARN] [1559810077.763313881]: tcp0: resolve error: Host not found (non-authoritative), try again later
[FATAL] [1559810077.764797647]: FCU: DeviceError:tcp: resolve:Bind address resolve failed
================================================================================REQUIRED process [mavros-2] has died!
process has finished cleanly
log file: /home/nvidia/.ros/log/e9c8318c-8835-11e9-9bef-00044bc4a5de/mavros-2*.log
Initiating shutdown!
================================================================================
[mavros-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

```
do I have to specify something else since it is a **TCP connection**?

vooon, 2019-06-07:
You supply incorrect URL. `@` symbol only used by UDP to separate bind (where mavros is listen) and remote.
See [Connection URL list][1].

[1]: https://github.com/mavlink/mavros/blob/master/mavros/README.md#connection-url

mohand150, 2019-06-07:
hello @vooon,  
you're right, it's a stupid mistake 
**what is all these warnings about TM : RTT too high for timesync: ??**
`
nvidia@tegra-ubuntu:~/catkin_ws$ roslaunch mavros apm.launch
... logging to /home/nvidia/.ros/log/6a5ef696-8921-11e9-8a99-00044bc4a5de/roslaunch-tegra-ubuntu-19677.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://tegra-ubuntu:39983/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_mav_type: ONBOARD_CONTROLLER
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 20.0
 * /mavros/distance_sensor/rangefinder_pub/field_of_view: 0.0
 * /mavros/distance_sensor/rangefinder_pub/frame_id: lidar
 * /mavros/distance_sensor/rangefinder_pub/id: 0
 * /mavros/distance_sensor/rangefinder_pub/send_tf: False
 * /mavros/distance_sensor/rangefinder_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/rangefinder_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/rangefinder_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/rangefinder_sub/id: 1
 * /mavros/distance_sensor/rangefinder_sub/orientation: PITCH_270
 * /mavros/distance_sensor/rangefinder_sub/subscriber: True
 * /mavros/fake_gps/eph: 2.0
 * /mavros/fake_gps/epv: 2.0
 * /mavros/fake_gps/fix_type: 3
 * /mavros/fake_gps/geo_origin/alt: 408.0
 * /mavros/fake_gps/geo_origin/lat: 47.3667
 * /mavros/fake_gps/geo_origin/lon: 8.55
 * /mavros/fake_gps/gps_rate: 5.0
 * /mavros/fake_gps/mocap_transform: True
 * /mavros/fake_gps/satellites_visible: 5
 * /mavros/fake_gps/tf/child_frame_id: fix
 * /mavros/fake_gps/tf/frame_id: map
 * /mavros/fake_gps/tf/listen: False
 * /mavros/fake_gps/tf/rate_limit: 10.0
 * /mavros/fake_gps/tf/send: False
 * /mavros/fake_gps/use_mocap: True
 * /mavros/fake_gps/use_vision: False
 * /mavros/fcu_protocol: v2.0
 * /mavros/fcu_url: tcp://192.168.1.1...
 * /mavros/gcs_url:
 * /mavros/global_position/child_frame_id: base_link
 * /mavros/global_position/frame_id: map
 * /mavros/global_position/gps_uere: 1.0
 * /mavros/global_position/rot_covariance: 99999.0
 * /mavros/global_position/tf/child_frame_id: base_link
 * /mavros/global_position/tf/frame_id: map
 * /mavros/global_position/tf/global_frame_id: earth
 * /mavros/global_position/tf/send: False
 * /mavros/global_position/use_relative_alt: True
 * /mavros/image/frame_id: px4flow
 * /mavros/imu/angular_velocity_stdev: 0.000349065850399
 * /mavros/imu/frame_id: base_link
 * /mavros/imu/linear_acceleration_stdev: 0.0003
 * /mavros/imu/magnetic_stdev: 0.0
 * /mavros/imu/orientation_stdev: 1.0
 * /mavros/local_position/frame_id: map
[... 271 lines cut ...]
[ INFO] [1559911245.328788047]: WP: item #37  F:0 C:206 p: 0.000000 0.000000 0.000000 0.000000 x: 0.000000 y: 0.000000 z: 0.000000
[ INFO] [1559911245.329845416]: WP: mission received
[ WARN] [1559911245.406499138]: TM : RTT too high for timesync: 23.23 ms.
[ WARN] [1559911245.709206822]: TM : RTT too high for timesync: 25.25 ms.
[ WARN] [1559911246.010488979]: TM : RTT too high for timesync: 26.55 ms.
[ WARN] [1559911246.305601898]: TM : RTT too high for timesync: 21.89 ms.
[ WARN] [1559911246.605786622]: TM : RTT too high for timesync: 22.34 ms.
[ WARN] [1559911246.905340727]: TM : RTT too high for timesync: 22.24 ms.
[ WARN] [1559911247.207335967]: TM : RTT too high for timesync: 23.49 ms.
[ WARN] [1559911247.507579731]: TM : RTT too high for timesync: 23.78 ms.
[ WARN] [1559911247.805251865]: TM : RTT too high for timesync: 21.43 ms.
[ WARN] [1559911248.106356263]: TM : RTT too high for timesync: 22.03 ms.
[ WARN] [1559911248.404997190]: TM : RTT too high for timesync: 21.80 ms.
[ WARN] [1559911248.705213051]: TM : RTT too high for timesync: 21.95 ms.
[ WARN] [1559911249.005437007]: TM : RTT too high for timesync: 22.05 ms.
[ INFO] [1559911249.294077969]: HP: requesting home position
[ WARN] [1559911249.305839874]: TM : RTT too high for timesync: 22.10 ms.
[ WARN] [1559911249.605534298]: TM : RTT too high for timesync: 21.56 ms.
[ WARN] [1559911249.908565755]: TM : RTT too high for timesync: 24.65 ms.
[ WARN] [1559911250.205012521]: TM : RTT too high for timesync: 21.35 ms.
[ WARN] [1559911250.505862745]: TM : RTT too high for timesync: 21.87 ms.
[ WARN] [1559911250.808295518]: TM : RTT too high for timesync: 24.01 ms.
[ WARN] [1559911251.106136291]: TM : RTT too high for timesync: 22.86 ms.
[ WARN] [1559911251.405945658]: TM : RTT too high for timesync: 21.54 ms.
`

closed 2019-06-07 by mohand150

vooon, 2019-06-07:
@mohand150 Round Trip Time, it means that you have too big deplay on your network.
Perhaps easiest to disable sys_time, or find why your ping so slow.

sergioma295, 2019-06-07:
Thank you very much for your help. I had a problem with the port assigned to the simulator. I finally fixed it.