mavlink/mavros #1248: MAVROS, SITL, GAZEBO NOT WORKING TOPICS

sergioma295, 2019-06-07:
### Issue details

I need some help. I'm doing the offboard tutorial explained in [https://dev.px4.io/en/ros/mavros_offboard.html](url). I have followed all the steps and finally when I launch the node no topic publishes anything. I have installed and deleted the packages 6 times, I have tried to install from source or binary. I have consulted many forums and I can not know what is the problem I have and because my drone does not take off. Please, I need someone's help because I'm desperate. 

I have uploaded all my configuration of both ros, catkin workspace and PX4/Firmware in my github repository because I don't find the problem
[https://github.com/sergioma295/Drone_Ros](url)

I explain the most important steps I have taken below:

For "make px4_sitl gazebo":
`sergio@sergio-ing:~/catkin_ws$ cd 
sergio@sergio-ing:~$ cd src/Firmware/
sergio@sergio-ing:~/src/Firmware$ make px4_sitl gazebo
-- PX4 config file: /home/sergio/src/Firmware/boards/px4/sitl/default.cmake
-- PX4 config: px4_sitl_default
-- PX4 platform: posix
-- PX4 lockstep: enabled
-- PX4 version: v1.9.0-199-gbe02ad3
-- cmake build type: RelWithDebInfo
-- The CXX compiler identification is GNU 5.4.0
-- The C compiler identification is GNU 5.4.0
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/cc
-- Check for working CXX compiler using: Ninja
-- Check for working CXX compiler using: Ninja -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler using: Ninja
-- Check for working C compiler using: Ninja -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Found PythonInterp: /usr/bin/python (found version "2.7.12") 
-- Found PY_jinja2: /usr/local/lib/python2.7/dist-packages/jinja2  
-- PX4 ECL: Very lightweight Estimation & Control Library v1.9.0-rc1-5-g54ac147
-- Configuring done
-- Generating done
-- Build files have been written to: /home/sergio/src/Firmware/build/px4_sitl_default
[20/687] git submodule src/drivers/gps/devices
[24/687] git submodule src/lib/ecl
[25/687] git submodule mavlink/include/mavlink/v2.0
[26/687] git submodule Tools/sitl_gazebo
[234/687] Performing configure step for 'sitl_gazebo'
-- install-prefix: /usr/local
-- The C compiler identification is GNU 5.4.0
-- The CXX compiler identification is GNU 5.4.0
-- Check for working C compiler using: Ninja
-- Check for working C compiler using: Ninja -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler using: Ninja
-- Check for working CXX compiler using: Ninja -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Performing Test COMPILER_SUPPORTS_CXX17
-- Performing Test COMPILER_SUPPORTS_CXX17 - Success
-- Performing Test COMPILER_SUPPORTS_CXX14
-- Performing Test COMPILER_SUPPORTS_CXX14 - Success
-- Performing Test COMPILER_SUPPORTS_CXX11
-- Performing Test COMPILER_SUPPORTS_CXX11 - Success
-- Performing Test COMPILER_SUPPORTS_CXX0X
-- Performing Test COMPILER_SUPPORTS_CXX0X - Success
-- Using C++14 compiler
[... 922 lines cut ...]
        value: "0.000000"
      - 
        key: "Mean RTT (ms)"
        value: "0.000000"
      - 
        key: "Last remote time (s)"
        value: "0.000000000"
      - 
        key: "Estimated time offset (s)"
        value: "0.000000000"
---

```

### Check ID

```
OK. I got messages from 1:1.

---
Received 106714 messages, from 1 addresses
sys:comp   list of messages
  1:1     0, 1, 2, 36, 74, 76, 77, 111, 147, 340, 245, 4, 26

```

hardeman, 2019-06-07:
If you're also running QGroundControl port 14557 can be occupied. Try 14556.

sergioma295, 2019-06-07:
Im not running QGroundControl. But i try your advise right now. Can you tell me how can i change the port?

hardeman, 2019-06-07:
How did you start mavros?

sergioma295, 2019-06-07:
I have started with this:
roslaunch mavros px4.launch

However, Also i have started with this:
roslaunch mavros px4.launch fcu_url:="udp://:14550@localhost:14556"

And i have the same problem...

khancyr, 2019-06-07:
@sergioma295 stop spamming

khancyr, 2019-06-07:
@sergioma295 stop spamming
@sergioma295
@sergioma295
@sergioma295

sergioma295, 2019-06-07:
@khancyr I'm just trying to find someone who's going through the same thing with me. You've already written to me once and I haven't put anything back. Please stop bothering me

sergioma295, 2019-06-07:
@hardeman Do you have any suggestions? Thank you very much for your help

TSC21, 2019-06-07:
@sergioma295 spamming your problem in multiple places is not going to solve your problem faster. Did you try to follow what's in https://dev.px4.io/master/en/simulation/ros_interface.html?

sergioma295, 2019-06-07:
Im so sorry. Yes i have followed what's in https://dev.px4.io/master/en/simulation/ros_interface.html

I have a repository in my github https://github.com/sergioma295/Drone_Ros where there are all my project. 

I can't found the problem..

hardeman, 2019-06-07:
Have you tried running the simulator as sudo?

sudo make px4_sitl gazebo

hardeman, 2019-06-07:
I also found this in your log:
INFO [mavlink] mode: Normal, data rate: 4000000 B/s on udp port 14570 remote port 14550

try:
roslaunch mavros px4.launch fcu_url:="udp://:14550@localhost:14570"

haniehes, 2020-07-28:
@sergioma295  I encountered the same problem with the same procedures, Could you finally solve it? The UAV doesn't fly in Gazebo. Also, I think there should be a connection between the gazebo node and controller. I don't know how to make it and I cannot see this connection in your node graph!

closed 2026-11-16 by github-actions[bot]