mavlink/mavros #1249: Play a tune via MAVCMD

smartexagro, 2019-06-10:
I'm trying to play a tune via rosrun command:
```rosrun mavros mavcmd 258 0 0 0 0 0 0 0```

anybody knows which params i need to send?

----

### MAVROS version and platform

ROS: Melodic
Ubuntu: 16.04

### Autopilot type and version

[  ] ArduPilot
[ x] PX4

Version: 3.7.1

vooon, 2019-06-12:
[MAV_CMD][1] doesn't have such command.

If you want to send [PLAY_TUNE][2], then you have to write a plugin.

[1]: https://mavlink.io/en/messages/common.html#mavlink-commands-mavcmd
[2]: https://mavlink.io/en/messages/common.html#PLAY_TUNE

mortenfyhn, 2021-05-11:
No worries, what you're missing might just be to enable the play_tune plugin. When you launch mavros, do you get a long list of plugins saying plugin loaded / blacklisted / similar? It tells you which plugins are enabled. You can add the `play_tune` plugin to the config file `apm_pluginlists.yaml` which is read by `apm.launch`. Hopefully that solves it.

mortenfyhn, 2021-05-11:
That's odd. Do you have QGroundControl or something similar running? You can use its "mavlink inspector" to check if you're seeing mavlink PLAY_TUNE_V2 message. Beyond that I think the other maintainers might be better suited to figure this out.

mortenfyhn, 2021-05-11:
It might be so that it won't show up in QGC even when it works, when I think about it. The message might go straight to the pixhawk, and the pixhawk might not pass it on to QGC.

AnneWendt, 2021-05-12:
Interesting to know, thank you for investigating!

Is there a (beginner-friendly) way to circumvent this issue?
Can I help with anything and where would be the most effective place to make the change?

mortenfyhn, 2021-05-12:
I can probably extend the plugin to support plain PLAY_TUNE messages. @vooon, what's the policy for API changes? I'm tempted to create a new subscriber, so the plugin reads `~play_tune` for mavros_msgs/PlayTune -> mavlink PLAY_TUNE, and `~play_tune_v2` for mavros_msgs/PlayTuneV2 -> mavlink PLAY_TUNE_V2. But that will break the old API, which uses `~play_tune` for v2 messages.

Alternatively I can add a parameter that determines if it expects v1 or v2 messages on `~play_tune`, and I'll output the matching mavlink message.

vooon, 2021-05-12:
@mortenfyhn since we at 1.0 stage, i'd prefer to keep existing api.
Also user shouldn't care about witch version of message is sent, so i'd prefer to add param plus detection.

Something like that:
```
void connection_cb() {
  int new_ver = 0
  if (nh.getParam("message_version", new_ver) {
    // check 1 vs 2
  } 
  if (new_ver == 0) {
    new_ver = m_uas->is_ardupilotmega() ? 1 : 2;
  }

  message_version = new_ver;
}
```

mortenfyhn, 2021-05-12:
> @mortenfyhn since we at 1.0 stage, i'd prefer to keep existing api.
:+1:

I'll do roughly as you suggest:
* Add a parameter.
* If unset, keep existing behaviour.
* If set to specify v1/v2, use that.
* If unset, and `m_uas->is_ardupilotmega() == true` then use v1.

vooon, 2021-05-12:
Yes, also i think it's ok to use 0 == auto (v1 for APM / v2 for others).

vooon, 2021-05-12:
Ah, also, we should simply convert PlayTuneV2.msg to PLAY_TUNE.
Need to check that format == MML (APM's implementation), then do something like that (pseudocode):
```
PLAY_TUNE pt{};

mavlink::set_string(pt.tune, tune->tune);
mavlink::set_string_z(pt.tune2, tune->tune[pt.tune.size():]);
``

AnneWendt, 2021-08-20:
Hi guys,
I was just wondering if there were any updates on this topic. This is probably not a top priority but I think it would be pretty neat if this worked as intended.
Cheers
Anne

mortenfyhn, 2021-09-02:
Hi! I would very much like to fix this but I cannot prioritize it, at least not for a while. Have you considered trying to implement the change?

abhi2198, 2022-01-11:
So the issue is that ardupilot accepts only PLAY_TUNE and you are publishing PLAY_TUNE_V2, a solution would be to send the message from mavlink as PLAY_TUNE instead of PLAY_TUNE_V2. You can change the callback function in    [https://github.com/mavlink/mavros/blob/ros2/mavros_extras/src/plugins/play_tune.cpp#:~:text=PlayTuneV2%3E%3A%3ASharedPtr%20sub%3B-,void%20callback(const%20mavros_msgs%3A%3Amsg%3A%3APlayTuneV2%3A%3ASharedPtr%20tune),%7D,-%7D%3B](url) as follows,

```
void callback(const mavros_msgs::PlayTuneV2::ConstPtr& tune)
	{
		auto msg = mavlink::common::msg::PLAY_TUNE{};
		m_uas->msg_set_target(msg);
		// msg.format = tune->format;
		mavlink::set_string_z(msg.tune, tune->tune);
		UAS_FCU(m_uas)->send_message_ignore_drop(msg);
	}
```
@AnneWendt

closed 2026-11-16 by github-actions[bot]