mavlink/mavros #1250: Publishing thrust command to mavros

rishab26201, 2019-06-14:
Hi,
I am working on a state space model for a drone and my inputs are the thrusts generated by my four rotors independently (k*u1**2, k*u2**2,...). I was wondering if there is any way I could directly publish these values to mavros. The setpoint_raw/attitude topic has one variable thrust, and so I'm not sure how to use it.

Thanks

amalpavithran, 2019-06-14:
By publishing the thrust do you mean sending separate PWM values to each motor? If so then you might be interested in RC Override topic found [here](http://wiki.ros.org/mavros#mavros.2BAC8-Plugins.rc_io)

rishab26201, 2019-06-14:
Yes, the PWM or better the angular speeds of each rotor blade

amalpavithran, 2019-06-14:
I don't think you will be able to get the angular speeds of each blade but you will be able to see the servo output in the topic `/mavros/rc/out` and the RC input in `/mavros/rc/in`. I think you will be able achieve your goal by publishing to the RC override topic mentioned above that is `/mavros/rc/override`

rishab26201, 2019-06-14:
Thanks, much appreciated. Will look into it

closed 2019-06-16 by rishab26201

xabierolaz, 2020-11-05:
Hi @Bibbidi-Babbidi-Boo , did you manage to directly publish these individual motor values to mavros?

Was it inside mavros/rc/out?

Thanks