mavlink/mavros #1254: I have a big problem! help

Fllying, 2019-06-20:
This is only bug and feature tracker, please use it
to report bugs or request features.

When I did it through the tutorials from officiall web, run the code:roslaunch mavros px4.launch ,the message is like this:

### Issue details

roslaunch mavros px4.launch 
... logging to /home/wswhu/.ros/log/af757752-9338-11e9-bf1c-00e05c68513a/roslaunch-wswhu-24135.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/loader.py:409: YAMLLoadWarning: calling yaml.load() without Loader=... is deprecated, as the default Loader is unsafe. Please read https://msg.pyyaml.org/load for full details.
  data = yaml.load(text)
error loading <rosparam> tag: 
	file /home/wswhu/catkin_ws/src/mavros/mavros/launch/px4_config.yaml contains invalid YAML:
could not determine a constructor for the tag '!degrees'
  in "<string>", line 63, column 27:
      angular_velocity_stdev: !degrees 0.02
                              ^
XML is <rosparam command="load" file="$(arg config_yaml)"/>
The traceback for the exception was written to the log file

### MAVROS version and platform

Mavros: latest version
ROS: Kinetic
Ubuntu: 16.04.06

Fllying, 2019-06-20:
How can fix it, please help!!!

kornee-mainblades, 2019-06-20:
Removing the '!' in "!degrees" worked for me.
Remember to do thing for all the "!degrees"' statements.

Fllying, 2019-06-20:
> Removing the '!' in "!degrees" worked for me.
> Remember to do this for all the "!degrees"' statements.

Thanks

closed 2026-11-16 by github-actions[bot]