mavlink/mavros #1255: Topic global_position/gp_vel not receive data

mohand150, 2019-06-20:
I work on a boat drone,
I am looking to get the speed of the meter in meters/s, preferably the speed coming out of the kelman filter
I am looking to get the Ground X Speed (m/s), Ground Y Speed (m/s) and Ground Z Speed (m/s)
so I search the information fused by FCU and raw GPS data.

----

### Issue details
this data is on message 33  and 63 that I receive :
```
  <message id="33" name="GLOBAL_POSITION_INT">
      <description>The filtered global position (e.g. fused GPS and accelerometers). The position is in GPS-frame (right-handed, Z-up). It
               is designed as scaled integer message since the resolution of float is not sufficient.</description>
      <field type="uint32_t" name="time_boot_ms" units="ms">Timestamp (time since system boot).</field>
      <field type="int32_t" name="lat" units="degE7">Latitude, expressed</field>
      <field type="int32_t" name="lon" units="degE7">Longitude, expressed</field>
      <field type="int32_t" name="alt" units="mm">Altitude (MSL). Note that virtually all GPS modules provide both WGS84 and MSL.</field>
      <field type="int32_t" name="relative_alt" units="mm">Altitude above ground</field>
      <field type="int16_t" name="vx" units="cm/s">Ground X Speed (Latitude, positive north)</field>
      <field type="int16_t" name="vy" units="cm/s">Ground Y Speed (Longitude, positive east)</field>
      <field type="int16_t" name="vz" units="cm/s">Ground Z Speed (Altitude, positive down)</field>
      <field type="uint16_t" name="hdg" units="cdeg">Vehicle heading (yaw angle), 0.0..359.99 degrees. If unknown, set to: UINT16_MAX</field>
    </message>
    <message id="63" name="GLOBAL_POSITION_INT_COV">
      <description>The filtered global position (e.g. fused GPS and accelerometers). The position is in GPS-frame (right-handed, Z-up). It  is designed as scaled integer message since the resolution of float is not sufficient. NOTE: This message is intended for onboard networks / companion computers and higher-bandwidth links and optimized for accuracy and completeness. Please use the GLOBAL_POSITION_INT message for a minimal subset.</description>
      <field type="uint64_t" name="time_usec" units="us">Timestamp (UNIX Epoch time or time since system boot). The receiving end can infer timestamp format (since 1.1.1970 or since system boot) by checking for the magnitude the number.</field>
      <field type="uint8_t" name="estimator_type" enum="MAV_ESTIMATOR_TYPE">Class id of the estimator this estimate originated from.</field>
      <field type="int32_t" name="lat" units="degE7">Latitude</field>
      <field type="int32_t" name="lon" units="degE7">Longitude</field>
      <field type="int32_t" name="alt" units="mm">Altitude in meters above MSL</field>
      <field type="int32_t" name="relative_alt" units="mm">Altitude above ground</field>
      <field type="float" name="vx" units="m/s">Ground X Speed (Latitude)</field>
      <field type="float" name="vy" units="m/s">Ground Y Speed (Longitude)</field>
      <field type="float" name="vz" units="m/s">Ground Z Speed (Altitude)</field>
      <field type="float[36]" name="covariance">Row-major representation of a 6x6 position and velocity 6x6 cross-covariance matrix (states: lat, lon, alt, vx, vy, vz; first six entries are the first ROW, next six entries are the second row, etc.). If unknown, assign NaN value to first element in the array.</field>
    </message>
```
**in the wiki mavros I saw that it exites already a topic
~ global_position/gp_vel (geometry_msgs / TwistStamped)
when I run the roslaunch of my mavros node, I don't find this topic**

### MAVROS version and platform

Mavros: 0.30.0
ROS: Kinetic
Ubuntu: 16.04

### Autopilot type and version

[ x ] ArduPilot
[  ] PX4

Version: 3.4.2

### Node logs

```
nvidia@tegra-ubuntu:~/catkin_ws$ roslaunch mavros apm.launch
WARNING: Package name "stereo_vision_SLAM" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits, underscores, and dashes.
... logging to /home/nvidia/.ros/log/5f72d66e-9351-11e9-9b47-00044bc4a5de/roslaunch-tegra-ubuntu-16350.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

WARNING: Package name "stereo_vision_SLAM" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits, underscores, and dashes.
started roslaunch server http://tegra-ubuntu:43688/

SUMMARY
========

[... 443 lines cut ...]
      -
        key: "Current"
        value: "1.5"
      -
        key: "Remaining"
        value: "98.0"

```

### Check ID

```
nvidia@tegra-ubuntu:~/catkin_ws$ rosrun mavros checkid
OK. I got messages from 1:1.

---
Received 1344 messages, from 1 addresses
sys:comp   list of messages
  1:1     0, 1, 2, 150, 65, 152, 147, 22, 24, 27, 29, 30, 33, 34, 35, 36, 165, 42, 178, 182, 193, 194, 74, 77, 163, 111, 241, 116, 125

```
**the topic  ~ global_position/gp_vel  Is it published by the FCU? or the problem comes from GPS which displays warnings "" No gps fix "" "?
or I have to write the handler for GLOBAL_POSITION_INT_COV?

how can I get the velocity in m / s**

elgarbe, 2020-03-08:
so, what is the topic that publish current boat velocity? I need it too...

vooon, 2020-03-09:
In current plugin code i do not see that topic: https://github.com/mavlink/mavros/blob/master/mavros/src/plugins/global_position.cpp#L77

Probably you may use `global_position/raw/gps_vel` or extract velocity from Odom.

closed 2026-11-16 by github-actions[bot]