mavlink/mavros #1256: I can't used /mavros/setpoint_raw/global

omeralierdemir, 2019-06-21:
This is only bug and feature tracker, please use it
to report bugs or request features.

----

### Issue details

Hi everyone,

I want to send my quadcopter to the latitude, longitude and altitude coordinates I have specified. I tried this command at the terminal for simulasion 
``` 
rostopic pub /mavros/spoint_raw/global mavros_msgs/GlobalPositionTarget "header:
  seq: 0
  stamp: now
  frame_id: 'world'
coordinate_frame: 11
type_mask: 4032
latitude: 47.3977427
longitude: 8.5455936
altitude: 435.384916013
velocity: {x: 0.0, y: 0.0, z: 3.0}" -r 10
   ```
But I get a timeout message when I switch to offboard mode
    ```rosrun mavros mavsys mode -c OFFBOARD    ```  ----->
   ```  
[ WARN] [1561125222.798518914]: CMD: Unexpected command 176, result 1
 ``` 

I don't know what the error is and what to do. Can you help me ?

### MAVROS version and platform

Mavros: ros-kinetic-mavros is already the newest version

ROS: Kinetic
Ubuntu: 16.04

### Autopilot type and version

[  ] ArduPilot
[ x ] PX4

installed from latest qgc
### Node logs

```
copy output of mavros_node. Usually console where you run roslaunch

efl4tun@efl4tun44:~/src/Firmware$ make px4_sitl gazebo_typhoon_h480
[2/6] Performing configure step for 'sitl_gazebo'
-- install-prefix: /usr/local
-- Using C++14 compiler
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   system
--   thread
--   filesystem
--   chrono
--   date_time
--   atomic
-- Boost version: 1.58.0
-- Building klt_feature_tracker without catkin
-- Building OpticalFlow with OpenCV
-- catkin DISABLED
Gazebo version: 7.0
-- Configuring done
-- Generating done
-- Build files have been written to: /home/efl4tun/src/Firmware/build/px4_sitl_default/build_gazebo
[3/6] Performing build step for 'sitl_gazebo'
ninja: no work to do.
[... 553 lines cut ...]
        value: "0.166549"
      - 
        key: "Mean RTT (ms)"
        value: "0.223549"
      - 
        key: "Last remote time (s)"
        value: "467.780000000"
      - 
        key: "Estimated time offset (s)"
        value: "1561125194.968325138"

```

### Check ID

```
efl4tun@efl4tun44:~/catkin_ws/src/dronemap/src/scripts$ rosrun mavros checkid
OK. I got messages from 1:1.

---
Received 7172 messages, from 1 addresses
sys:comp   list of messages
  1:1     0, 1, 2, 4, 265, 140, 141, 147, 24, 132, 26, 30, 31, 32, 33, 36, 74, 331, 76, 340, 87, 230, 105, 111, 241, 242, 245

```

Jaeyoung-Lim, 2019-06-25:
Global position setpoints are not supported by PX4

omeralierdemir, 2019-06-25:
Really ? I want to send my quadcopter to the latitude, longitude and altitude coordinates I have specified. How can I do ? What can I use ?

Jaeyoung-Lim, 2019-06-25:
@omeralierdemir You can send waypoints to the flight controller and put the vehicle in mission mode. Its just offboard position setpoints are not supported in PX4

omeralierdemir, 2019-06-25:
thanks for your help :)

closed 2026-11-16 by github-actions[bot]