mavlink/mavros #1264: Build is failing

Juched, 2019-06-28:
### Issue details

The build with the commit: 881cda6ab8c3f655cefbbe20a592d4d76a66f2dd by vooon and mrivi is failing for all platforms. The problem is a trajectory message that they are using they updated it to use an array that doesn't exist within the mavlink message.

### MAVROS version and platform

Mavros: master branch
ROS: All versions
Ubuntu: 16.04

Jaeyoung-Lim, 2019-06-29:
I can see the same here

TSC21, 2019-06-29:
The main source is https://github.com/mavlink/mavros/commit/19c2c3e0a7092da0f9ed4265fa9d4cb07b230590. @vooon

vooon, 2019-06-30:
Will fix on upcoming mavlink release.
Please note, that Lunar is EOLed, so probably that was last update for that distro.

Jaeyoung-Lim, 2019-07-02:
Is there a ETA for this?

vooon, 2019-07-02:
http://wiki.ros.org/Distributions

vooon, 2019-07-02:
Oh, you are asking about release. Usually i do monthly when day == month.

vooon, 2019-07-06:
- mavlink kinetic https://github.com/ros/rosdistro/pull/21706
- mavlink melodic https://github.com/ros/rosdistro/pull/21707
- mavros kinetic https://github.com/ros/rosdistro/pull/21708
- mavros melodic https://github.com/ros/rosdistro/pull/21709

Release for Lunar blocked because of EOL.

closed 2026-11-16 by github-actions[bot]