mavlink/mavros #1267: Cannot takeoff in stabilized mode

rishab26201, 2019-07-05:
I am not able to takeoff in stabilized mode. I get
```
Taking-off from current coord: Lat: -35.3633684 Long: 149.1652296
Request failed. Check mavros logs. ACK: 4
```
 However I can takeoff in guided mode 

Here is the mavros logs
```
... logging to /home/rishab/.ros/log/fc9a55e2-9f3c-11e9-bef0-1657fc255eba/roslaunch-rishab-HP-Notebook-11417.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://rishab-HP-Notebook:39537/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_mav_type: ONBOARD_CONTROLLER
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/rangefinder_pub/field_of_view: 0.0
 * /mavros/distance_sensor/rangefinder_pub/frame_id: lidar
 * /mavros/distance_sensor/rangefinder_pub/id: 0
 * /mavros/distance_sensor/rangefinder_pub/send_tf: False
 * /mavros/distance_sensor/rangefinder_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/rangefinder_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/rangefinder_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/rangefinder_sub/id: 1
 * /mavros/distance_sensor/rangefinder_sub/orientation: PITCH_270
 * /mavros/distance_sensor/rangefinder_sub/subscriber: True
 * /mavros/fake_gps/eph: 2.0
 * /mavros/fake_gps/epv: 2.0
 * /mavros/fake_gps/fix_type: 3
 * /mavros/fake_gps/geo_origin/alt: 408.0
 * /mavros/fake_gps/geo_origin/lat: 47.3667
 * /mavros/fake_gps/geo_origin/lon: 8.55
 * /mavros/fake_gps/gps_rate: 5.0
 * /mavros/fake_gps/mocap_transform: True
 * /mavros/fake_gps/satellites_visible: 5
 * /mavros/fake_gps/tf/child_frame_id: fix
 * /mavros/fake_gps/tf/frame_id: map
 * /mavros/fake_gps/tf/listen: False
 * /mavros/fake_gps/tf/rate_limit: 10.0
 * /mavros/fake_gps/tf/send: False
 * /mavros/fake_gps/use_mocap: True
 * /mavros/fake_gps/use_vision: False
 * /mavros/fcu_protocol: v2.0
 * /mavros/fcu_url: udp://:14550@
 * /mavros/gcs_url: 
 * /mavros/global_position/child_frame_id: base_link
 * /mavros/global_position/frame_id: map
 * /mavros/global_position/gps_uere: 1.0
 * /mavros/global_position/rot_covariance: 99999.0
 * /mavros/global_position/tf/child_frame_id: base_link
 * /mavros/global_position/tf/frame_id: map
 * /mavros/global_position/tf/global_frame_id: earth
 * /mavros/global_position/tf/send: False
 * /mavros/global_position/use_relative_alt: True
 * /mavros/image/frame_id: px4flow
 * /mavros/imu/angular_velocity_stdev: 0.000349065850399
 * /mavros/imu/frame_id: base_link
[... 208 lines cut ...]
[ WARN] [1562346900.054955339, 4667.392500000]: TM : RTT too high for timesync: 3082.50 ms.
[ INFO] [1562346900.061033934, 4667.410000000]: FCU: ArduCopter V3.7.0-dev (f21e4d83)
[ INFO] [1562346900.061871787, 4667.410000000]: FCU: 21be32fa82fa4f5796098745c848d41a
[ INFO] [1562346900.063218928, 4667.412500000]: FCU: Frame: QUAD
[ WARN] [1562346900.492171123, 4667.770000000]: TM : RTT too high for timesync: 2860.00 ms.
[ INFO] [1562346901.158257989, 4668.327500000]: PR: parameters list received
[ WARN] [1562346901.305770079, 4668.402500000]: TM : RTT too high for timesync: 2890.00 ms.
[ INFO] [1562346901.308924900, 4668.405000000]: FCU: ArduCopter V3.7.0-dev (f21e4d83)
[ INFO] [1562346901.309364355, 4668.407500000]: FCU: 21be32fa82fa4f5796098745c848d41a
[ INFO] [1562346901.309837482, 4668.407500000]: FCU: Frame: QUAD
[ WARN] [1562346901.788358040, 4668.807500000]: WP: timeout, retries left 2
[ WARN] [1562346902.321699242, 4669.207500000]: TM : RTT too high for timesync: 3100.00 ms.
[ INFO] [1562346902.444675233, 4669.277500000]: FCU: Arming motors
[ WARN] [1562346902.465797185, 4669.290000000]: TM : RTT too high for timesync: 2580.00 ms.
[ WARN] [1562346902.472907023, 4669.302500000]: TM : RTT too high for timesync: 1992.50 ms.
[ INFO] [1562346902.480115342, 4669.310000000]: WP: mission received
[ WARN] [1562346902.484070006, 4669.310000000]: TM : RTT too high for timesync: 1402.50 ms.
[ WARN] [1562346902.510198580, 4669.330000000]: TM : RTT too high for timesync: 822.50 ms.
[ INFO] [1562346902.521064952, 4669.330000000]: WP: seems GCS requesting mission
[ INFO] [1562346902.521565030, 4669.330000000]: WP: scheduling pull after GCS is done
[ WARN] [1562346902.523824443, 4669.335000000]: TM : RTT too high for timesync: 227.50 ms.
[ INFO] [1562346907.409932951, 4674.337500000]: WP: mission received
[ INFO] [1562346912.315861469, 4679.335000000]: FCU: Disarming motors

```

TSC21, 2019-07-05:
Those are not the logs we are looking for. And please post here what's the output of the autpilot.

Though, I am sure that this is not a problem related with MAVROS but actually with the FCU - so I would bring the issue to the Ardupilot forums.

khancyr, 2019-07-05:
Hello,

That is the correct behavior. You cannot use takeoff function on manual modes like stabilize.

rishab26201, 2019-07-05:
So I have to takeoff in guided mode and then switch to stabilize?

rishab26201, 2019-07-05:
But if I switch to stabilize after takeoff the drone just falls to the ground

TSC21, 2019-07-05:
> But if I switch to stabilize after takeoff the drone just falls to the ground

Why do you want to takeoff and change to stabilize in the first place? Auto Takeoff is by definition an autonomous (or aided) behavior - means that you expect the vehicle to take control, not you.

rishab26201, 2019-07-05:
No, what I want to do is to take off and then make the drone track and land on a target. However, the acceleration and angular velocities (pitch and roll) of the drone are high and this produces error in my visual servoing estimates. To change these parameters, I wish to use stabilized mode

rishab26201, 2019-07-05:
The problem is that I am not able to takeoff in this mode without using auto takeoff

TSC21, 2019-07-05:
> No, what I want to do is to take off and then make the drone track and land on a target. However, the acceleration and angular velocities (pitch and roll) of the drone are high and this produces error in my visual servoing estimates. To change these parameters, I wish to use stabilized mode

Why don't you just use position mode then? Do you have valid GPS or a vision system that provides a position estimate of the vehicle?

rishab26201, 2019-07-05:
Yes I do. So if I were to use Position Mode, I would be able to provide as input the normalized thrust and desired pitch and roll and yaw rate (to /mavros/setpoint_raw/attitude) and can use that to control my drone. Am I correct? And also how would I change my acceleration and ang. velocity parameters for this (for simulation in Gazebo)

TSC21, 2019-07-05:
> Yes I do. So if I were to use Position Mode, I would be able to provide as input the normalized thrust and desired pitch and roll and yaw rate (to /mavros/setpoint_raw/attitude) and can use that to [...]

This is getting out of the spec of this repository. I suggest you move this dicussion to the Ardupilot forums/Gitter. I am closing this since this is no longer an issue related with MAVROS.

closed 2019-07-05 by TSC21