mavlink/mavros #1268: OverrideRCIN is slow

fazlurnu, 2019-07-07:
Hi Guys,

I'm using Pixhawk PX4 as my flight controller, I use OverrideRCIn to replace the command from my remote control. It works but somehow the rate is so slow, the update is around 200 milliseconds.

Do you have any idea how to make it faster? I need to set the update period to be 20 milliseconds

Here's my code:
https://github.com/fazlurnu/ROS/blob/SimpleControl/control.py

This is only bug and feature tracker, please use it
to report bugs or request features.

----

### Issue details

_Please describe the problem, or desired feature_

### MAVROS version and platform

Mavros: ?0.18.4?
ROS: ?Kinetic?
Ubuntu: ?16.04?

### Autopilot type and version

[  ] ArduPilot
[  ] PX4

Version: ?3.7.1?

### Node logs

```
copy output of mavros_node. Usually console where you run roslaunch
```

### Diagnostics

```
place here result of:
rostopic echo -n1 /diagnostics
```

### Check ID

```
rosrun mavros checkid
```

khancyr, 2019-07-07:
Hello,

Put a faster update rate in your control loop!

fazlurnu, 2019-07-07:
I tried to update the rate but it doesn't work. Seems like it has something to do with the algorithm architecture. The way the callback function is called.

vooon, 2019-07-08:
Please add diagnostics as described in issue template.
Then:
1. where mavros is runninng?
2. how fast link between mavros host and FCU?
3. what frequency you setup in your loop?

fazlurnu, 2019-07-13:
I found out the problem. The callback function is called only when the imu/data message arrives. The frequency of /mavros/imu/data topic is around 4 Hz that's why I got this problem. I need to increase its frequency

closed 2019-07-13 by fazlurnu