mavlink/mavros #1272: Comunication problem pixhawk, arducopter, mavros 0.32

elgarbe, 2019-07-14:
### Issue details

I have a pixhawk 1 connected through an ftdi adapter (chines clone) to an odroid xu4 runing ubuntu 16.04 and mavros 0.32 from binary. I've tried with 115200, 921600 and the problem is always the same, I have heartbeat but don't receive several ros topics (/mavros/imu/data, /mavros/battery....) after some time I have CON Lost message. I've tested with a desktop PC too and the same problem. I add some details info

### MAVROS version and platform

Mavros: 0.32.0-1xenial-20190709-034638+0000 500
ROS: 1.12.14
Ubuntu: 16.04

### Autopilot type and version

[ X] ArduPilot
[  ] PX4

Version: 3.5.7

### Node logs

```
~$ roslaunch mavros apm.launch fcu_url:=/dev/ttyUSB0:115200
... logging to /home/elgarbe/.ros/log/700e820c-a648-11e9-ae86-7c67a237b388/roslaunch-notebook-13975.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://notebook:37139/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_mav_type: ONBOARD_CONTROLLER
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/rangefinder_pub/field_of_view: 0.0
 * /mavros/distance_sensor/rangefinder_pub/frame_id: lidar
 * /mavros/distance_sensor/rangefinder_pub/id: 0
 * /mavros/distance_sensor/rangefinder_pub/send_tf: False
 * /mavros/distance_sensor/rangefinder_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/rangefinder_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/rangefinder_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/rangefinder_sub/id: 1
 * /mavros/distance_sensor/rangefinder_sub/orientation: PITCH_270
 * /mavros/distance_sensor/rangefinder_sub/subscriber: True
 * /mavros/fake_gps/eph: 2.0
 * /mavros/fake_gps/epv: 2.0
 * /mavros/fake_gps/fix_type: 3
 * /mavros/fake_gps/geo_origin/alt: 408.0
 * /mavros/fake_gps/geo_origin/lat: 47.3667
 * /mavros/fake_gps/geo_origin/lon: 8.55
 * /mavros/fake_gps/gps_rate: 5.0
 * /mavros/fake_gps/mocap_transform: True
 * /mavros/fake_gps/satellites_visible: 5
 * /mavros/fake_gps/tf/child_frame_id: fix
 * /mavros/fake_gps/tf/frame_id: map
 * /mavros/fake_gps/tf/listen: False
 * /mavros/fake_gps/tf/rate_limit: 10.0
 * /mavros/fake_gps/tf/send: False
 * /mavros/fake_gps/use_mocap: True
 * /mavros/fake_gps/use_vision: False
 * /mavros/fcu_protocol: v2.0
 * /mavros/fcu_url: /dev/ttyUSB0:115200
 * /mavros/gcs_url: 
[... 479 lines cut ...]
OK. I got messages from 1:1.

---
Received 166 messages, from 1 addresses
sys:comp   list of messages
  1:1     0, 77, 111

```
/mavros/state:

---
header: 
  seq: 112
  stamp: 
    secs: 1563116678
    nsecs: 677276851
  frame_id: ''
connected: True
armed: False
guided: True
mode: "RTL"
system_status: 3
---

``

elgarbe, 2019-07-14:
I've disabled timesync_rate and now I get this on mavros_node:

`
[ INFO] [1563116991.303955198]: MAVROS started. MY ID 1.240, TARGET ID 1.1
[ INFO] [1563116991.675892743]: CON: Got HEARTBEAT, connected. FCU: ArduPilot
[ INFO] [1563116992.694977679]: VER: 1.1: Capabilities         0x0000000000001bcf
[ INFO] [1563116992.695099206]: VER: 1.1: Flight software:     030507ff (b11c6af3b535f9741bcae90b)
[ INFO] [1563116992.695188703]: VER: 1.1: Middleware software: 00000000 (b535f9741bcae90b)
[ INFO] [1563116992.695287656]: VER: 1.1: OS software:         00000000 (1bcae90b)
[ INFO] [1563116992.695377927]: VER: 1.1: Board hardware:      00000000
[ INFO] [1563116992.695455686]: VER: 1.1: VID/PID:             0000:0000
[ INFO] [1563116992.695534395]: VER: 1.1: UID:                 0000000000000000
[ WARN] [1563116992.695883199]: CMD: Unexpected command 520, result 0
[ INFO] [1563117001.676468153]: HP: requesting home position
[ERROR] [1563117001.678061981]: FCU: PreArm: Check mag field
[ERROR] [1563117001.682867300]: FCU: PreArm: GPS speed error 1.8 (needs 1.0)
[ERROR] [1563117001.687861799]: FCU: PreArm: Throttle below Failsafe
[ INFO] [1563117001.695083118]: FCU: APM:Copter V3.5.7 (b11c6af3)
[ INFO] [1563117001.699828328]: FCU: PX4: b535f974 NuttX: 1bcae90b
[ INFO] [1563117001.704751007]: FCU: Frame: HEXA
[ INFO] [1563117001.710773202]: FCU: PX4v2 00540035 3436510A 32393637
[ INFO] [1563117006.703650981]: WP: item #0  F:0 C: 16 p: 0.000000 0.000000 0.000000 0.000000 x: 0.000000 y: 0.000000 z: 0.000000
[ INFO] [1563117006.712472503]: WP: item #1  F:0 C:178 p: 1.000000 2.000000 -1.000000 0.000000 x: 0.000000 y: 0.000000 z: 0.000000
[ INFO] [1563117006.722397902]: WP: item #2  F:3 C: 16 p: 0.000000 0.000000 0.000000 0.000000 x: -33.507919 y: -60.063168 z: 60.000000
[ INFO] [1563117006.733367488]: WP: item #3  F:3 C: 16 p: 0.000000 0.000000 0.000000 0.000000 x: -33.508533 y: -60.062431 z: 60.000000
[ INFO] [1563117006.743375769]: WP: item #4  F:3 C: 16 p: 0.000000 0.000000 0.000000 0.000000 x: -33.508427 y: -60.062332 z: 60.000000
[ INFO] [1563117006.753318749]: WP: item #5  F:3 C: 16 p: 0.000000 0.000000 0.000000 0.000000 x: -33.507816 y: -60.063061 z: 60.000000
[ INFO] [1563117006.764309769]: WP: item #6  F:3 C: 16 p: 0.000000 0.000000 0.000000 0.000000 x: -33.507713 y: -60.062958 z: 60.000000
[ INFO] [1563117006.776193945]: WP: item #7  F:3 C: 16 p: 0.000000 0.000000 0.000000 0.000000 x: -33.508327 y: -60.062222 z: 60.000000
[ INFO] [1563117006.795023919]: WP: item #8  F:3 C: 16 p: 0.000000 0.000000 0.000000 0.000000 x: -33.508224 y: -60.062111 z: 60.000000
[ INFO] [1563117006.805968430]: WP: item #9  F:3 C: 16 p: 0.000000 0.000000 0.000000 0.000000 x: -33.507614 y: -60.062847 z: 60.000000
[ INFO] [1563117006.815931857]: WP: item #10  F:3 C: 16 p: 0.000000 0.000000 0.000000 0.000000 x: -33.507511 y: -60.062740 z: 60.000000
[ INFO] [1563117006.825940823]: WP: item #11  F:3 C: 16 p: 0.000000 0.000000 0.000000 0.000000 x: -33.508125 y: -60.062004 z: 60.000000
[ INFO] [1563117006.835853496]: WP: item #12  F:3 C: 16 p: 0.000000 0.000000 0.000000 0.000000 x: -33.508022 y: -60.061893 z: 60.000000
[ INFO] [1563117006.845757790]: WP: item #13  F:3 C: 16 p: 0.000000 0.000000 0.000000 0.000000 x: -33.507408 y: -60.062630 z: 60.000000
[ INFO] [1563117006.856794825]: WP: item #14  F:3 C: 16 p: 0.000000 0.000000 0.000000 0.000000 x: -33.507309 y: -60.062531 z: 60.000000
[ INFO] [1563117006.867644987]: WP: item #15  F:3 C: 16 p: 0.000000 0.000000 0.000000 0.000000 x: -33.507923 y: -60.061794 z: 60.000000
[ INFO] [1563117006.877608838]: WP: item #16  F:3 C: 16 p: 0.000000 0.000000 0.000000 0.000000 x: -33.507820 y: -60.061684 z: 60.000000
[ INFO] [1563117006.895579146]: WP: item #17  F:3 C: 16 p: 0.000000 0.000000 0.000000 0.000000 x: -33.507526 y: -60.062035 z: 60.000000
[ INFO] [1563117006.905479274]: WP: item #18  F:3 C: 16 p: 0.000000 0.000000 0.000000 0.000000 x: -33.508018 y: -60.062504 z: 20.000000
[ INFO] [1563117006.915454068]: WP: item #19  F:0 C:178 p: 1.000000 5.000000 -1.000000 0.000000 x: 0.000000 y: 0.000000 z: 0.000000
[ INFO] [1563117006.915680353]: WP: mission received
[ INFO] [1563117011.676762071]: HP: requesting home position
[ INFO] [1563117018.284208018]: PR: parameters list received
[ INFO] [1563117021.676955221]: HP: requesting home position
[ INFO] [1563117031.676612257]: HP: requesting home position
[ERROR] [1563117031.678624269]: FCU: PreArm: Check mag field
[ERROR] [1563117031.684627384]: FCU: PreArm: GPS speed error 1.8 (needs 1.0)
[ERROR] [1563117031.689554496]: FCU: PreArm: Throttle below Failsafe
[ WARN] [1563117037.678868657]: FCU: Flight mode change failed
[ INFO] [1563117041.676345485]: HP: requesting home position
[ INFO] [1563117051.677084537]: HP: requesting home position
[ INFO] [1563117061.676795446]: HP: requesting home position
[ERROR] [1563117061.680308349]: FCU: PreArm: Check mag field
[ INFO] [1563117071.676977080]: HP: requesting home position
[ INFO] [1563117081.676975490]: HP: requesting home position
[ INFO] [1563117091.676403993]: HP: requesting home position
[ERROR] [1563117091.677067297]: FCU: PreArm: Check mag field
[ INFO] [1563117101.676745227]: HP: requesting home position
[ INFO] [1563117111.676954279]: HP: requesting home position
[ERROR] [1563117121.676602998]: FCU: PreArm: Check mag field
[ INFO] [1563117121.676858938]: HP: requesting home position
`

and the diagnostic:

`
~$ rtopic echo -n1 /diagnostics
header: 
  seq: 153
[... 113 lines cut ...]
      - 
        key: "Errors count #3"
        value: "0"
      - 
        key: "Errors count #4"
        value: "0"
  - 
    level: 2
    name: "mavros: Battery"
    message: "No data"
    hardware_id: "/dev/ttyUSB0:115200"
    values: 
      - 
        key: "Voltage"
        value: "-1.00"
      - 
        key: "Current"
        value: "0.0"
      - 
        key: "Remaining"
        value: "0.0"
---
`

I don't receive any data on /mavros/imu/data /mavros/battery, etc...

khancyr, 2019-07-14:
Hello,

Try to activate the stream with rosservice call /mavros/set_stream_rate 0 10 1

elgarbe, 2019-07-14:
> Hello,
> 
> Try to activate the stream with rosservice call /mavros/set_stream_rate 0 10 1

WTF!!! 
That solve my problem!! 
Do I miss something in the installation/configuration steps? why do I need to run that command? 
anyway, thank for the help!

closed 2026-11-15 by github-actions[bot]