mavlink/mavros #1273: CON: Lost connection, HEARTBEAT timed out.

elgarbe, 2019-07-14:
### Issue details

I'm trying to connect pixhawk with odroid xu4 through an ftdi cable (telem2 to USB) I've tested this with a desktop pc and it is working ok. When I connect to odroid I get CON: Lost connection after a couple of minutes. 

### MAVROS version and platform

Mavros: 0.32.0
ROS: Kinetic
Ubuntu: 16.04

### Autopilot type and version

[ X] ArduPilot
[  ] PX4

Version: 3.5.7

### Node logs

```
odroid@odroid:~$ roslaunch mavros apm.launch fcu_url:=/dev/ttyUSB0:921600
... logging to /home/odroid/.ros/log/85ae4be6-a658-11e9-ac38-7cdd90b11a1e/roslaunch-odroid-2839.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://odroid:34565/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_mav_type: ONBOARD_CONTROLLER
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 20.0
 * /mavros/conn/timesync_rate: 0.0
 * /mavros/distance_sensor/rangefinder_pub/field_of_view: 0.0
 * /mavros/distance_sensor/rangefinder_pub/frame_id: lidar
 * /mavros/distance_sensor/rangefinder_pub/id: 0
 * /mavros/distance_sensor/rangefinder_pub/send_tf: False
 * /mavros/distance_sensor/rangefinder_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/rangefinder_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/rangefinder_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/rangefinder_sub/id: 1
 * /mavros/distance_sensor/rangefinder_sub/orientation: PITCH_270
 * /mavros/distance_sensor/rangefinder_sub/subscriber: True
 * /mavros/fake_gps/eph: 2.0
 * /mavros/fake_gps/epv: 2.0
 * /mavros/fake_gps/fix_type: 3
 * /mavros/fake_gps/geo_origin/alt: 408.0
 * /mavros/fake_gps/geo_origin/lat: 47.3667
 * /mavros/fake_gps/geo_origin/lon: 8.55
 * /mavros/fake_gps/gps_rate: 5.0
 * /mavros/fake_gps/mocap_transform: True
 * /mavros/fake_gps/satellites_visible: 5
 * /mavros/fake_gps/tf/child_frame_id: fix
 * /mavros/fake_gps/tf/frame_id: map
 * /mavros/fake_gps/tf/listen: False
 * /mavros/fake_gps/tf/rate_limit: 10.0
 * /mavros/fake_gps/tf/send: False
 * /mavros/fake_gps/use_mocap: True
 * /mavros/fake_gps/use_vision: False
 * /mavros/fcu_protocol: v2.0
 * /mavros/fcu_url: /dev/ttyUSB0:921600
 * /mavros/gcs_url: 
[... 598 lines cut ...]
      - 
        key: "Current"
        value: "0.0"
      - 
        key: "Remaining"
        value: "99.0"
---
odroid@odroid:~$ 

```

### Check ID

```
odroid@odroid:~$ rosrun mavros checkid
OK. I got messages from 1:1.

---
Received 1118 messages, from 1 addresses
sys:comp   list of messages
  1:1     0, 1, 2, 150, 193, 136, 152, 147, 22, 24, 27, 29, 30, 33, 35, 36, 165, 42, 178, 182, 62, 65, 74, 77, 163, 241, 116, 125

```

is raspberry pi 3 a better companion computer than an odroid xu4? I need to run some video processing node so I thought that odroid will be better, but I'm stuck with this problem for last week...

elgarbe, 2019-07-14:
there is some differences from mavros_node in my desktop pc and on the odroid:

PC:
```
[ INFO] [1563124284.971818423]: Known MAVLink dialects: common ardupilotmega ASLUAV autoquad icarous matrixpilot paparazzi slugs standard uAvionix ualberta
[ INFO] [1563124284.971840154]: MAVROS started. MY ID 1.240, TARGET ID 1.1
[ INFO] [1563124284.996930605]: IMU: Raw IMU message used.
[ WARN] [1563124284.996999181]: IMU: linear acceleration on RAW_IMU known on APM only.
[ WARN] [1563124284.997035977]: IMU: ~imu/data_raw stores unscaled raw acceleration report.
[ INFO] [1563124285.002833453]: RC_CHANNELS message detected!
[ INFO] [1563124285.519159688]: CON: Got HEARTBEAT, connected. FCU: ArduPilot
[ INFO] [1563124285.610534251]: RC_CHANNELS message detected!
[ INFO] [1563124286.532676371]: VER: 1.1: Capabilities         0x0000000000001bcf
[ INFO] [1563124286.532786795]: VER: 1.1: Flight software:     030507ff (b11c6af3b535f9741bcae90b)
[ INFO] [1563124286.532911062]: VER: 1.1: Middleware software: 00000000 (b535f9741bcae90b)
[ INFO] [1563124286.533031355]: VER: 1.1: OS software:         00000000 (1bcae90b)
[ INFO] [1563124286.533139581]: VER: 1.1: Board hardware:      00000000
[ INFO] [1563124286.533247361]: VER: 1.1: VID/PID:             0000:0000
[ INFO] [1563124286.533371792]: VER: 1.1: UID:                 0000000000000000
[ WARN] [1563124286.533518613]: CMD: Unexpected command 520, result 0
[ INFO] [1563124295.520311544]: HP: requesting home position
[ INFO] [1563124295.524672160]: FCU: APM:Copter V3.5.7 (b11c6af3)
[ INFO] [1563124295.525490641]: FCU: PX4: b535f974 NuttX: 1bcae90b
[ INFO] [1563124295.526484478]: FCU: Frame: HEXA
[ INFO] [1563124295.526604240]: FCU: PX4v2 00540035 3436510A 32393637
[ INFO] [1563124300.530056030]: WP: item #0  F:0 C: 16 p: 0.000000 0.000000 0.000000 0.000000 x: 0.000000 y: 0.000000 z: 0.000000
[ INFO] [1563124300.534880946]: WP: item #1  F:0 C:178 p: 1.000000 2.000000 -1.000000 0.000000 x: 0.000000 y: 0.000000 z: 0.000000
[ INFO] [1563124300.540135822]: WP: item #2  F:3 C: 16 p: 0.000000 0.000000 0.000000 0.000000 x: -33.507919 y: -60.063168 z: 60.000000
[ INFO] [1563124300.544949604]: WP: item #3  F:3 C: 16 p: 0.000000 0.000000 0.000000 0.000000 x: -33.508533 y: -60.062431 z: 60.000000
[ INFO] [1563124300.549869548]: WP: item #4  F:3 C: 16 p: 0.000000 0.000000 0.000000 0.000000 x: -33.508427 y: -60.062332 z: 60.000000
[ INFO] [1563124300.554811519]: WP: item #5  F:3 C: 16 p: 0.000000 0.000000 0.000000 0.000000 x: -33.507816 y: -60.063061 z: 60.000000
[ INFO] [1563124300.559910611]: WP: item #6  F:3 C: 16 p: 0.000000 0.000000 0.000000 0.000000 x: -33.507713 y: -60.062958 z: 60.000000
[ INFO] [1563124300.564826786]: WP: item #7  F:3 C: 16 p: 0.000000 0.000000 0.000000 0.000000 x: -33.508327 y: -60.062222 z: 60.000000
[ INFO] [1563124300.569780193]: WP: item #8  F:3 C: 16 p: 0.000000 0.000000 0.000000 0.000000 x: -33.508224 y: -60.062111 z: 60.000000
[ INFO] [1563124300.574776715]: WP: item #9  F:3 C: 16 p: 0.000000 0.000000 0.000000 0.000000 x: -33.507614 y: -60.062847 z: 60.000000
[ INFO] [1563124300.579698742]: WP: item #10  F:3 C: 16 p: 0.000000 0.000000 0.000000 0.000000 x: -33.507511 y: -60.062740 z: 60.000000
[ INFO] [1563124300.584686277]: WP: item #11  F:3 C: 16 p: 0.000000 0.000000 0.000000 0.000000 x: -33.508125 y: -60.062004 z: 60.000000
[ INFO] [1563124300.589687268]: WP: item #12  F:3 C: 16 p: 0.000000 0.000000 0.000000 0.000000 x: -33.508022 y: -60.061893 z: 60.000000
[ INFO] [1563124300.594633747]: WP: item #13  F:3 C: 16 p: 0.000000 0.000000 0.000000 0.000000 x: -33.507408 y: -60.062630 z: 60.000000
[ INFO] [1563124300.599566313]: WP: item #14  F:3 C: 16 p: 0.000000 0.000000 0.000000 0.000000 x: -33.507309 y: -60.062531 z: 60.000000
[ INFO] [1563124300.604745354]: WP: item #15  F:3 C: 16 p: 0.000000 0.000000 0.000000 0.000000 x: -33.507923 y: -60.061794 z: 60.000000
[ INFO] [1563124300.609804877]: WP: item #16  F:3 C: 16 p: 0.000000 0.000000 0.000000 0.000000 x: -33.507820 y: -60.061684 z: 60.000000
[ INFO] [1563124300.614616655]: WP: item #17  F:3 C: 16 p: 0.000000 0.000000 0.000000 0.000000 x: -33.507526 y: -60.062035 z: 60.000000
[ INFO] [1563124300.619535352]: WP: item #18  F:3 C: 16 p: 0.000000 0.000000 0.000000 0.000000 x: -33.508018 y: -60.062504 z: 20.000000
[ INFO] [1563124300.624486658]: WP: item #19  F:0 C:178 p: 1.000000 5.000000 -1.000000 0.000000 x: 0.000000 y: 0.000000 z: 0.000000
[ INFO] [1563124300.624768545]: WP: mission received
[ INFO] [1563124305.520216898]: HP: requesting home position
[ INFO] [1563124312.195706488]: PR: parameters list received
[ INFO] [1563124315.520140750]: HP: requesting home position
```

Odroid

```
[ INFO] [1563123477.372029634]: Known MAVLink dialects: common ardupilotmega ASLUAV autoquad icarous matrixpilot paparazzi slugs standard uAvionix ualberta
[ INFO] [1563123477.372124465]: MAVROS started. MY ID 1.240, TARGET ID 1.1
[ INFO] [1563123477.850184936]: CON: Got HEARTBEAT, connected. FCU: ArduPilot
[ INFO] [1563123478.858505114]: VER: 1.1: Capabilities         0x0000000000001bcf
[ INFO] [1563123478.858638569]: VER: 1.1: Flight software:     030507ff (b11c6af3b535f9741bcae90b)
[ INFO] [1563123478.858778066]: VER: 1.1: Middleware software: 00000000 (b535f9741bcae90b)
[ INFO] [1563123478.858900812]: VER: 1.1: OS software:         00000000 (1bcae90b)
[ INFO] [1563123478.859029892]: VER: 1.1: Board hardware:      00000000
[ INFO] [1563123478.859206763]: VER: 1.1: VID/PID:             0000:0000
[... cut ...]
[ INFO] [1563123497.850929061]: HP: requesting home position
[ WARN] [1563123499.970782449]: PR: Param STAT_FLTTIME (65535/774): <value><i4>9001</i4></value> different index: 602/774
[ WARN] [1563123499.971684804]: PR: Param STAT_RUNTIME (65535/774): <value><i4>90043</i4></value> different index: 603/774
[ WARN] [1563123505.470396228]: PR: request param #55 timeout, retries left 2, and 6 params still missing
[ INFO] [1563123505.829522989]: PR: parameters list received
```

could it be related to those STAT_FLTTIME param warning?

elgarbe, 2019-07-14:
I'm testing with a raspberry pi 3 and, using usb connection (telem2 -> ftdi -> USB raspberry) got the same problem, CON Lost.
Then I've tried with direct telem2 -> uart raspberry and got no problem.
So, sould it be ftdi dongle (china clone) but with the dongle and a desktp pc works ok, or maybe odroid/raspberry USB capacity? 
has someone tried pixhawk and odroid conection without problem?

vooon, 2019-07-15:
Let's do following tests:
- try to connect to pixhawk USB port directly and run mavros (port = /dev/ttyACM0)
- check wiring (especially GND)
- try to use another USB-UART bridge, for example pl2302

elgarbe, 2019-07-16:
Hi, I've connected odroid uart (from gpio header) through a transistor level shifter to pixhawk telem 2 and the problem was solved. I can't run at 921600bps but it is working at 115200. I'm buying a txb0102 level shifter to get more speed.
I will try direct USB connection and let you know.

vooon, 2019-07-16:
So issue is solved. Closing.

closed 2019-07-16 by vooon

stokekld, 2022-03-23:
Hi,

Is there some solution to connect via USB in the last version? I lost HEARTBEAT messages too when I connect the Pixhawk to RPI3 via USB. Thanks.