mavlink/mavros #1276: [ENH] Option for 1:1 transparent mode

tuloski, 2019-07-18:
Right now MAVROS manipulate lots of the data coming and going to UAVs. There is not a 1:1 mapping between MAVlink messages and topics.
I'd like an option to do that, i.e. that every MAVlink message arriving gets translated into a topic with the same fields of the mavlink message. This would require a new ROS msg for every MAVlink message type, and maybe a single plugin that handles all the incoming MAVlink and simply publish all the topics.

Jaeyoung-Lim, 2019-07-18:
@tuloski Are `mavlink/to` `mavlink/from` topics not enough for your usecase?

tuloski, 2019-07-18:
> @tuloski Are `mavlink/to` `mavlink/from` topics not enough for your usecase?

Well, that is really too low level. There is only the payload as uint64 if I remember well. I need to parse manually every single message. Probably (but I'm not sure) I can use easily MAVlink library to parse.
Right now I'm adding manually all the messages I need in mavros_msgs and I wrote a plugin that simply parse and publish the topics I need and it's really good IMO.

TSC21, 2019-07-18:
@tuloski one of the main features in MAVROS that we tend to keep and protect is to translate Mavlink msgs to std ROS msgs, and only if there isn't a std ROS msg that doesn't fit the msg structure of a specific Mavlink msg we add new msg to `mavros_msgs`. We are not aiming to do a one-to-one translation to ROS msgs and that was never the purpose.

Jaeyoung-Lim, 2019-07-18:
@tuloski But wouldn't that mean that you will need to have a plugin that needs to be constantly looking at mavlink messages being added / removed in upstream mavlink?

I understand that there are some mavlink messages you need that are needed, but having all mavlinks map to ROS topics in a single plugin sounds like a huge work to maintain for a not super clear use case.

tuloski, 2019-07-18:
> @tuloski But wouldn't that mean that you will need to have a plugin that needs to be constantly looking at mavlink messages being added / removed in upstream mavlink?
>
> I understand that there are [...]

MAVlink protocol is already completely defined. No need to update it nor check which messages are added. ALL MAVlink messages should be added.

But I understand that actually MAVROS is more like higher level, with the risk of not having really a MAVlink to ROS bridge. Actually it's very PX4 oriented and many of MAVlink messages are not parsed and published. Only the ones useful to PX4 are parsed.

TSC21, 2019-07-18:
> Actually it's very PX4 oriented and many of MAVlink messages are not parsed and published. Only the ones useful to PX4 are parsed.

That statement is false. You can check that most of thr plugins are a translation for the common Mavlink set and we do even have plugins for Ardupilot specific msgs. We provide translation for the msgs that are more commonly used. If you feel that there are msgs you need, you will have to add the plugins yourself.

tuloski, 2019-07-18:
> That statement is false. You can check that most of thr plugins are a translation for the common Mavlink set and we do even have plugins for Ardupilot specific msgs. We provide translation for the [...]

My biggest problem is to understand if a MAVlink message is read by some plugin and the info published in some topic.
For example, take the number of satellites. In MAVlink I can check here https://mavlink.io/en/messages/common.html and I see instantly that GPS_RAW_INT ( #24 ) has the information.
Now in MAVROS I don't know. I'm not even sure it's published somewhere.

BTW I think the easiest way is that I'll make my 1:1 plugin that does that and I disable all the other plugin. I'll do a PR. Will it be accepted? Otherwise I won't spend time on that copy paste work.

Jaeyoung-Lim, 2019-07-18:
> MAVlink protocol is already completely defined. No need to update it nor check which messages are added. ALL MAVlink messages should be added.

I have to disagree with this. I have had experience on regressions when my system is using a outdated mavlink definitions. It doesn't happen too often, but still means that it needs to be continuously maintained. If this plugin is to be well maintained, I am not against it.

TSC21, 2019-07-18:
> BTW I think the easiest way is that I'll make my 1:1 plugin that does that and I disable all the other plugins. I'll do a PR. Will it be accepted? Otherwise I won't spend time on that copy paste work.

I don't think you understood what we said above: we are not going to support a plugin that basically is going to translate all single msgs 1:1. That's no the aim of MAVROS. If you feel that you need that, write one to yourself. Closing this as it's not a relevant change to bring upstream.

closed 2019-07-18 by TSC21

vooon, 2019-07-19:
While ago i read about generator that makes ros package out of mavlink definitions.
Need to find...

Upd:
https://github.com/posilva/mav2rosgenerator
You probably may resurrect that code.

You also may connect it to `mavlink/{to,from}`, so it'll be optional transparent thing.