mavlink/mavros #1277: when I publish vision position estimate

start0821, 2019-07-18:
This is only bug and feature tracker, please use it
to report bugs or request features.

----

### Issue details
I use mavros on raspberry pi 3 B+, and pixhawk1, px4.
When I use `vision pose estimate`  using `/mavros/vision_pose/pose_cov` topic. It works nice but there are timeout errors very frequently.

### MAVROS version and platform

Mavros: 0.32.0
ROS: Kinetic
Ubuntu: 16.04

### Autopilot type and version

[  ] ArduPilot
[x] PX4

Version: 1.8.0 lpe and 1.8.2 lpe

### Node logs

```
[ INFO] [1563478761.600113899]: HP: requesting home position
[ERROR] [1563478770.400896903]: FCU: [lpe] vision position timeout 
[ INFO] [1563478770.502498410]: FCU: [lpe] vision position init: -0.00  0.00  0.00 m s
[ERROR] [1563478770.554006034]: FCU: [lpe] vision position timeout 
[ INFO] [1563478770.602714666]: FCU: [lpe] vision position init: -0.00  0.00  0.00 m s
[ERROR] [1563478770.653201307]: FCU: [lpe] vision position timeout 
[ INFO] [1563478770.715288235]: FCU: [lpe] vision position init: -0.00  0.00  0.00 m s
[ERROR] [1563478770.752205280]: FCU: [lpe] vision position timeout 
[ INFO] [1563478770.812787895]: FCU: [lpe] vision position init:  0.00  0.00  0.00 m s
[ERROR] [1563478770.854053348]: FCU: [lpe] vision position timeout 
[ INFO] [1563478770.922755961]: FCU: [lpe] vision position init:  0.00  0.00  0.00 m s
[ERROR] [1563478770.952183785]: FCU: [lpe] vision position timeout 
[ INFO] [1563478771.032685746]: FCU: [lpe] vision position init: -0.00  0.00  0.00 m s
[ERROR] [1563478771.508026730]: TM : Time jump detected. Resetting time synchroniser.

```

### Diagnostics

```
header:   seq: 1362
  stamp: 
    secs: 1563479617
    nsecs: 803817226
  frame_id: ''
status: 
  - 
    level: 0
    name: "mavros: FCU connection"
    message: "connected"
    hardware_id: "/dev/ttyAMA0:921600"
    values: 
      - 
        key: "Received packets:"
        value: "36841"
      - 
        key: "Dropped packets:"
        value: "0"
      - 
        key: "Buffer overruns:"
        value: "0"
      - 
        key: "Parse errors:"
        value: "0"
      - 
        key: "Rx sequence number:"
        value: "163"
      - 
        key: "Tx sequence number:"
        value: "54"
      - 
        key: "Rx total bytes:"
        value: "15482224"
      - 
        key: "Tx total bytes:"
        value: "5200820"
      - 
        key: "Rx speed:"
        value: "17912.000000"
      - 
        key: "Tx speed:"
        value: "9549.000000"
  - 
    level: 2
    name: "mavros: GPS"
    message: "No satellites"
    hardware_id: "/dev/ttyAMA0:921600"
    values: 
      - 
        key: "Satellites visible"
        value: "0"
      - 
        key: "Fix type"
        value: "0"
      - 
        key: "EPH (m)"
        value: "Unknown"
      - 
        key: "EPV (m)"
        value: "Unknown"
  - 
    level: 0
    name: "mavros: Heartbeat"
    message: "Normal"
    hardware_id: "/dev/ttyAMA0:921600"
    values: 
      - 
        key: "Heartbeats since startup"
        value: "897"
      - 
        key: "Frequency (Hz)"
        value: "0.959997"
      - 
        key: "Vehicle type"
        value: "Quadrotor"
      - 
        key: "Autopilot type"
        value: "PX4 Autopilot"
      - 
        key: "Mode"
        value: "MANUAL"
      - 
        key: "System status"
        value: "Standby"
  - 
    level: 2
    name: "mavros: System"
    message: "Sensor health"
    hardware_id: "/dev/ttyAMA0:921600"
    values: 
      - 
        key: "Sensor present"
        value: "0x0007000F"
      - 
        key: "Sensor enabled"
        value: "0x0001000F"
      - 
        key: "Sensor health"
        value: "0x0006000F"
      - 
        key: "3D gyro"
        value: "Ok"
      - 
        key: "3D accelerometer"
        value: "Ok"
      - 
        key: "3D magnetometer"
        value: "Ok"
      - 
        key: "absolute pressure"
        value: "Ok"
      - 
        key: "rc receiver"
        value: "Fail"
      - 
        key: "CPU Load (%)"
        value: "54.1"
      - 
        key: "Drop rate (%)"
        value: "0.0"
      - 
        key: "Errors comm"
        value: "0"
      - 
        key: "Errors count #1"
        value: "0"
      - 
        key: "Errors count #2"
        value: "0"
      - 
        key: "Errors count #3"
        value: "0"
      - 
        key: "Errors count #4"
        value: "0"
  - 
    level: 0
    name: "mavros: Battery"
    message: "Normal"
    hardware_id: "/dev/ttyAMA0:921600"
    values: 
      - 
        key: "Voltage"
        value: "11.58"
      - 
        key: "Current"
        value: "0.1"
      - 
        key: "Remaining"
        value: "66.0"
  - 
    level: 0
    name: "mavros: Time Sync"
    message: "Normal"
    hardware_id: "/dev/ttyAMA0:921600"
    values: 
      - 
        key: "Timesyncs since startup"
        value: "8449"
      - 
        key: "Frequency (Hz)"
        value: "9.999948"
      - 
        key: "Last RTT (ms)"
        value: "5.076940"
      - 
        key: "Mean RTT (ms)"
        value: "3.621689"
      - 
        key: "Last remote time (s)"
        value: "1092.091512000"
      - 
        key: "Estimated time offset (s)"
        value: "1563478525.616504908"
---

```

### Check ID

```
OK. I got messages from 1:1.

---
Received 6261 messages, from 2 addresses
sys:comp   list of messages
  1:1     0, 1, 2, 4, 140, 141, 147, 26, 30, 31, 32, 36, 65, 70, 74, 76, 77, 83, 230, 102, 105, 111, 241, 245
 28:0     96000

```

vooon, 2019-07-19:
Maybe rate of vision messages is to low?

start0821, 2019-07-19:
Thank you for replying!
I think it's not the reason.
Because I change the frequency several times (1,5,7,10,20,50 Hz..).
And I think there may be some problem on timestamp...
I set timestamp on `vision topic` to `ros::time::now`. How can I fix it?

start0821, 2019-07-24:
Hi, guys!! 
I resolved the problem. I don't know what's the main problem of timeout, but when I change the `ROS_MASTER_URI` from remote laptop ip to `localhost` on raspberry pi. And it works well!!
Additionally, if your code is running on raspberry pi, it's low throughput also can be reason of timeout.

closed 2019-07-24 by start0821