mavlink/mavros #1278: Publishing transforms both directions is incorrect

tfoote, 2019-07-19:
The comment that you need to publish the transform the other way is incorrect. 

https://github.com/mavlink/mavros/blob/bdc5c7386f9451ac244b5a85c087ad620fd3a989/mavros/src/lib/uas_data.cpp#L57

Only one of these should be used:

https://github.com/mavlink/mavros/blob/366a9a74847b2e231035b47fa06651527a4556b9/mavros/src/lib/uas_data.cpp#L56-L58

They will fight with each other and cause continuous reparenting such that at any given time the last one recieved will win. Likely the first version tested caused a disconnected tree by inadvertent reparenting and adding the second call masked it out. 

And while I'm looking at this code. It's cleaner and more efficient to publish a vector of static transforms to take advantage of latching rather than publishing a sequence of individual transforms.

mhkabir, 2019-07-20:
We brought this up here before: https://github.com/mavlink/mavros/pull/1252#issuecomment-503084522

mhkabir, 2019-07-20:
@jkflying @baumanta Can you guys try to test if the timestamp issue exists with the duplicate publisher removed?

baumanta, 2019-07-22:
@tfoote you were right with your you comment on [issue #388 ](https://github.com/ros/geometry2/issues/388#issuecomment-513409748). The tf loop in mavros seems to be responsible for getting always the tf at ros::Time::now().

I put a tf listener in the avoidance code which retrieved the transforms at the timestamp of the last pointcloud and at ros::Time::now() and printed both transforms. As the drone was moving I would expenct the transforms to be slightly different. Using Mavros 0.29.2. both transforms were always identical even thought the timestamps were not. Using Mavros [PR #1252](https://github.com/mavlink/mavros/pull/1252) the transforms were different.

mhkabir, 2019-07-22:
OK, I suspect why the `odom` plugin code wasn't working is due to an obscure limitation of latched topics - you cannot publish different messages to a latched topic from the same process.

See https://github.com/ros/ros_comm/issues/146 and https://answers.ros.org/question/261815/how-can-i-access-all-static-tf2-transforms/.

The workaround for this is to construct a vector of all the transforms and publish them in one shot (pass to specialized TF2 function http://docs.ros.org/melodic/api/tf2_ros/html/c++/classtf2__ros_1_1StaticTransformBroadcaster.html#a83d0664cb0cc85688b60a2236e9d81ff)

TSC21, 2019-10-07:
Fixed https://github.com/mavlink/mavros/pull/1300.

closed 2019-10-07 by TSC21