mavlink/mavros #1279: coordinate frames mavros/gazebo

elgarbe, 2019-07-20:
I'm runing px4 in SITL with gazebo. I spawn 2 iris in gazebo world:

`
name: [ground_plane, asphalt_plane, iris_2, iris_1]
pose: 
  - 
    position: 
      x: 0.0
      y: 0.0
      z: 0.0
    orientation: 
      x: 0.0
      y: 0.0
      z: 0.0
      w: 1.0
  - 
    position: 
      x: 0.0
      y: 0.0
      z: 0.0
    orientation: 
      x: 0.0
      y: 0.0
      z: 0.0
      w: 1.0
  - 
    position: 
      x: -0.019805065431
      y: -4.99587068079
      z: 0.12498486787
    orientation: 
      x: -0.000230943368958
      y: -0.000417520484615
      z: 0.0648003868379
      w: 0.99789813218
  - 
    position: 
      x: -0.0197702045516
      y: 5.00411155332
      z: 0.124996387745
    orientation: 
      x: -9.26127168815e-05
      y: -0.000172612596903
      z: 0.0648782734594
      w: 0.997893166256
`

As you can see they are at 0,5,0 and 0,-5,0. So they are displaced on gazebo Y axis.
/mavros/global_position/global reports:

`
elgarbe@i7:~$ rostopic echo -n1 /uav1/mavros/global_position/global 
header: 
  seq: 132598
  stamp: 
    secs: 633440121
    nsecs: 418143808
  frame_id: "base_link"
status: 
  status: 0
  service: 1
latitude: 47.3977872
longitude: 8.5455931
altitude: 535.488938054
position_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
position_covariance_type: 1
elgarbe@i7:~$ rostopic echo -n1 /uav2/mavros/global_position/global 
header: 
  seq: 132556
  stamp: 
    secs: 633440150
    nsecs: 107474688
  frame_id: "base_link"
status: 
  status: 0
  service: 1
latitude: 47.397697
longitude: 8.5455937
altitude: 535.091894966
position_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
position_covariance_type: 1
---
`
so, a change in latitude (East-West) is translated to a Y axis, so the coordinate frame is NED, right?

then y transform that lla message to ECEF and get:

`
[INFO] [1563652484.586929, 2830.334000]: ECEF1: [ 4277578.83208723   642768.77958409  4672206.45458772]
[INFO] [1563652484.602226, 2830.350000]: ECEF2: [ 4277586.00679073   642769.87295696  4672199.54669639]
`

So, now the gazebo Y offset is traduced to a X change. What am I missing?

thank

elgarbe, 2019-08-19:
can someone try this? I have not find a solution...

TSC21, 2019-08-19:
Gazebo uses the ENU frame convention.

elgarbe, 2019-08-19:
but, mavros does not take into acount NED to ENU convertion?

closed 2026-11-15 by github-actions[bot]