mavlink/mavros #1280: mavros installation on Raspberry Pi 4 buster OS

bozkurthan, 2019-07-23:
Mavros: Latest
ROS: Kinetic || Melodic
OS: Raspbian Buster
Platform: Raspberry Pi 4

Hello, I'm trying to install MAVROS to RPi 4. However, I couldn't finish it without any error.

Here is the list of what I tried:

### 1. Followed this guide to install kinetic and getting error by Boost version: 

http://wiki.ros.org/ROSberryPi/Installing%20ROS%20Kinetic%20on%20the%20Raspberry%20Pi

I found details which is about this issue are here:
https://answers.ros.org/question/328303/i-cannot-install-ros-kinetic-desktop-following-the-tutorial/

> So there is two big issues that make impossible install ROS in RPi4,
> 
> I believe the issue is more that you're trying to build a version of ROS on a platform that is not officially supported, with versions of dependencies that are not supported (by Kinetic at least).
> 
> ros/rospack#92 seems to be your exact issue.
> 
> The problem appears to be that you are attempting to build ROS Kinetic on an unsupported platform. Kinetic is only supported on (from REP 3: Target Platforms - Kinetic Kame (May 2016 - May 2021)):
> 
> Required Support for:
> 
> Ubuntu Wily (15.10)
> Ubuntu Xenial (16.04)
> Recommended Support for:
> 
> Debian Jessie
> And with Boost version 1.55. Anything else is not explicitly supported.
> 
> The comments on ros/rospack#92 seem to suggest that the Kinetic version of rospack (and probably other packages) depend on the tr1 headers. Newer versions no longer require them (ros/rospack#80).
> 
> If you have the option, I would suggest you try to build Melodic from source, as that should be compatible with Boost 1.65.
> 
> There is still no guarantee though, as Melodic is not officially supported on Raspbian.

### 2. Try to install Melodic version of ROS by following source instillation and there is no error with melodic 

(I just change command: `rosdep install -y --from-paths src --ignore-src --rosdistro melodic -r --os=debian:buster`)
http://wiki.ros.org/melodic/Installation/Source

3. After installed melodic try to install MAVROS by source and getting error:

```
pi@raspberrypi:~/catkin_ws $ rosdep install --from-paths src --ignore-src -y --rosdistro melodic
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
mavros: No definition of [rosunit] for OS version [buster]
mavlink: No definition of [catkin] for OS version [buster]
mavros_msgs: No definition of [geographic_msgs] for OS version [buster]
libmavconn: No definition of [rosunit] for OS version [buster]
test_mavros: No definition of [control_toolbox] for OS version [buster]
mavros_extras: No definition of [tf2_eigen] for OS version [buster]
```
I want to use Jessie or Ubuntu Mate for RPi 4 to install without error but there is no support for RPi 4 .

Is there any guide or possible way to install MAVROS to RPi 4 or do you plan to support RPi 4 soon?

Thanks.

fnoop, 2019-08-25:
To progress the build, you can slightly adapt the process to this:

# Adapted from here: https://github.com/mavlink/mavros/blob/master/mavros/README.md#source-installation

```
mkdir -p ~/catkin_ws/src
cd ~/catkin_ws
catkin init
wstool init src
rosinstall_generator --rosdistro melodic mavlink | tee /tmp/mavros.rosinstall
rosinstall_generator --upstream-development mavros mavros_extras test_mavros mavros_msgs sensor_msgs tf control_toolbox realtime_tools tf2_ros urdf python_orocos_kdl |tee -a /tmp/mavros.rosinstall
wstool merge -t src /tmp/mavros.rosinstall
wstool update -t src -j4
rosdep install --from-paths src --ignore-src -y
sudo ./src/mavros/mavros/scripts/install_geographiclib_datasets.sh
catkin build
```

However, it then fails compiling one of the dependencies - realtime_tools:
```
Errors     << realtime_tools:cmake /srv/maverick/catkin_ws/logs/realtime_tools/build.cmake.000.log
CMake Warning at /opt/ros/melodic/share/catkin/cmake/test/gtest.cmake:151 (message):
  skipping gmock 'realtime_box_tests' in project 'realtime_tools' because
  gmock was not found
Call Stack (most recent call first):
  /opt/ros/melodic/share/catkin/cmake/test/gtest.cmake:80 (_catkin_add_executable_with_google_test)
  /opt/ros/melodic/share/catkin/cmake/test/gtest.cmake:54 (_catkin_add_google_test)
  CMakeLists.txt:23 (catkin_add_gmock)

CMake Error at /srv/maverick/catkin_ws/src/realtime_tools/CMakeLists.txt:24 (target_link_libraries):
  Cannot specify link libraries for target "realtime_box_tests" which is not
  built by this project.
```

fnoop, 2019-08-25:
@vooon @TSC21 Is there any way of skipping test_mavros?

vooon, 2019-08-25:
@fnoop yes, you can blacklist packages by `catkin config --blacklist "list of packages"`

fnoop, 2019-08-25:
@vooon no matter how I try to blacklist test_mavros, it keeps coming back:
```
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
test_mavros: No definition of [control_toolbox] for OS version [buster]
```

vooon, 2019-08-25:
Maybe building `control_toolbox` from source will help.
```
rosinstall_generator --rosdistro melodic control_toolbox | tee -a /tmp/mavros.rosinstall

fnoop, 2019-08-25:
Hi @vooon thanks for the suggestions, but I couldn't get any of those to work.  What eventually worked was:
```
mkdir -p ~/catkin_ws/src
cd ~/catkin_ws
catkin init
wstool init src
rosinstall_generator --rosdistro melodic mavlink | tee /tmp/mavros.rosinstall
rosinstall_generator --upstream-development mavros mavros_extras mavros_msgs test_mavros sensor_msgs  control_toolbox realtime_tools tf tf2_ros python_orocos_kdl urdf |tee -a /tmp/mavros.rosinstall
wstool merge -t src /tmp/mavros.rosinstall
wstool update -t src -j4
rosdep install --from-paths src --ignore-src -y
sudo ./src/mavros/mavros/scripts/install_geographiclib_datasets.sh
catkin build -DCATKIN_ENABLE_TESTING=0
```

Crucial bit is the `-DCATKIN_ENABLE_TESTING=0`

juanmed, 2019-09-01:
@fnoop Thanks for the installation instructions on your platform. I actually just followed them on a Nvidia Jetson Nano and can confirm I could successfully install MAVROS on it. 

Memory  4 GB
Processor ARMv8 Processor rev 1 (v8l) × 4
Graphics NVIDIA Tegra X1 (nvgpu)/integrated
OS: Ubuntu 18.04LTS 64-bit
ROS Melodic

fishworx, 2019-09-07:
@juanmed , @vooon , I have installed Ros2 dashing on an Nvidia nano. Is it the same process to install for dashing?

TSC21, 2019-09-07:
> @juanmed , @vooon , I have installed Ros2 dashing on an Nvidia nano. Is it the same process to install for dashing?

MAVROS is a ROS (1) package. Not a ROS2 package. And there's no current plan to port it. So the answer is no.

fishworx, 2019-09-07:
@TSC21 , could i use mavros with the ros-dashing-ros1-bridge?

TSC21, 2019-09-07:
> @TSC21 , could i use mavros with the ros-dashing-ros1-bridge?

You can but that means you need to recreate the `mavros_msgs` package on ROS2 as well so there can be a map between them.

fishworx, 2019-09-07:
@TSC21 , i dont want to recreate the wheel. How are other people running Ros/python3/mavros? Which ros install are they using?

TSC21, 2019-09-07:
> @TSC21 , i dont want to recreate the wheel. How are other people running Nvidia Nano/Ros/python3/mavros? Which ros install are they using?

Usins ROS (1) distros, which are the ones that MAVROS was written to.

fishworx, 2019-09-07:
@TSC21 , can you post a link please, everything i've seen says use ros2 dashing....

TSC21, 2019-09-07:
> @TSC21 , can you post a link please, everything i've seen says use ros2 dashing....

You must be mistaken. Where exactly are you seeing MAVROS assotiated with ROS2? Now I am curious? You can start by checking the above comments - they are using `catkin`, which is a build system supported in ROS, while ROS2 uses `ament` and `colcon`.

TSC21, 2019-09-07:
Our README is quite clear as well: https://github.com/mavlink/mavros/tree/master/mavros#installation

fishworx, 2019-09-07:
@TSC21 , I'm just new to this. My research told me python3 only works on ros2. Then the kinetic install failed. However the 'dashing' install worked perfectly.... I noticed mavros is for kinetic, and was wondering if it worked on dashing too? I'm just looking for a working solution to use nano/python3/Ros/Pixhawk.

https://index.ros.org/doc/ros2/Installation/Dashing/

fnoop, 2019-09-07:
@fishworx Even if ROS(1) is compiled using python2 (it's actually quite difficult to get it to build with python3 even if you try), you can still happily use python3 scripts while loading the python2-based mavros modules. (add /opt/ros/melodic/lib/python2.7/dist-packages to PYTHONPATH).  There are also various python3 ros packages for building:
```
[dev] [mav@dev ~]$ apt search python3-ros
Sorting... Done
Full Text Search... Done
python3-ros-buildfarm/bionic 3.0.0-100 all
  Build farm used to build the ROS ecosystem's packages.

python3-ros-buildfarm-modules/bionic 3.0.0-1 all
  Build farm used to build the ROS ecosystem's packages.

python3-rosdep/bionic 0.15.2-1 all
  rosdep package manager abstraction tool for ROS

python3-rosdep2/bionic 0.11.8-1 all
  rosdep package manager abstraction tool for Robot OS (Python 3)

python3-rosdistro/bionic 0.7.4-100 all
  A tool to work with rosdistro files

python3-rosdistro-modules/bionic 0.7.4-1 all
  A tool to work with rosdistro files

python3-rosinstall/bionic 0.7.8-1 all
  The installer for ROS

python3-rosinstall-generator/bionic 0.1.17-1 all
  A tool for generating rosinstall files

python3-rospkg/bionic 1.1.10-100 all
  ROS package library

python3-rospkg-modules/bionic,now 1.1.10-1 all [installed]
  ROS package library
```

You can find some build hints here:
https://github.com/goodrobots/maverick/blob/master/manifests/maverick-modules/maverick_ros/manifests/init.pp
Our system is entirely python3 based, but even when you build ROS/MAVROS from source it still uses python2.  There is a way to override and tell it to build ROS itself from python3 but I've had no luck with that method (https://github.com/goodrobots/maverick/issues/647#issuecomment-422161778).  ROS has announced that the next/final version of ROS1 will fully support python3 as python2 will be EOL by that point.
Also: https://github.com/mavlink/mavros/issues/940

TL;DR - you can use python3 scripts just fine with ros/mavros, even though it's based underneath on python2.

TSC21, 2019-09-07:
> @TSC21 , I'm just new to this. My research told me python3 only works on ros2. Then the kinetic install failed. However the 'dashing' install worked perfectly.... I noticed mavros is for kinetic, [...]

Why do you strictly required Python 3? If you are new to this is I recommend you start by understanding how ROS works and what are the main differences between ROS and ROS2 (it's not only in Python 3). Now, if you want to use MAVROS, you need to check what MAVROS depends on and where it is actually supported. It's not only in Kinetic. You can use latest ROS 1 distro which is Melodic.

fnoop, 2019-09-07:
@fishworx btw that build manifest also works perfectly on jetson nano running the latest jetpack.  I'm intending to upload a prebuilt OS image at some point with ros/mavros included but haven't had the time as of yet.

fishworx, 2019-09-07:
you know python2 is being discontinued??? also we use openCV which is python3....

https://pythonclock.org/

fishworx, 2019-09-07:
@fnoop , so just copy and paste, no need to change melodic?

TSC21, 2019-09-07:
> you know python2 is being discontinued??? also we use openCV which is python3....
> 
> https://pythonclock.org/

Absolutely. But as someone said already, being EOL doesn't mean it doesn't work ;) Also, please read what me and @fnoop wrote.

fnoop, 2019-09-07:
Also note that the whole python3 question isn't so much a mavros (or any other ros module) issue, but a ROS issue.  The ros modules just have to support both, which mavros now does.
https://discourse.ros.org/t/planning-future-ros-1-distribution-s/6538

To have a future, MAVROS needs to move to ROS2, but there are no plans for that currently.  
https://github.com/mavlink/mavros/issues/771
https://github.com/mavlink/mavros/issues/1000
Note the RTSP comments are a red herring - that has nothing to do with MAVROS, which is a *mavlink* gateway and is absolutely needed going forward.

fishworx, 2019-09-07:
@fnoop , can you confirm your using melodic on a nano/ubuntu 18.04? I cant get melodic to work....

fnoop, 2019-09-07:
```
[dev] [mav@maverick-tegra ~]$ uname -a
Linux maverick-tegra 4.9.140-tegra #1 SMP PREEMPT Tue Jul 16 17:04:49 PDT 2019 aarch64 aarch64 aarch64 GNU/Linux
[dev] [mav@maverick-tegra ~]$ maverick info

Maverick - UAV Companion Computer System - Version 2.0.0-dev

WARNING: Maverick is using branch:master, not stable

Maverick Environment:    dev

Vendor:			Nvidia
Model:			Jetson Nano
CPU Cores:		4
Memory:			4059504 kB
Swap:			2029744 kB
SD Card Size:		30707.712 Mb
 ----------------------------------
OS:			Ubuntu
OS Version:		18.04
Architecture:		aarch64
Kernel:			4.9.140-tegra
Timezone:		BST
FQDN:			maverick-tegra.lan
IP Address:		192.168.1.86
```

```
[dev] [mav@maverick-tegra ~]$ ps -ef |grep roslaunch
mav       8111  6066  1 Aug25 ?        04:13:39 /srv/maverick/software/python/bin/python3 /srv/maverick/software/ros/current/bin/roslaunch mavros apm.launch fcu_url:=tcp://localhost:5770
```

```
[dev] [mav@maverick-tegra ~]$ ls -l /opt/ros
total 0
lrwxrwxrwx 1 root root 34 Aug 12 12:22 current -> /srv/maverick/software/ros/melodic
lrwxrwxrwx 1 root root 34 Aug 12 12:22 melodic -> /srv/maverick/software/ros/melodic
```

Yes, works perfectly :)  Note roslaunch is running under python3.

fishworx, 2019-09-07:
@fnoop , can you share a link on how to install ros please?

juanmed, 2019-09-07:
@fishworx  https://www.stereolabs.com/blog/ros-and-nvidia-jetson-nano/

fnoop, 2019-09-07:
See above - this is my build manifest that works across all platforms:
https://github.com/goodrobots/maverick/blob/master/manifests/maverick-modules/maverick_ros/manifests/init.pp

Also check out jetsonhacks:
https://github.com/JetsonHacksNano/installROS

fishworx, 2019-09-07:
@fnoop , some unmet dependencies? is this a success?

[image]

fnoop, 2019-09-07:
@fishworx No, but unrelated to this issue - suggest you raise a new issue for that.

ashirsat, 2020-07-23:
I am unable to build mavlink packages from source for raspberry pi3 model b running buster and ros-kinetic. I followed the guide 

> To progress the build, you can slightly adapt the process to this:
> 
> Adapted from here: https://github.com/mavlink/mavros/blob/master/mavros/README.md#source-installation
> 
> ```
> mkdir -p ~/catkin_ws/src
> cd ~/catkin_ws
> catkin init
> wstool init src
> rosinstall_generator --rosdistro melodic mavlink | tee /tmp/mavros.rosinstall
> rosinstall_generator --upstream-development mavros mavros_extras mavros_msgs test_mavros sensor_msgs  control_toolbox realtime_tools tf tf2_ros python_orocos_kdl urdf |tee -a /tmp/mavros.rosinstall
> wstool merge -t src /tmp/mavros.rosinstall
> wstool update -t src -j4
> rosdep install --from-paths src --ignore-src -y
> sudo ./src/mavros/mavros/scripts/install_geographiclib_datasets.sh
> catkin build
> ```

as listed above but I cannot find any support for geographic lib for buster.and all mavlink packages listed above also fail to build

Any comments or updates would be appreciated.

fnoop, 2020-07-23:
Hi, try installing libgeographic-dev through apt:
 `apt install -y libgeographic-dev`

Also kinetic doesn't support buster - you should be using the latest noetic release, I think.

cjmv, 2020-12-23:
Hi all,

Exactly the subject I was looking for.

In my particular case I've installed Ros-noetic, and the step 4 from the Mavros [source installation](https://github.com/mavlink/mavros/blob/master/mavros/README.md#installation) is the only that actually get's me in trouble.

When I run 
`rosdep install --from-paths src --ignore-src -y`

I get the error:
```
ERROR: the following rosdeps failed to install
  apt: command [sudo -H apt-get install -y ros-noetic-angles] failed
```
and actually looking in the apt, there's no such package. I did find a couple of packages that could be related like:
```
libangles-dev - Robot OS set of simple math utilities to work with angles
python-angles - Robot OS set of simple math utilities to work with angles (Python 2)
python3-angles - Robot OS set of simple math utilities to work with angles (Python 3)
```

and one by one I've installed them, to see if would make a difference. Any ideas on how to get these dependencies installed?

bozkurthan, 2022-11-28:
I think the issue is timeout since the many changes are applied for this issue. I'm closing as stale.

closed 2022-11-28 by bozkurthan