mavlink/mavros #1283: Cannot launch mavros.

Kumaron, 2019-07-25:
### Issue details
Cannot launch mavros px4.launch. Always `REQUIRED process [mavros-2] has died! `

### MAVROS version and platform

Mavros: ?0.32.0?
ROS: ?Kinetic?
Ubuntu: ?16.04?

### Autopilot type and version

[  ] ArduPilot
[  ] PX4 √

Version: ?1.8.2?

### Node logs

```
kumaron@ubuntu:~$ roslaunch mavros px4.launch fcu_url:="udp://:14540@127.0.0.1:14557"
... logging to /home/kumaron/.ros/log/8c04f8ea-aec4-11e9-970a-843a4b933d20/roslaunch-ubuntu-4105.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ubuntu:36503/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/hrlv_ez4_pub/field_of_view: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/frame_id: hrlv_ez4_sonar
 * /mavros/distance_sensor/hrlv_ez4_pub/id: 0
 * /mavros/distance_sensor/hrlv_ez4_pub/orientation: PITCH_270
 * /mavros/distance_sensor/hrlv_ez4_pub/send_tf: True
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/laser_1_sub/id: 3
 * /mavros/distance_sensor/laser_1_sub/orientation: PITCH_270
 * /mavros/distance_sensor/laser_1_sub/subscriber: True
 * /mavros/distance_sensor/lidarlite_pub/field_of_view: 0.0
 * /mavros/distance_sensor/lidarlite_pub/frame_id: lidarlite_laser
 * /mavros/distance_sensor/lidarlite_pub/id: 1
 * /mavros/distance_sensor/lidarlite_pub/orientation: PITCH_270
 * /mavros/distance_sensor/lidarlite_pub/send_tf: True
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/sonar_1_sub/id: 2
 * /mavros/distance_sensor/sonar_1_sub/orientation: PITCH_270
 * /mavros/distance_sensor/sonar_1_sub/subscriber: True
 * /mavros/fake_gps/eph: 2.0
 * /mavros/fake_gps/epv: 2.0
 * /mavros/fake_gps/fix_type: 3
 * /mavros/fake_gps/geo_origin/alt: 408.0
 * /mavros/fake_gps/geo_origin/lat: 47.3667
 * /mavros/fake_gps/geo_origin/lon: 8.55
 * /mavros/fake_gps/gps_rate: 5.0
 * /mavros/fake_gps/mocap_transform: True
 * /mavros/fake_gps/satellites_visible: 5
[... 220 lines cut ...]
[mavros-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```

### Diagnostics

```
place here result of:
rostopic echo -n1 /diagnostics
```

### Check ID

```
ERROR. I got 0 addresses, but not your target 1:1

---
Received 0 messages, from 0 addresses
sys:comp   list of messages

```

TSC21, 2019-10-15:
@Kumaron without more information it will be quite difficult to track what's happening? What exactly are you trying to do? Connect with PX4 SITL? What's your setup?

bafu, 2019-12-24:
Hi @TSC21 I met the same `exit code -11` on Ubuntu 16.04.

I am happy to help debug if you can share some suggestions and I can provide more logs.

---
**Steps to Reproduce**

1. Install ROS by following the official steps http://wiki.ros.org/kinetic/Installation/Ubuntu

1. Install mavros packages and geographiclib datasets

    ```
    $ sudo apt-get install ros-kinetic-mavros ros-kinetic-mavros-extras

    $ wget https://raw.githubusercontent.com/mavlink/mavros/master/mavros/scripts/install_geographiclib_datasets.sh
    $ chmod 755 install_geographiclib_datasets.sh
    $ sudo ./install_geographiclib_datasets.sh
    ```

1. Install px4 firmware

    ```
    $ mkdir -p ~/ros_ws/src && cd ~/ros_ws/src
    $ git clone https://github.com/PX4/Firmware.git
    $ cd Firmware
    $ make px4_sitl_default gazebo
    ```

1. Add paths to the end of `~/.bashrc`

    ```
    source /opt/ros/kinetic/setup.bash
    rosws=~/ros_ws/
    px4ws=$rosws/src/Firmware
    source $px4ws/Tools/setup_gazebo.bash $px4ws $px4ws/build/px4_sitl_default
    export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:$px4ws:$px4ws/Tools/sitl_gazebo
    export ROS_HOSTNAME=localhost
    export ROS_MASTER_URI=http://localhost:11311
    ```

1. Run `roslaunch mavros px4.launch`
1. Got the `exit code -11`

    ```
    auto-starting new master
    process[master]: started with pid [25340]
    ROS_MASTER_URI=http://localhost:11311

    setting /run_id to 987e3c3e-2643-11ea-a10e-5cf9388f7c86
    process[rosout-1]: started with pid [25353]
    started core service [/rosout]
    process[mavros-2]: started with pid [25357]
    ================================================================================REQUIRED process [mavros-2] has died!
    process has died [pid 25357, exit code -11, cmd /opt/ros/kinetic/lib/mavros/mavros_node __name:=mavros __log:=/home/debug/.ros/log/987e3c3e-2643-11ea-a10e-5cf9388f7c86/mavros-2.log].
    log file: /home/debug/.ros/log/987e3c3e-2643-11ea-a10e-5cf9388f7c86/mavros-2*.log
    Initiating shutdown!
    ================================================================================
    [mavros-2] killing on exit
    [rosout-1] killing on exit
    [master] killing on exit
    shutting down processing monitor...
    ... shutting down processing monitor complete
    done
```

closed 2026-11-15 by github-actions[bot]