mavlink/mavros #1284: sensor_msgs::BatteryState  percentage with a negative value

FaboNo, 2019-07-27:
This is only bug and feature tracker, please use it
to report bugs or request features.

----

### Issue details

_Please describe the problem, or desired feature_
When I am reading the battery status (/mavros/battery -- see below) I noticed that the percentage value is always small and negative. However in the documentation this value should be between 0 and 1. Can you tell me what is missing? Thank you

```
  seq: 216
  stamp: 
    secs: 1528206242
    nsecs: 464688044
  frame_id: ''
voltage: 65.5350036621
current: 0.00999999977648
charge: nan
capacity: nan
design_capacity: nan
percentage: -0.00999999977648
power_supply_status: 2
power_supply_health: 0
power_supply_technology: 3
present: True
cell_voltage: []
location: "id0"
serial_number: ''
---
header: 
  seq: 217
  stamp: 
    secs: 1528206242
    nsecs: 665512459
  frame_id: ''
voltage: 65.5350036621
current: 0.00999999977648
charge: nan
capacity: nan
design_capacity: nan
percentage: -0.00999999977648
power_supply_status: 2
power_supply_health: 0
power_supply_technology: 3
present: True
cell_voltage: []
location: "id0"
serial_number: ''
```

I added a screenshot of the px4 parameters associated with the battery in case I missed something?

[image]

### MAVROS version and platform

Mavros: ?0.18.4?
ROS: ?Kinetic?
Ubuntu: ?16.04?

### Autopilot type and version

[  ] ArduPilot
[ x ] PX4

Version: ?3.7.1?

### Node logs

[... 333 lines cut ...]
    message: "Normal"
    hardware_id: "/dev/ttyPIXEXT:921600"
    values: 
      - 
        key: "Voltage"
        value: "65.54"
      - 
        key: "Current"
        value: "-0.0"
      - 
        key: "Remaining"
        value: "-1.0"
---

### Check ID

```
rosrun mavros checkidOK. I got messages from 1:1.

---
Received 5724 messages, from 1 addresses
sys:comp   list of messages
  1:1     0, 1, 132, 140, 141, 147, 4, 26, 30, 31, 32, 36, 74, 76, 83, 85, 230, 105, 111, 241, 242, 245

```

vooon, 2019-07-28:
Battery messages are processed here:
- https://github.com/mavlink/mavros/blob/master/mavros/src/plugins/sys_status.cpp#L748
- https://github.com/mavlink/mavros/blob/8aabb0bf1b7ba05ddd5da487bbb729655a37a2e9/mavros/src/plugins/sys_status.cpp#L748

Math is simple `remaining / 100.0f` (to map 0-100 to 0.0-1.0), and remaining may be `-1` in case of unknown.
So probably we should define what to return in that case.
In any case, you may use `r < 0` to check that condition.

FaboNo, 2019-07-29:
@vooon Thank you for your message. I got the same values with Analyze (qgc) and I am surprised by the voltage value (65 volts and I have a 4C lipo battery). I am wondering whether I did not correctly configure the power panel with QGC or not.

vooon, 2019-07-29:
@FaboNo yes, likely that battery monitor is misconfigured. But i don't know what exactly wrong, so probably first better to try reconfigure until you got possible values in QGC.
Also i suppose, that unknown remaining percentage may be because lack of current monitor (in config).

FaboNo, 2019-08-04:
@vooon the cable from the power module was plugged inot POWER2 instead of POWER1 on the pixhawk v2.1. This is why the pixhawk s not able to measure the voltage. Now it works fine.

closed 2019-08-04 by FaboNo