mavlink/mavros #1291: /mavros/global_position/local lags to /gazebo/model_state

elgarbe, 2019-08-13:
Hi, I'm working with px4 in sitl and gazebo with iris model.
I've noticed that odometry msg from /global_position/local is not accurate compared with /gazebo/model_state. Is there a way to adjust the odometry in order to get better result?

Thank

TSC21, 2019-08-13:
@elgarbe the odometry by itself has noise and drift assotiated - it's a result of the estimator fusing the data of different sensors. `/gazebo/model_state` basically gives you the ground truth, so it's basically the real vehicle position. So, in that sense, odometry and ground truth will always differ. Now what can be evaluated is the size of the error that you are getting: is it meters? centimeters? Also take into account that the origin of the odometry may be different from the origin of the ground truth, which is likely the case.

elgarbe, 2019-08-19:
Ok, I've tested again, closing gazebo gui and work a lot better, so seems to be a cpu load problem.
thank

closed 2019-08-19 by elgarbe