mavlink/mavros #1293: SITL error: unconnected trees

TobiMiller, 2019-08-14:
Hey,

with actual PX4 master and mavros i am getting following error

`ODOM: Ex: Could not find a connection between 'local_origin_ned' and 'fcu' because they are not part of the same tree.Tf has two or more unconnected trees.`

[frames.pdf](https://github.com/mavlink/mavros/files/3500358/frames.pdf)

TSC21, 2019-08-14:
That's a direct consequence of https://github.com/mavlink/mavros/pull/1252. A definite solution is being worked on so stay tunned.

senthilarul, 2019-08-31:
Is there any update on this issue. PX4 firmware was working fine for me over the last few months, but since the last 3 days it shows the this error 

`ODOM: Ex: Could not find a connection between 'local_origin_ned' and 'fcu' because they are not part of the same tree.Tf has two or more unconnected trees.`

I even tried building the latest version of firmware from scratch and it shows the same error. Is there a fix for this?

Apurv5, 2019-09-01:
Has anyone got any solution for this? 

I am also facing the same problem.

jungx148, 2019-09-02:
Having same problem and waiting the update.

TSC21, 2019-09-02:
Solution coming in https://github.com/mavlink/mavros/pull/1300

sasakou0911, 2019-09-06:
Hi, I've tried to build the latest firmware and I'm facing with the same problem. I'm not sure what should I do for solving it. Should I wait for a while for the update or can I modify something inside the Firmware?

TSC21, 2019-09-06:
@all the issue your are facing won't break any functionality of MAVROS. Please allow #1300 to be reviewed and merged and as soon as it gets upstream, we will do the release.

dftossem, 2019-09-16:
Having same problem and waiting the update.

Thanks for all your work guys.

Apurv5, 2019-09-16:
Use this launch file and problem will be solved.

<?xml version="1.0"?>
<launch>
    <include file="$(find px4)/launch/mavros_posix_sitl.launch"/>

    <node name="task_3" pkg="task_3" type="task_3_node" output="screen"/>
    <node name ="link1_broadcaster" pkg="tf" type="static_transform_publisher" args="0 0 0 0 0 0 local_origin_ned fcu 10" />
</launch>

TSC21, 2019-09-16:
> Use this launch file and problem will be solved.
> 
> ```
> <node name="task_3" pkg="task_3" type="task_3_node" output="screen"/>
> <node name ="link1_broadcaster" pkg="tf" type="static_transform_publisher" args="0 0 0 0 0 0 local_origin_ned fcu 10" />
> ```

That doesn't solve anything. Just hides the problem ;)

closed 2019-10-07 by TSC21

alcidesmig, 2019-10-19:
I'm trying to collect data from mavros/global_position/compass_hdg and mavros/global_position/global and this topics are not being published. I'm having the error 

> ODOM: Ex: Could not find a connection between 'local_origin_ned' and 'fcu' because they are not part of the same tree.Tf has two or more unconnected trees.

Is there any relationship between the error and the topics?
What can I do?

Thank you all!

TSC21, 2019-10-19:
> I'm trying to collect data from mavros/global_position/compass_hdg and mavros/global_position/global and this topics are not being published. I'm having the error the same error of the issue:
> 
> > ODOM: Ex: Could not find a connection between 'local_origin_ned' and 'fcu' because they are not part of the same tree.Tf has two or more unconnected trees.
> 
> Is there any relationship between the error and the topics?
> What can I do?
> 
> Thank you all!

Yes. 0.33 was already release and soon the fix should be available through apt. Anyway, if you want this fixed now, build or install MAVROS from source.

TSC21, 2019-10-19:
Actually, I rephrase: the problem you are having has nothing to do with the TF issue. The more obvious issue I see is that you don't have GPS data on your autopilot. Anyway, different issue, so please do not continue the discussion in this topic.

liangpu597, 2019-12-25:
i have same troubles when i use : roslaunch px4 mavros_posix_sitl.launch
I can't understand these solution.
please tell me how to solve? thanks a lot !

alcidesmig, 2019-12-25:
> My Systemis : Ubuntu 18.04 + PX4+ ROS Melodic Morenia.
> I am in full simulate, I don't have pixwark
> i have same troubles when i use : roslaunch px4 mavros_posix_sitl.launch
> I can't understand these solution.
> please tell me how to solve? thanks a lot !

Try to source catkin_ws/devel/setup.bash and /opt/ros/.../setup.bash. My problem was resolved doing it. Maybe you can try to run the command inside the catkin_ws folder and rebuild the project before sourcing the environment variables.

PR #1300 (2019-10-07): TF tree and odometry plugin

With this PR a proper tf tree is built and the odometry plugin simplified. This should simplify providing the FCU with external position/odometry information.

Fixes #1293

The suggested changes include:
- odom plugin is not publishing tfs anymore.
- multiple static tfs are published in uas_data.cpp. (includes a fix proposed by @mhkabir )
- transform between local frame and body frame is published by local_position. 
  (needed to change the frame_id in px4_config.yaml)
- change default frame_id in px4_config.yaml for all plugin.  (to be discussed)
- simplify the od [... cut]