mavlink/mavros #1294: unexpected command while setting mode

v-to, 2019-08-14:
### Issue details

_Please describe the problem, or desired feature_

I am trying to send mavlink commands via mavros to ardupilot rover.

mavsafety arm/disarm is working fine, mavcmd for setting servos as well, but I can't change mode of the autopilot

I am sending message with: rosrun mavros mavsys mode -b 4
I am getting response: CMD: Unexpected command 11, result 3

I run out of ideas how to solve this. Any idea of what might be the issue?
### MAVROS version and platform

Mavros: 0.29.2
ROS: Lunar
Ubuntu: 16.04

### Autopilot type and version

[ x ] ArduPilot
[  ] PX4

Version: 3.5.1 ArduRover

### Node logs

[ INFO] [1565769114.857175129]: FCU URL: /dev/ttyS1:57600
[ INFO] [1565769114.858655472]: serial0: device: /dev/ttyS1 @ 57600 bps
[ INFO] [1565769114.858987507]: GCS URL: udp://@
[ INFO] [1565769114.859150206]: udp1: Bind address: 0.0.0.0:14555
[ INFO] [1565769114.878820334]: Plugin 3dr_radio loaded
[ INFO] [1565769114.880262191]: Plugin 3dr_radio initialized
[ INFO] [1565769114.880326545]: Plugin actuator_control loaded
[ INFO] [1565769114.883385227]: Plugin actuator_control initialized
[ INFO] [1565769114.891112436]: Plugin adsb loaded
[ INFO] [1565769114.903722850]: Plugin adsb initialized
[ INFO] [1565769114.904065093]: Plugin altitude loaded
[ INFO] [1565769114.907913076]: Plugin altitude initialized
[ INFO] [1565769114.908197238]: Plugin cam_imu_sync loaded
[ INFO] [1565769114.910433955]: Plugin cam_imu_sync initialized
[ INFO] [1565769114.910780415]: Plugin command loaded
[ INFO] [1565769114.929798884]: Plugin command initialized
[ INFO] [1565769114.930089141]: Plugin companion_process_status loaded
[ INFO] [1565769114.940013238]: Plugin companion_process_status initialized
[ INFO] [1565769114.940284215]: Plugin debug_value loaded
[ INFO] [1565769114.960504624]: Plugin debug_value initialized
[ INFO] [1565769114.960797651]: Plugin distance_sensor loaded
[ WARN] [1565769114.964736514]: DS: plugin not configured!
[ INFO] [1565769114.964815354]: Plugin distance_sensor initialized
[ INFO] [1565769114.965073022]: Plugin fake_gps loaded
[ INFO] [1565769114.999936357]: Plugin fake_gps initialized
[ INFO] [1565769115.000323849]: Plugin ftp loaded
[ INFO] [1565769115.023821379]: Plugin ftp initialized
[ INFO] [1565769115.024202377]: Plugin global_position loaded
[ INFO] [1565769115.078218371]: Plugin global_position initialized
[ INFO] [1565769115.078530845]: Plugin gps_rtk loaded
[ INFO] [1565769115.088804507]: Plugin gps_rtk initialized
[ INFO] [1565769115.089232083]: Plugin hil loaded
[ INFO] [1565769115.132297298]: Plugin hil initialized
[ INFO] [1565769115.132528141]: Plugin home_position loaded
[ INFO] [1565769115.139230362]: Plugin home_position initialized
[ INFO] [1565769115.139434098]: Plugin imu loaded
[ INFO] [1565769115.150187645]: Plugin imu initialized
[ INFO] [1565769115.150401325]: Plugin local_position loaded
[ INFO] [1565769115.160923382]: Plugin local_position initialized
[ INFO] [1565769115.161126213]: Plugin log_transfer loaded
[ INFO] [1565769115.166055331]: Plugin log_transfer initialized
[ INFO] [1565769115.166277394]: Plugin manual_control loaded
[ INFO] [1565769115.172050811]: Plugin manual_control initialized
[... 239 lines cut ...]
    message: "Low voltage"
    hardware_id: "/dev/ttyS1:57600"
    values: 
      - 
        key: "Voltage"
        value: "0.00"
      - 
        key: "Current"
        value: "-0.0"
      - 
        key: "Remaining"
        value: "-1.0"
---
### Check ID

WARNING: mavros/target_system_id not set. Used default value: 1
WARNING: mavros/target_component_id not set. Used default value: 1
NOTE: Target parameters may be unset, but that may be result of incorrect --mavros-ns option.Double check results!
OK. I got messages from 1:1.

---
Received 442 messages, from 2 addresses
sys:comp   list of messages
 255:190   0, 111
  1:1     0, 1, 2, 22, 129, 137, 152, 150, 24, 27, 29, 30, 32, 33, 34, 163, 36, 165, 42, 49, 178, 158, 182, 62, 65, 194, 74, 193, 111, 241, 242, 116, 125

vooon, 2019-08-14:
I suppose that better to use string-custom modes, i'm not sure does base mode mask still works ok.
Take a look at http://wiki.ros.org/mavros/CustomModes , it's a bit outdated, but most of modes should be there.

v-to, 2019-08-16:
Thanks vooon. I've tried this already. For custom modes I got "Unsupported FCU" :(
Any ideas why?

rgds 

VV

vooon, 2019-08-16:
Check Rover modes, if not helps, then check that https://github.com/mavlink/mavros/blob/master/mavros/src/lib/uas_stringify.cpp#L197

v-to, 2021-06-21:
Hi vooon,

After shifting another project I am back to this place.
The problm is still unsolved however :(
I have looked over all the parameters.
The newest idea is that maybe there is an issue with FCU type?

I can see in code that MAV_TYPE should be SURFACE_BOAT. 
In my mavros/diagnostics I have 
key: "Vehicle type"
value: "Surface vessel" and not surface boat.
Can this be an issue?
Or maybe you have other ideas?

Kind regards

VV

vooon, 2021-06-24:
That's ok, vehicle type uses shortened description for entry: https://mavlink.io/en/messages/common.html#MAV_TYPE_SURFACE_BOAT

`rosrun mavros mavsys mode -c AUTO` - that didn't work?

closed 2026-11-15 by github-actions[bot]