mavlink/mavros #1309: Pixhawk PWM AUX Control

caio-freitas, 2019-09-06:
I've been trying to control a servo PWM motor with a Pixhawk 2.4.8, on AUX1, running PX4 v1.9.2 and ROS Kinetic, and **all** I could get was an Error LED from the Pixhawk.
This is the code:
`#include <ros/ros.h>
#include <mavros_msgs/CommandBool.h>
#include <std_msgs/String.h>
#include <mavros_msgs/ActuatorControl.h>
#include <mavros_msgs/CommandLong.h>
#include <mavros_msgs/CommandLongRequest.h>

std_msgs::String message;
mavros_msgs::ActuatorControl servocon;

int main(int argc, char **argv)	{
    ros::init(argc, argv, "servo_control");
    ros::NodeHandle nh;
    ros::Rate rate(20.0);
    ros::ServiceClient command_client = nh.serviceClient<mavros_msgs::CommandLong>("/mavros/cmd/command");
    int index=1;
    float value=1900;   // thruster value
    int pulse_width = (value + 1) * 400 + 1100;
    mavros_msgs::CommandLong srv;
    srv.request.command = 184; //because it didn't accept mavros_msgs::CommandLong::CMD_DO_SET_SERVO;//
		srv.request.param1 = index + 1; // servos are 1-indexed here
	  srv.request.param2 = pulse_width;
		while (ros::ok()) {
	  bool result = command_client.call(srv);
		ros::spinOnce();
		rate.sleep();
	}
	return 0;
}
`
Does anyone know how to 
### Issue details

_Please describe the problem, or desired feature_

### MAVROS version and platform

Mavros: 0.29.2
ROS: Kinetic
Ubuntu: 16.04.5

### Autopilot type and version

[  ] ArduPilot
[ X ] PX4

Version: ?3.7.1?

### Node logs
It logs nothing

### Diagnostics
header: 
  seq: 59
  stamp: 
    secs: 1567798296
    nsecs: 746577989
  frame_id: ''
status: 
  - 
    level: 0
    name: "mavros: FCU connection"
    message: "connected"
    hardware_id: "/dev/ttyUSB0:57600"
    values: 
      - 
        key: "Received packets:"
        value: "1150"
      - 
        key: "Dropped packets:"
        value: "0"
      - 
        key: "Buffer overruns:"
        value: "0"
      - 
        key: "Parse errors:"
        value: "0"
      - 
        key: "Rx sequence number:"
        value: "121"
      - 
        key: "Tx sequence number:"
        value: "38"
      - 
        key: "Rx total bytes:"
        value: "54243"
      - 
        key: "Tx total bytes:"
        value: "16199"
      - 
        key: "Rx speed:"
        value: "2253.000000"
      - 
        key: "Tx speed:"
        value: "736.000000"
  - 
    level: 2
    name: "mavros: GPS"
    message: "No satellites"
    hardware_id: "/dev/ttyUSB0:57600"
    values: 
      - 
        key: "Satellites visible"
        value: "0"
      - 
        key: "Fix type"
        value: "0"
      - 
        key: "EPH (m)"
        value: "99.99"
      - 
        key: "EPV (m)"
        value: "99.99"
  - 
    level: 0
    name: "mavros: Heartbeat"
    message: "Normal"
    hardware_id: "/dev/ttyUSB0:57600"
    values: 
      - 
        key: "Heartbeats since startup"
        value: "32"
      - 
        key: "Frequency (Hz)"
        value: "0.999983"
      - 
        key: "Vehicle type"
        value: "Hexarotor"
      - 
        key: "Autopilot type"
        value: "PX4 Autopilot"
      - 
        key: "Mode"
        value: "MANUAL"
      - 
        key: "System status"
        value: "Standby"
  - 
    level: 2
    name: "mavros: System"
    message: "Sensor helth"
    hardware_id: "/dev/ttyUSB0:57600"
    values: 
      - 
        key: "Sensor present"
        value: "0x000E002F"
      - 
        key: "Sensor enabled"
        value: "0x0021000F"
      - 
        key: "Sensor helth"
        value: "0x000E002F"
      - 
        key: "3D gyro"
        value: "Ok"
      - 
        key: "3D accelerometer"
        value: "Ok"
      - 
        key: "3D magnetometer"
        value: "Ok"
      - 
        key: "absolute pressure"
        value: "Ok"
      - 
        key: "rc receiver"
        value: "Fail"
      - 
        key: "AHRS subsystem health"
        value: "Fail"
      - 
        key: "CPU Load (%)"
        value: "33.1"
      - 
        key: "Drop rate (%)"
        value: "0.0"
      - 
        key: "Errors comm"
        value: "0"
      - 
        key: "Errors count #1"
        value: "0"
      - 
        key: "Errors count #2"
        value: "0"
      - 
        key: "Errors count #3"
        value: "0"
      - 
        key: "Errors count #4"
        value: "0"
  - 
    level: 0
    name: "mavros: Battery"
    message: "Normal"
    hardware_id: "/dev/ttyUSB0:57600"
    values: 
      - 
        key: "Voltage"
        value: "65.54"
      - 
        key: "Current"
        value: "-0.0"
      - 
        key: "Remaining"
        value: "-1.0"
  - 
    level: 2
    name: "mavros: Time Sync"
    message: "No events recorded."
    hardware_id: "/dev/ttyUSB0:57600"
    values: 
      - 
        key: "Timesyncs since startup"
        value: "0"
      - 
        key: "Frequency (Hz)"
        value: "0.000000"
      - 
### Check ID

OK. I got messages from 1:1.

---
Received 481 messages, from 1 addresses
sys:comp   list of messages
  1:1     0, 1, 83, 36, 230, 32, 105, 74, 331, 77, 111, 141, 241, 147, 340, 245, 22, 24, 4, 30

Ken-Thorburn, 2019-09-19:
Were you able to control an aux port with a mavros command? I just started researching the possibility today.

dibukk95, 2019-09-25:
hai
i am also going through the same for last few days.but i am not able to control the aux via mavros.i tried diiferent msgs & various topics but still not able to do.if you figured it please share.

thanks in advance

purvikpatel, 2020-11-20:
Hi @dibukk95
 Did you find the solution ?

ntrinite, 2023-03-01:
Bump for solution?

tiralonghipol, 2023-03-07:
@caio-freitas  hei, any news on this?

closed 2026-11-14 by github-actions[bot]