mavlink/mavros #1310: Installing mavros on Ros melodic for Nvidia Nano

fishworx, 2019-09-07:
Please see below, mavros for melodic did not install....

sudo apt-get install ros-melodic-mavros ros-melodic-mavros-extras

Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 ros-melodic-mavros : Depends: ros-melodic-diagnostic-updater but it is not going to be installed
                      Depends: ros-melodic-pluginlib but it is not going to be installed
                      Depends: ros-melodic-rosconsole-bridge but it is not going to be installed
                      Depends: ros-melodic-roscpp but it is not going to be installed
                      Depends: ros-melodic-rospy but it is not going to be installed
                      Depends: ros-melodic-tf2-ros but it is not going to be installed
 ros-melodic-mavros-extras : Depends: ros-melodic-roscpp but it is not going to be installed
                             Depends: ros-melodic-tf but it is not going to be installed
                             Depends: ros-melodic-urdf but it is not going to be installed
E: Unable to correct problems, you have held broken packages.

[image]
This is only bug and feature tracker, please use it
to report bugs or request features.

----

### Issue details

_Please describe the problem, or desired feature_

### MAVROS version and platform

Mavros: ?0.18.4?
ROS: ?Kinetic?
Ubuntu: ?16.04?

### Autopilot type and version

[  ] ArduPilot
[  ] PX4

Version: ?3.7.1?

### Node logs

```
copy output of mavros_node. Usually console where you run roslaunch
```

### Diagnostics

```
place here result of:
rostopic echo -n1 /diagnostics
```

### Check ID

```
rosrun mavros checkid
```

Jaeyoung-Lim, 2019-09-07:
Try 'source /opt/ros/melodic/setup.bash'

fishworx, 2019-09-07:
not sure if (Try 'source /opt/ros/melodic/setup.bash') worked....

[image]

[image]

TSC21, 2019-09-07:
@fishworx do you even have ROS Melodic installed in your system? From the output above, it seems like you don't.

TSC21, 2019-09-07:
http://wiki.ros.org/melodic/Installation/Ubuntu

fishworx, 2019-09-08:
@TSC21 , reflashing sd card now and restarting melodic install from scratch....

fishworx, 2019-09-09:
It looks like it worked.... How do i test it works? Also, it is necessary to install thee GeographicLib datasets?

https://github.com/mavlink/mavros/blob/master/mavros/README.md#installation

[image]

canberkgurel, 2019-12-12:
yes, GeographicLib is required.

ashirsat, 2021-06-02:
@fishworx If that solved your issue please close the ticket. Thanks

closed 2021-06-02 by vooon