mavlink/mavros #1320: error when build the mavros

onionysy, 2019-09-26:
This is only bug and feature tracker, please use it
to report bugs or request features.

----

### Issue details

_Please describe the problem, or desired feature_
when i build the mavros, there are something wrong with the build
i hope you can give me some advice with this error, thank you!

`Errors     << mavros_extras:make /home/coamc/catkin_ws/logs/mavros_extras/build.make.002.log
/home/coamc/catkin_ws/src/mavros/mavros_extras/src/plugins/onboard_computer_status.cpp: In member function ‘void mavros::extra_plugins::OnboardComputerStatusPlugin::status_cb(const ConstPtr&)’:
/home/coamc/catkin_ws/src/mavros/mavros_extras/src/plugins/onboard_computer_status.cpp:59:25: error: ‘ONBOARD_COMPUTER_STATUS’ is not a member of ‘mavlink::common::msg’
   mavlink::common::msg::ONBOARD_COMPUTER_STATUS status {};
                         ^~~~~~~~~~~~~~~~~~~~~~~
/home/coamc/catkin_ws/src/mavros/mavros_extras/src/plugins/onboard_computer_status.cpp:59:25: note: suggested alternative: ‘SMART_BATTERY_STATUS’
   mavlink::common::msg::ONBOARD_COMPUTER_STATUS status {};
                         ^~~~~~~~~~~~~~~~~~~~~~~
                         SMART_BATTERY_STATUS
/home/coamc/catkin_ws/src/mavros/mavros_extras/src/plugins/onboard_computer_status.cpp:60:3: error: ‘status’ was not declared in this scope
   status.time_usec = req->header.stamp.toNSec() / 1000;     //!< [microsecs]
   ^~~~~~
/home/coamc/catkin_ws/src/mavros/mavros_extras/src/plugins/onboard_computer_status.cpp:60:3: note: suggested alternative: ‘static’
   status.time_usec = req->header.stamp.toNSec() / 1000;     //!< [microsecs]
   ^~~~~~
   static
make[2]: *** [CMakeFiles/mavros_extras.dir/src/plugins/onboard_computer_status.cpp.o] Error 1
make[1]: *** [CMakeFiles/mavros_extras.dir/all] Error 2
make: *** [all] Error 2
`

### MAVROS version and platform

Mavros: ?0.18.4?
ROS: melodic
Ubuntu: 18.04

### Autopilot type and version

[  ] ArduPilot
[  ] PX4

Version: ?3.7.1?

### Node logs

```
copy output of mavros_node. Usually console where you run roslaunch
```

### Diagnostics

```
place here result of:
rostopic echo -n1 /diagnostics
```

### Check ID

```
rosrun mavros checkid
```

Jaeyoung-Lim, 2019-09-26:
You need to update mavlink in your workspace

onionysy, 2019-09-26:
> You need to update mavlink in your workspace

sorry， i am new in this field. Can you give me the specific method of operation?

onionysy, 2019-09-26:
> update mavlink in your workspace

    version: release/melodic/mavlink/2019.9.9-1
and my version is this

Jaeyoung-Lim, 2019-09-27:
The tag seems correct. Try to do a clean build

closed 2020-08-12 by onionysy