mavlink/mavros #1330: Cannot specify frame for output of obstacle_distance plugin

mortenfyhn, 2019-10-11:
### Issue details

The `obstacle_distance` plugin sends a mavlink [OBSTACLE_DISTANCE](https://mavlink.io/en/messages/common.html#OBSTACLE_DISTANCE) message, but the plugin does not specify the `frame` field of the message, so it defaults to `MAV_FRAME_GLOBAL`. I reckon the user should be able to specify this through a ROS parameter for the plugin. Do you agree? I can make the PR.

The upcoming v1.10.0 PX4 release has some changes to the collision prevention code where it now supports frames GLOBAL, LOCAL_NED, and BODY_FRD for obstacle distances, so it would be convenient to update the plugin accordingly.

### MAVROS version and platform

Mavros: 0.33.0
ROS: Melodic
Ubuntu: 18.04

### Autopilot type and version

[ ] ArduPilot
[x] PX4

Version: v1.10.0-beta2

### Node logs

N/A.

### Diagnostics

N/A.
### Check ID

N/A.

TSC21, 2019-10-11:
@mortenfyhn a PR is welcomed. Though, take into consideration that it should take some time before a new MAVROS release is done, considering 0.33.0 was released yesterday.

mortenfyhn, 2019-10-12:
Cool, I'll make the PR. And thanks for notifying me about the release, shouldn't be a problem though.

PR #1332 (2019-10-15): obstacle distance plugin: Add ROS param for mavlink frame
Makes it possible to specify the 'frame' field in the MAVLink
OBSTACLE_DISTANCE message sent by this plugin. Previously the
frame was not defined, which means it defaulted to MAV_FRAME_GLOBAL.
(See https://mavlink.io/en/messages/common.html#OBSTACLE_DISTANCE)
The default frame is therefore still MAV_FRAME_GLOBAL.

Fix #1330.

Seems to work fine, for instance when I set frame `BODY_FRD` and check the corresponding message in the px4 simulator, I get `frame: 12` which matches (https://mavlink.io/en/messages/common.html#MAV_FRAME).

[image [... cut]
closed 2019-10-15 by vooon