mavlink/mavros #1331: SITL error: "fcu" passed to lookupTransform argument source_frame does not exist.

ryrocha, 2019-10-11:
### Issue details

After building 0.33.0 I ran into two errors I hadn't seen before:
```
Plugin onboard_computer_status load exception: MultiLibraryClassLoader: Could not create object of class type mavros::extra_plugins::OnboardComputerStatusPlugin as no factory exists for it. Make sure that the library exists and was explicitly loaded through MultiLibraryClassLoader::loadLibrary()
```
```
ODOM: Ex: "fcu" passed to lookupTransform argument source_frame does not exist.
```
I was able to reproduce with a clean build of v1.9.2 and 0.33.0 running:
```
roslaunch px4 mavros_posix_sitl.launch
```
My tests on a real MAV using v1.9.2 and 0.33.0 do not show either of these errors.

### MAVROS version and platform

Mavros: 0.33.0
ROS: Kinetic
Ubuntu: 16.04

### Autopilot type and version

PX4 - v1.9.2

### Node logs

```
started roslaunch server http://localhost:35881/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/hrlv_ez4_pub/field_of_view: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/frame_id: hrlv_ez4_sonar
 * /mavros/distance_sensor/hrlv_ez4_pub/id: 0
 * /mavros/distance_sensor/hrlv_ez4_pub/orientation: PITCH_270
 * /mavros/distance_sensor/hrlv_ez4_pub/send_tf: True
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/laser_1_sub/id: 3
 * /mavros/distance_sensor/laser_1_sub/orientation: PITCH_270
 * /mavros/distance_sensor/laser_1_sub/subscriber: True
 * /mavros/distance_sensor/lidarlite_pub/field_of_view: 0.0
 * /mavros/distance_sensor/lidarlite_pub/frame_id: lidarlite_laser
 * /mavros/distance_sensor/lidarlite_pub/id: 1
 * /mavros/distance_sensor/lidarlite_pub/orientation: PITCH_270
 * /mavros/distance_sensor/lidarlite_pub/send_tf: True
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/sonar_1_sub/id: 2
 * /mavros/distance_sensor/sonar_1_sub/orientation: PITCH_270
 * /mavros/distance_sensor/sonar_1_sub/subscriber: True
 * /mavros/fake_gps/eph: 2.0
 * /mavros/fake_gps/epv: 2.0
 * /mavros/fake_gps/fix_type: 3
 * /mavros/fake_gps/geo_origin/alt: 408.0
 * /mavros/fake_gps/geo_origin/lat: 47.3667
 * /mavros/fake_gps/geo_origin/lon: 8.55
[... 509 lines cut ...]
      - 
        key: "Last RTT (ms)"
        value: "0.000000"
      - 
        key: "Mean RTT (ms)"
        value: "0.341880"
      - 
        key: "Last remote time (s)"
        value: "12.852000000"
      - 
        key: "Estimated time offset (s)"
        value: "0.147999996"
---
```

### Check ID

```
OK. I got messages from 1:1.

---
Received 6131 messages, from 1 addresses
sys:comp   list of messages
  1:1     0, 1, 2, 4, 140, 141, 147, 24, 30, 31, 32, 33, 36, 74, 331, 76, 340, 87, 230, 105, 111, 241, 242, 245
```

TSC21, 2019-10-11:
@ryrocha are you using a custom px4_config file or the one provided with the release?

ryrocha, 2019-10-11:
I did not modify the px4_config.

TSC21, 2019-10-11:
@kamilritz I think this is related with the latest changes on the odom plugin and config file. Can you please cross check? Thank you.

kamilritz, 2019-10-11:
Yeah, that seems related to the changes in the odom plugin. The parameters in the px4_config file seem to be right:
```
 * /mavros/odometry/fcu/odom_child_id_des: base_link
 * /mavros/odometry/fcu/odom_parent_id_des: map
```
The only other way that it would look up a transform related to frame "fcu" is when something is publishing on the `mavros/odometry/out` topic. 
@TSC21 Do you know if there is something publishing on this topic?

kamilritz, 2019-10-11:
@ryrocha Thanks a lot for raising this. Can you check if something is publishing on `mavros/odometry/out`?

ryrocha, 2019-10-11:
No problem! Thanks for your recent work with the GPS/vision fusion. Great stuff! 

Doesn't look like anything is being published on that topic.

kamilritz, 2019-10-13:
I can have a look at this on Monday. But in general, I am not sure if the Firmware version 1.9.2  will run successfully with the updates in the MAVROS odometry plugin, in case you want to use this. I also changed a few things in the mavlink receiver on the Firmware side. So you would need to use this [Firmware version](https://github.com/PX4/Firmware/releases/tag/v1.10.0-beta1).

ryrocha, 2019-10-14:
Sounds good. Thank you!

kamilritz, 2019-10-15:
I tried now with v1.10.0-beta1 and current master. I can not reproduce your error with any of these Firmware versions.
@TSC21 Can you try to recreate this error?

ryrocha, 2019-10-16:
I tried a clean build of mavros again and am not longer seeing the error with v1.10.0-beta1. The only thing I can think of that might have caused this error before was failing to clean the catkin workspace prior to building a new mavros version. Sorry for any inconveniences.

closed 2019-10-16 by ryrocha