mavlink/mavros #1333: Change Pixhawk Flight Mode With MAVROS

apoissant95, 2019-10-14:
----

### Issue details

I am trying to change the flight mode of my Pixhawk 4 from Mission to Manual. I am able to do so through qgroundcontrol (qgc), but am unable to do so through my Odroid. My Odroid is running the ROS master and is able to properly communicate to my Pixhawk. I have tried changing the flight mode through my Odroids terminal using the command <rostopic pub -r 10 /mavors/state mavros/msgs/State '{mode: "MANUAL"}'>, but the new mode is not changed in qgc. However, I am able to see the publisher command is received by the Pixhawk because when I execute <rostopic echo /mavros/state>, I see the mode as MANUAL. 

So how do I properly use MAVROS to change the flight mode? 

### MAVROS version and platform

Mavros: 0.28.0
ROS: Melodic
Ubuntu: 18.04

### Autopilot type and version

[  ] ArduPilot
[ X ] PX4

Version: 1.9.2

### Node logs

```
... logging to /home/odroid/.ros/log/f395ada2-ee83-11e9-84d5-001e0636a853/roslaunch-odroid-2937.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://172.16.0.192:40337/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/hrlv_ez4_pub/field_of_view: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/frame_id: hrlv_ez4_sonar
 * /mavros/distance_sensor/hrlv_ez4_pub/id: 0
 * /mavros/distance_sensor/hrlv_ez4_pub/orientation: PITCH_270
 * /mavros/distance_sensor/hrlv_ez4_pub/send_tf: True
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/laser_1_sub/id: 3
 * /mavros/distance_sensor/laser_1_sub/orientation: PITCH_270
 * /mavros/distance_sensor/laser_1_sub/subscriber: True
 * /mavros/distance_sensor/lidarlite_pub/field_of_view: 0.0
 * /mavros/distance_sensor/lidarlite_pub/frame_id: lidarlite_laser
 * /mavros/distance_sensor/lidarlite_pub/id: 1
 * /mavros/distance_sensor/lidarlite_pub/orientation: PITCH_270
 * /mavros/distance_sensor/lidarlite_pub/send_tf: True
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/sonar_1_sub/id: 2
 * /mavros/distance_sensor/sonar_1_sub/orientation: PITCH_270
 * /mavros/distance_sensor/sonar_1_sub/subscriber: True
 * /mavros/fake_gps/eph: 2.0
 * /mavros/fake_gps/epv: 2.0
 * /mavros/fake_gps/fix_type: 3
 * /mavros/fake_gps/geo_origin/alt: 408.0
 * /mavros/fake_gps/geo_origin/lat: 47.3667
[... 407 lines cut ...]
        value: "3.082309"
      - 
        key: "Mean RTT (ms)"
        value: "4.070913"
      - 
        key: "Last remote time (s)"
        value: "96.123676000"
      - 
        key: "Estimated time offset (s)"
        value: "1571058588.440886021"
---

```

### Check ID

```
OK. I got messages from 1:1.

---
Received 1433 messages, from 1 addresses
sys:comp   list of messages
  1:1     0, 65, 2, 36, 70, 1, 42, 76, 77, 111, 147, 340, 245, 4, 74

```

netgittemp, 2019-10-31:
Hi, I think it seems likethe same issue with mime.
[https://github.com/mavlink/mavros/issues/1340#issue-515723520](https://github.com/mavlink/mavros/issues/1340#issue-515723520)

I am working with PX4 and odroid XU4 for running offboard mode code, and I got the same issue with yours. Now I am working on my laptop to figure out what is the problem with gazebo simulation.

It would be happy if there will be a solution..

TSC21, 2019-11-01:
> Hi, I think it seems like the same issue with mime.
> [#1340 (comment)](https://github.com/mavlink/mavros/issues/1340#issue-515723520)
> 
> I am working with PX4 and odroid XU4 for running offboard mode code, and I got the same issue with yours. Now I am working on my laptop to figure out what is the problem with gazebo simulation.
> 
> It would be happy if there will be a solution..

@netgittemp not at all a related issue.

@apoissant95 the command you are looking for is:
```console
rosrun mavros mavsys mode -b <ENUM VALUE>
```
Where the `ENUM_VALUE` is the mode you are trying to change to, like `OFFBOARD` or `MANUAL`.

closed 2019-11-01 by TSC21

apoissant95, 2019-11-01:
@TSC21 That worked perfectly. Thank you!