mavlink/mavros #1337: Understanding of gcs_bridge and gcs_url

TobiMiller, 2019-10-29:
Hey,

i want to understand the gcs_bridge plugin and  the gcs_url. I want to forward the mavlink messages to QGroundControl to get telemtry data and to send params/set waypoints etc. over wifi. Is this possible?

I have set up <arg name="gcs_url" default="tcp-l://0.0.0.0:5760" />.

I can connect via QGroundControl. But i have just a one way communication. What do i need to do to get two way communication?

best regards

wly2014, 2020-11-20:
do you have any idea about this now? thx

EricPedley, 2024-11-14:
I just set it up the other way around. Have QGroundControl forward the messages to mavros with its "mavlink forwarding" setting (https://docs.qgroundcontrol.com/master/en/qgc-user-guide/settings_view/mavlink.html). I suppose getting mavros to forward the packets is possible somehow but I also couldn't find _any_ documentation for how to do it.

closed 2026-11-13 by github-actions[bot]