mavlink/mavros #1339: RTT timesync issue after reboot (and sometimes during normal operation)

nbelanger99, 2019-10-30:
This is only bug and feature tracker, please use it
to report bugs or request features.

----

### Issue details

Issue happens when rebooting pixhawk with Mavros running, and sometimes after just running for some time (this is difficult to replicate). We see "TM : RTT too high for timesync" on the Mavros side, and invoking mavlink status in NSH, we see that packets are being dropped and the "tx rate mult" coefficient decreases to less than 1.

### MAVROS version and platform

Mavros: ?0.18.4?
ROS: ?Melodic?
Ubuntu: ?18.04?

### Autopilot type and version

[  ] ArduPilot
[ x] PX4

Version: ?1.8.2?

[image]

guomingyang0, 2019-11-19:
https://github.com/mavlink/mavros/pull/1300

mtbsteve, 2020-01-22:
Same problem here with ROS Melodic, Ubuntu 18.04 and Arducopter 3.6.12.
Workaround: set 
`timesync_rate: 0    # TIMESYNC rate in Hertz (10.0 is default, feature disabled if 0.0)`
in file `/opt/ros/melodic/share/mavros/launch/apm_config.yaml`
I could not encounter any negative side effects yet.

kylelindgren, 2020-05-22:
I'm afraid changing `timesync_rate` to 0 in the mavros launch config file did not solve my issue. I still see the warnings output and also see erratic behavior from the sim vehicle.

closed 2026-11-13 by github-actions[bot]