mavlink/mavros #1340: Offboard control does not work!

netgittemp, 2019-10-31:
### Issue details

Hi,
I installed mavros using
`$ sudo apt-get install ros-kinetic-mavros ros-kinetic-mavros-extras`
and
I successfully worked with gazebo simulation for offboard example (https://dev.px4.io/v1.9.0/en/ros/mavros_offboard.html).
I used the following commands;
```
~/src/Firmware$ sudo make px4_sitl gazebo
$ roslaunch mavros px4.launch fcu_url:="udp://:14540@127.0.0.1:14557"
~/catkin_ws$ rosrun px4_offboard offboard_ex
```

However, after installing rotor_simulator in "catkin_ws_RotorS" folder (https://github.com/ethz-asl/rotors_simulator)
offboard mode does not work with the same commands....
I am not sure it is the problem, but for your information,
I found mavros package is connected in the "catkin_ws_RotorS" using `roscd`

Could you give some advice for resolve this issue?

### MAVROS version and platform

Mavros:
ROS: Kinetic
Ubuntu: 16.04

### Autopilot type and version

[  ] ArduPilot
[ 0 ] PX4

Version: 

### Node logs

```
~$ roslaunch mavros px4.launch fcu_url:="udp://:14540@127.0.0.1:14557"
... logging to /home/ubuntu-dell/.ros/log/b141ce30-fc1c-11e9-ae44-50eb719ed77f/roslaunch-ubuntu-dell-464.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/loader.py:409: YAMLLoadWarning: calling yaml.load() without Loader=... is deprecated, as the default Loader is unsafe. Please read https://msg.pyyaml.org/load for full details.
  data = yaml.load(text)
started roslaunch server http://ubuntu-dell:35247/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/hrlv_ez4_pub/field_of_view: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/frame_id: hrlv_ez4_sonar
 * /mavros/distance_sensor/hrlv_ez4_pub/id: 0
 * /mavros/distance_sensor/hrlv_ez4_pub/orientation: PITCH_270
 * /mavros/distance_sensor/hrlv_ez4_pub/send_tf: True
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/laser_1_sub/id: 3
 * /mavros/distance_sensor/laser_1_sub/orientation: PITCH_270
 * /mavros/distance_sensor/laser_1_sub/subscriber: True
[... 435 lines cut ...]
        value: "1.194847"
      - 
        key: "Mean RTT (ms)"
        value: "1.471830"
      - 
        key: "Last remote time (s)"
        value: "91.744000000"
      - 
        key: "Estimated time offset (s)"
        value: "1572553561.050406933"
---

```

### Check ID

```
$ rosrun mavros checkid
OK. I got messages from 1:1.

---
Received 5259 messages, from 1 addresses
sys:comp   list of messages
  1:1     0, 1, 2, 4, 140, 141, 147, 24, 30, 31, 32, 36, 74, 331, 76, 77, 83, 340, 230, 105, 111, 241, 245
```

TSC21, 2019-11-01:
@netgittemp I am trying to understand what exact problem you are describing but the information you provide is kinda short.

> However, after installing rotor_simulator in "catkin_ws_RotorS" folder (https://github.com/ethz-asl/rotors_simulator)

What does this have to do with what is described in https://dev.px4.io/v1.9.0/en/ros/mavros_offboard.html? What you see on that page is using [sitl_gazebo](https://github.com/PX4/sitl_gazebo) with PX4 SITL, meaning that PX4 is running over SITL and connected to Gazebo through Mavlink. `rotors_simulator` is a package set (where `sitl_gazebo` has its origin), but for a long time the PX4 ecosystem focused its development on `sitl_gazebo`. So if you are trying to use `rotors_simulator` with the PX4 SITL, I advise you to bring your questions there instead.

In any case, I don't see anywhere this is an issue with MAVROS, but rather with what you are trying to do with `rotors_simulator`.

netgittemp, 2019-11-01:
@TSC21  I am sorry for confusing you.
I am trying to use offboard mode with MAVROS.
As I mentioned in the post, I used

```
~/src/Firmware$ sudo make px4_sitl gazebo
$ roslaunch mavros px4.launch fcu_url:="udp://:14540@127.0.0.1:14557"
~/catkin_ws$ rosrun px4_offboard offboard_ex
```
However, the quadrotor in Gazebo does not move at all. The output of `$ rostopic echo /mavros/state/` is
```
---
header: 
  seq: 48
  stamp: 
    secs: 1572622117
    nsecs: 473311580
  frame_id: ''
connected: True
armed: False
guided: True
manual_input: False
mode: "AUTO.LOITER"
system_status: 3
---
```
Could you help me to resolve this issue?
Thanks.

netgittemp, 2019-11-01:
FYI, I reinstalled ubuntu 16.04.
and Installed ROS and mavros using Binary installation (deb)
( MAVROS pacakge is not in the `catkin_ws`, it is in the following directory.
```
$ roscd mavros
/opt/ros/kinetic/share/mavros$
```
)

Then, following commands work correctly hovering the quadrotor in Gazebo
```
~/src/Firmware$ sudo make px4_sitl gazebo
$ roslaunch mavros px4.launch fcu_url:="udp://:14540@127.0.0.1:14557"
~/catkin_ws$ rosrun px4_offboard offboard_ex
```

When I install mavros using Source installation, however, I do not know how to run offboard example. Because installing mavros generates` catkin_ws`...
I tried to create package in` /catkin_ws/src`, and run $ catkin build.
It showed error. So I make another workspace like` catkin_ws_offboard`, and build using `catkin_build`.
Then, I got the error;
```
fatal error: mavros_msgs/CommandBool.h: No such file or directory
compilation terminated.
offboard_test/CMakeFiles/offboard_test.dir/build.make:62: recipe for target 'offboard_test/CMakeFiles/offboard_test.dir/src/offboard_test.cpp.o' failed
make[2]: *** [offboard_test/CMakeFiles/offboard_test.dir/src/offboard_test.cpp.o] Error 1
CMakeFiles/Makefile2:651: recipe for target 'offboard_test/CMakeFiles/offboard_test.dir/all' failed
make[1]: *** [offboard_test/CMakeFiles/offboard_test.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
```

So now, my question is that
**How to run offboard example with MAVROS which is installed via Source installation
In this case MAVROS package is in `catkin_ws/src`**

fabrizioschiano, 2020-04-14:
Any updates here? I seem to have the same problem. It seems to be a problem with the mavros version but I don't know yet how to solve it.

closed 2026-11-13 by github-actions[bot]