mavlink/mavros #1346: Change update rate of my IMU

frme92, 2019-11-13:
### Issue details

I need to change the update rate of my 3DR Pixhawk IMU from 50 Hz to 100 Hz. I've tried with `rosrun mavros mavsys rate --all 100` from terminal but it still remains at 50. 
Can somebody show me how to do this? Thanks.

### MAVROS version and platform

Mavros: ?0.18.4?
ROS: Melodic
Ubuntu: 18.04

### Autopilot type and version

[  ] ArduPilot
[ X ] PX4

Version: Fimware 1.9.2

### Node logs

```
$ roslaunch mavros px4.launch
... logging to /home/francesco/.ros/log/9f8841d2-0631-11ea-bf51-7cb0c2639a2f/roslaunch-Enterprise-Ubuntu-5233.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

/opt/ros/melodic/lib/python2.7/dist-packages/roslaunch/loader.py:412: YAMLLoadWarning: calling yaml.load() without Loader=... is deprecated, as the default Loader is unsafe. Please read https://msg.pyyaml.org/load for full details.
  data = yaml.load(text)
started roslaunch server http://Enterprise-Ubuntu:42721/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/hrlv_ez4_pub/field_of_view: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/frame_id: hrlv_ez4_sonar
 * /mavros/distance_sensor/hrlv_ez4_pub/id: 0
 * /mavros/distance_sensor/hrlv_ez4_pub/orientation: PITCH_270
 * /mavros/distance_sensor/hrlv_ez4_pub/send_tf: True
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/laser_1_sub/id: 3
 * /mavros/distance_sensor/laser_1_sub/orientation: PITCH_270
 * /mavros/distance_sensor/laser_1_sub/subscriber: True
 * /mavros/distance_sensor/lidarlite_pub/field_of_view: 0.0
 * /mavros/distance_sensor/lidarlite_pub/frame_id: lidarlite_laser
 * /mavros/distance_sensor/lidarlite_pub/id: 1
 * /mavros/distance_sensor/lidarlite_pub/orientation: PITCH_270
 * /mavros/distance_sensor/lidarlite_pub/send_tf: True
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/sonar_1_sub/id: 2
 * /mavros/distance_sensor/sonar_1_sub/orientation: PITCH_270
 * /mavros/distance_sensor/sonar_1_sub/subscriber: True
 * /mavros/fake_gps/eph: 2.0
 * /mavros/fake_gps/epv: 2.0
 * /mavros/fake_gps/fix_type: 3
 * /mavros/fake_gps/geo_origin/alt: 408.0
 * /mavros/fake_gps/geo_origin/lat: 47.3667
[... 424 lines cut ...]
        value: "1.598668"
      - 
        key: "Mean RTT (ms)"
        value: "1.001017"
      - 
        key: "Last remote time (s)"
        value: "392.844721000"
      - 
        key: "Estimated time offset (s)"
        value: "1573661690.480628252"
---

```

### Check ID

```
OK. I got messages from 1:1.

---
Received 6075 messages, from 1 addresses
sys:comp   list of messages
  1:1     0, 1, 2, 4, 140, 141, 147, 26, 30, 31, 32, 36, 74, 331, 77, 83, 340, 230, 105, 111, 241, 116, 245

```

jinwandou, 2019-12-26:
hello, I got the same issue,  I want to change imu rate from 50 Hz to 100 Hz too, I use ardupilot flight stack，did you sloved it ? any help is welcome, thanks!

frme92, 2020-01-07:
> hello, I got the same issue, I want to change imu rate from 50 Hz to 100 Hz too, I use ardupilot flight stack，did you sloved it ? any help is welcome, thanks!

Hello, yeah I've solved in this way: 
with APM software from terrminal you can digit
`rosrun mavros mavsys rate --all 100`
in this way you change all your update rate, but if you want modify something specific you can use the help tool `rosrun mavros mavsys rate -h `
Otherwise like in my situation,with PX4 firmware, you should create an etc folder in Pixhawk micro SD containing an extra.txt file. 
It can be done directly by inserting the SD through an adapter or from the MavLInk Console (found in QGround Control, in the menu of the sheet with magnifying glass) by typing:

`cd /fs/microsd/etc`
`echo "mavlink stream -d /dev/ttyACM0 -s HIGHRES_IMU -r 100" > extras.txt`

In this way the topic _mavros / imu / data_raw_ will be written at 100 HZ. However, if we want _mavros / imu /data_ to be written, so also the attitude at this frequency, the extras file must contain:

_mavlink stream -d / dev / ttyACM0 -s ATTITUDE -r 100
mavlink stream -d / dev / ttyACM0 -s ATTITUDE_QUATERNION -r 100
mavlink stream -d / dev / ttyACM0 -s HIGHRES_IMU -r 100_

jinwandou, 2020-01-09:
> > hello, I got the same issue, I want to change imu rate from 50 Hz to 100 Hz too, I use ardupilot flight stack，did you sloved it ? any help is welcome, thanks!
> 
> Hello, yeah I've solved in this way:
> with APM software from terminal you can digit
> `rosrun mavros mavsys rate --all 100`
> in this way you change all your update rate, but if you want modify something specific you can use the help tool `rosrun mavros mavsys rate -h `
> Otherwise like in my situation,with PX4 firmware, you should create an etc folder in Pixhawk micro SD containing an extra.txt file.
> It can be done directly by inserting the SD through an adapter or from the MavLInk Console (found in QGround Control, in the menu of the sheet with magnifying glass) by typing:
> 
> `cd /fs/microsd/etc`
> `echo "mavlink stream -d /dev/ttyACM0 -s HIGHRES_IMU -r 100" > extras.txt`
> 
> In this way the topic _mavros / imu / data_raw_ will be written at 100 HZ. However, if we want _mavros / imu /data_ to be written, so also the attitude at this frequency, the extras file must contain:
> 
> _mavlink stream -d / dev / ttyACM0 -s ATTITUDE -r 100 mavlink stream -d / dev / ttyACM0 -s ATTITUDE_QUATERNION -r 100 mavlink stream -d / dev / ttyACM0 -s HIGHRES_IMU -r 100_

Thank you very much. At first I try the way you said, the max rate I can get was only 50hz on ardupilot firmware, finally I tried to flash to px4 firmware, and tested it accoring to your reply, anyway it works.

mzahana, 2023-09-05:
I would like increase the IMU rate to 200Hz. I am using ros2. Any one figured this out?
When I tried `ros2 run mavros mav sys message-interval --id 27 --rate 20`, I get ` [mavros.sys]: SYS: command plugin service call failed!`

gregoryw3, 2023-09-05:
Not sure about the error message but the --rate 20 means 20hz so if you want 200Hz youd write --rate 200

mzahana, 2023-09-05:
I solved by using the correct message ID 
[HIGHRES_IMU](https://mavlink.io/en/messages/common.html#HIGHRES_IMU)
`ros2 run mavros mav sys message-interval --id 105 --rate 100`

gregoryw3, 2023-09-05:
It's not that id 27 is wrong it's just for some reason your system didn't support it.

nnln-lno, 2025-06-28:
> I solved by using the correct message ID [HIGHRES_IMU](https://mavlink.io/en/messages/common.html#HIGHRES_IMU) `ros2 run mavros mav sys message-interval --id 105 --rate 100`

Thanks a lot.

Actually, I'm not fully understand about 105 id, anyway I will look for it

closed 2026-11-13 by github-actions[bot]