mavlink/mavros #1348: No plugin publishing for px4 estimator status.

saifullah3396, 2019-11-14:
This is only bug and feature tracker, please use it
to report bugs or request features.

----

### Issue details

Hi, my work involves using the px4-mavros integration for uavs in simulations. For that I require to  repeatedly switch on/off the estimator after every simulation iteration for which I use the px4-ekf2/local_position_estimator binaries. However I found no reliable way to get the current status of the estimator and to check whether the estimator has stabilized and started giving outputs with valid states in mavros. I saw that mavlink already provides a message for that but its not being published by any mavros plugin.

To solve the problem I've made a estimator_status plugin that publishes it with the following message:

`std_msgs/Header header
bool stable_attitude
bool stable_hor_velocity_est
bool stable_ver_velocity_est
bool stable_hor_position_est_rel
bool stable_hor_position_est_abs
bool stable_ver_position_est_abs
bool stable_ver_position_est_agl
bool const_position_mode
bool pred_hor_position_rel
bool pred_hor_position_abs
bool gps_glitch
bool accel_error`

Each bool represents the corresponding ESTIMATOR_STATUS_FLAGS defined in mavlink message definitions.h. Is this the right way to solve the problem or is there any other way? If yes then can I PR this addition?

### MAVROS version and platform

Mavros: ?0.18.4?
ROS: ?Melodic?
Ubuntu: ?18.04?

### Autopilot type and version

[  ] ArduPilot
[x] PX4

Version: ?3.7.1?

Jaeyoung-Lim, 2019-11-14:
@saifullah3396 Did you make a pull request?

saifullah3396, 2019-11-14:
@Jaeyoung-Lim Not yet.

TSC21, 2019-11-14:
Everything status related pleasr put it under the sys_status plugin. No need for another plugin.

PR #1349 (2019-11-20): Added a publisher for estimator status in sys_status.
@TSC21 I believe this way is correct? Let me know if any changes are to be made.

Fixes #1348.
closed 2019-11-20 by TSC21