mavlink/mavros #1352: px4 stil mavros offboard does not work

woongjoonchoi, 2019-11-17:
This is only bug and feature tracker, please use it
to report bugs or request features.

---- 

I do  PX4 AVOIDANCE SIMULATION
Then i run below command
rosrun mavros mavsaefety arm

arming works!

rosrun mavros mavsys mode -c OFFBOARD
Timed out!

OFFBOARD MODE does not work!

### MAVROS version and platform

Mavros: 0.33
ROS: Melodic
Ubuntu: 18.04

### Autopilot type and version

[  ] ArduPilot
[  ] PX4

Version: ?3.7.1?

### Node logs
NODES
  /
    convert_metric (nodelet/nodelet)
    gazebo (gazebo_ros/gzserver)
    local_planner_node (local_planner/local_planner_node)
    mavros (mavros/mavros_node)
    points_xyz (nodelet/nodelet)
    rqt_reconfigure (rqt_reconfigure/rqt_reconfigure)
    rviz (rviz/rviz)
    sitl_0 (px4/px4)
    spawn_model (gazebo_ros/spawn_model)
    standalone_nodelet (nodelet/nodelet)
    tf_depth_camera (tf/static_transform_publisher)

```
copy output of mavros_node. Usually console where you run roslaunch
```

### Diagnostics

```
place here result of:
rostopic echo -n1 /diagnostics
```WARNING: no messages received and simulated time is active.
Is /clock being published?

### Check ID

```
rosrun mavros checkid
```

Jaeyoung-Lim, 2019-11-17:
You need send setpoints before you can switch to offboard mode

TSC21, 2019-11-17:
Documentation: https://docs.px4.io/master/en/flight_modes/offboard.html.

closed 2019-11-17 by TSC21

woongjoonchoi, 2019-11-18:
Thanks For Answer. But How do i send SetPoints?

TSC21, 2019-11-18:
> Thanks For Answer. But How do i send SetPoints?

Please check the docs. Example with ROS and MAVROS: https://dev.px4.io/master/en/ros/mavros_offboard.html