mavlink/mavros #1354: GCS Bridge fails to launch (TCPROS error)

jveitchmichaelis, 2019-11-19:
### Issue details

Trying to run `gcs_bridge` alongside mavros results in the following error:

```
[ERROR] [1574200648.225888273]: Header from subscriber did not have the required elements: md5sum, topic, callerid
```

and:

```
drone_ws$ rosrun mavros gcs_bridge
[ INFO] [1574200648.121763753]: udp0: Bind address: 0.0.0.0:14555
[ERROR] [1574200648.228493718]: Received an invalid TCPROS header.  Each line must have an equals sign.
```

### Launch file:

```
<launch>
	<!-- vim: set ft=xml noet : -->
	<!-- example launch script for ArduPilot based FCU's -->

        <arg name="fcu_url" default="/dev/ttyS3:57600" />
	<arg name="gcs_url" default="" />
	<arg name="tgt_system" default="0" />
	<arg name="tgt_component" default="0" />
	<arg name="log_output" default="screen" />
	<arg name="fcu_protocol" default="v2.0" />
	<arg name="respawn_mavros" default="false" />

	<include file="$(find mavros)/launch/node.launch">
		<arg name="pluginlists_yaml" value="$(find mavros)/launch/apm_pluginlists.yaml" />
		<arg name="config_yaml" value="$(find mavros)/launch/apm_config.yaml" />

		<arg name="fcu_url" value="$(arg fcu_url)" />
		<arg name="gcs_url" value="$(arg gcs_url)" />
		<arg name="tgt_system" value="$(arg tgt_system)" />
		<arg name="tgt_component" value="$(arg tgt_component)" />
		<arg name="log_output" value="$(arg log_output)" />
		<arg name="fcu_protocol" value="$(arg fcu_protocol)" />
		<arg name="respawn_mavros" default="$(arg respawn_mavros)" />
	</include>
</launch>
```
GCS Bridge is run using:

`rosrun mavros gcs_bridge`

### MAVROS version and platform
```
ros-melodic-mavros/stretch,now 0.33.0-1stretch.20191112.011047 arm64 [installed]
ros-melodic-mavros-extras/stretch,now 0.33.0-1stretch.20191112.015939 arm64 [installed]
ros-melodic-mavros-extras-dbgsym/stretch 0.33.0-1stretch.20191112.015939 arm64
ros-melodic-mavros-msgs/stretch,now 0.33.0-1stretch.20191112.004211 arm64 [installed,automatic]
```

### Autopilot type and version

PX4 (hardware in the loop)
Not sure which version, but unrelated in this issue.

### Diagnostics

```
header: 
  seq: 31
  stamp: 
    secs: 1574200828
    nsecs: 129502907
  frame_id: ''
status: 
  - 
    level: 1
    name: "mavros: FCU connection"
    message: "not connected"
    hardware_id: "/dev/ttyS3:57600"
    values: 
      - 
        key: "Received packets:"
        value: "1878"
      - 
        key: "Dropped packets:"
        value: "0"
      - 
        key: "Buffer overruns:"
        value: "0"
      - 
        key: "Parse errors:"
        value: "0"
      - 
        key: "Rx sequence number:"
        value: "177"
      - 
        key: "Tx sequence number:"
        value: "0"
      - 
        key: "Rx total bytes:"
        value: "58641"
      - 
        key: "Tx total bytes:"
        value: "10480"
      - 
        key: "Rx speed:"
        value: "2640.000000"
      - 
        key: "Tx speed:"
        value: "485.000000"
  - 
    level: 0
    name: "mavros: GPS"
    message: "3D fix"
    hardware_id: "/dev/ttyS3:57600"
    values: 
      - 
        key: "Satellites visible"
        value: "9"
      - 
        key: "Fix type"
        value: "3"
      - 
        key: "EPH (m)"
        value: "1.18"
      - 
        key: "EPV (m)"
        value: "2.17"
  - 
    level: 2
    name: "mavros: Heartbeat"
    message: "No events recorded."
    hardware_id: "/dev/ttyS3:57600"
    values: 
      - 
        key: "Heartbeats since startup"
        value: "0"
      - 
        key: "Frequency (Hz)"
        value: "0.000000"
      - 
        key: "Vehicle type"
        value: "Generic micro air vehicle"
      - 
        key: "Autopilot type"
        value: "Generic autopilot"
      - 
        key: "Mode"
        value: ''
      - 
        key: "System status"
        value: "Uninit"
  - 
    level: 0
    name: "mavros: System"
    message: "Normal"
    hardware_id: "/dev/ttyS3:57600"
    values: 
      - 
        key: "Sensor present"
        value: "0x0360FC2F"
      - 
        key: "Sensor enabled"
        value: "0x02601C2F"
      - 
        key: "Sensor health"
        value: "0x0360FC2F"
      - 
        key: "3D gyro"
        value: "Ok"
      - 
        key: "3D accelerometer"
        value: "Ok"
      - 
        key: "3D magnetometer"
        value: "Ok"
      - 
        key: "absolute pressure"
        value: "Ok"
      - 
        key: "GPS"
        value: "Ok"
      - 
        key: "3D angular rate control"
        value: "Ok"
      - 
        key: "attitude stabilization"
        value: "Ok"
      - 
        key: "yaw position"
        value: "Ok"
      - 
        key: "AHRS subsystem health"
        value: "Ok"
      - 
        key: "Terrain subsystem health"
        value: "Ok"
      - 
        key: "Battery"
        value: "Ok"
      - 
        key: "CPU Load (%)"
        value: "45.1"
      - 
        key: "Drop rate (%)"
        value: "0.0"
      - 
        key: "Errors comm"
        value: "0"
      - 
        key: "Errors count #1"
        value: "0"
      - 
        key: "Errors count #2"
        value: "0"
      - 
        key: "Errors count #3"
        value: "0"
      - 
        key: "Errors count #4"
        value: "0"
  - 
    level: 1
    name: "mavros: Battery"
    message: "Low voltage"
    hardware_id: "/dev/ttyS3:57600"
    values: 
      - 
        key: "Voltage"
        value: "0.85"
      - 
        key: "Current"
        value: "1.2"
      - 
        key: "Remaining"
        value: "99.0"
  - 
    level: 0
    name: "mavros: Time Sync"
    message: "Normal"
    hardware_id: "/dev/ttyS3:57600"
    values: 
      - 
        key: "Timesyncs since startup"
        value: "297"
      - 
        key: "Frequency (Hz)"
        value: "9.000374"
      - 
        key: "Last RTT (ms)"
        value: "24.833529"
      - 
        key: "Mean RTT (ms)"
        value: "50.090870"
      - 
        key: "Last remote time (s)"
        value: "290.161266000"
      - 
        key: "Estimated time offset (s)"
        value: "0.000000000"
---
```
### Check ID

```
ERROR. I got 2 addresses, but not your target 0:0

---
Received 854 messages, from 2 addresses
sys:comp   list of messages
  2:5     262666
  1:1     0, 1, 2, 133, 129, 136, 137, 152, 147, 150, 24, 27, 29, 30, 33, 163, 165, 42, 178, 253, 182, 62, 193, 74, 111, 241, 116, 125
```

(that's odd, not sure where 2:5 is coming from)

Let me know if specific logs would help. I've omitted here as they're quite long and look mostly normal. The offending messages are above.

jveitchmichaelis, 2019-11-20:
Possibly related to https://github.com/ros/ros_comm/issues/1404?

dino-no, 2020-01-10:
Hey,
I have the same issue, can't connect through serial. 
Any updates on this issue?

But the it does work on Kintetic?

jveitchmichaelis, 2020-01-10:
Yes, it does. I had to go through some extra hoops to get this working on
Buster/aarch64 but I think on Jessie it'd be fine. Happy to help if you run
into trouble.

dino-no, 2020-01-13:
Hey again,
Thank you for the quick response. 
I'm struggling a bit to find the right platform to run the kinetic mavros on either Jetson Nano, Raspberry Pi 3B or 3B+.

Do you have any suggestions?

jveitchmichaelis, 2020-01-16:
I've run it fine on a 3B+, all those boards should be OK. I'm using a Rock960 these days.

dino-no, 2020-01-16:
Hmm, interesting. 
Which platform did you use for the 3B+? buster/jessie/xenial?

Also, i'm currently trying to set up mavros on a Pi 3B with xenial.
But I keep getting "initial compiler error: Killed (program cc1plus)", which, if i've understood it correctly, is due to insufficient memory.
Did you encounter this issue?

jveitchmichaelis, 2020-01-16:
You should add swap space, just remember to turn it off when you're done
https://linuxize.com/post/create-a-linux-swap-file/

Alternatively you can use catkin build with memory limiting, but adding
swap is easy and will be quicker:
https://catkin-tools.readthedocs.io/en/latest/verbs/catkin_build.html#configuring-memory-use

e.g. I have in my Dockerfile:

RUN pip install catkin-tools psutil
RUN cd ros_catkin_ws && catkin init && catkin config --install-space
/opt/ros/kinetic --install
 RUN cd ros_catkin_ws && catkin build --mem-limit 80% -j2 -p2

dino-no, 2020-01-16:
Cool, I'll try out Stretch.

As for the limited memory, I got it built with 'catkin build -j1'

dino-no, 2020-01-16:
One more question, just to verify.
Is it possible to establish gcs_bridge connection through radio links, like the SiK radios, or other custom radios?
(I am new to this game, and might have misunderstood the features)

jveitchmichaelis, 2020-01-16:
Yep, I'm using an RFD radio. Just set the gcs bridge url to the serial port
the radio is connected to on the companion computer (eg /dev/ttyAMA0) , and
then connect the other radio to mission planner or whatever your gcs is.

Any tcp/ip or serial link should work.

dino-no, 2020-01-16:
Great!
So i just have to the kinetic running properly...

Thank you for all the help, bwt :)

closed 2026-11-13 by github-actions[bot]